espressif/esp-adf #329: When esp_audio_play() failed, next esp_audio_sync_play() maybe return same error

xiruilin, 2020-01-22:
Hi,
When ``esp_audio_play()`` play failed, next ``esp_audio_sync_play()`` maybe return same error.
``esp_adf_libs`` is v1.5.2
The code:
```
    printf("First play: [0x%x]\n", esp_audio_play(handle, AUDIO_CODEC_TYPE_DECODER, 
	    "https://dl.spressif.com/dl/audio/ff-16b-2c-44100hz.mp3", 0));
    printf("Second play: [0x%x]\n", esp_audio_sync_play(handle, 
	    "https://dl.espressif.com/dl/audio/ff-16b-2c-44100hz.mp3", 0));
```
First play url is invalid, second  is correct.

The log:
```
I (3765) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5, que:0x3ffc7e30
I (3772) ESP_AUDIO_TASK: It's a decoder
I (3776) ESP_AUDIO_TASK: 1.CUR IN:[IN_http],CODEC:[DEC_mp3],RESAMPLE:[48000],OUT:[OUT_iis]
I (3786) ESP_AUDIO_TASK: 2.Handles,IN:0x3ffc9568,CODEC:0x3ffc9c40,FILTER:0x3ffcb464,OUT:0x3ffcb05c
I (3795) ESP_AUDIO_TASK: 2.2 Update all pipeline
I (3800) ESP_AUDIO_TASK: 2.3 Linked new pipeline
I (3806) AUDIO_PIPELINE: link el->rb, el:0x3ffc9568, tag:IN_http, rb:0x3ffcb840
I (3815) AUDIO_PIPELINE: link el->rb, el:0x3ffc9c40, tag:DEC_mp3, rb:0x3ffce17c
I (3822) AUDIO_PIPELINE: link el->rb, el:0x3ffcb464, tag:resample, rb:0x3ffd02b8
I (3830) ESP_AUDIO_TASK: 3. Previous starting...
I (3835) AUDIO_ELEMENT: [IN_http] Element task created
I (3841) AUDIO_ELEMENT: [DEC_mp3] Element task created
I (3847) AUDIO_ELEMENT: [resample] Element task created
I (3846) AUDIO_ELEMENT: [DEC_mp3] AEL_MSG_CMD_RESUME,state:1
I (3853) AUDIO_ELEMENT: [OUT_iis] Element task created
I (3859) STAGEFRIGHTMP3_DECODER: MP3 opened
I (3853) AUDIO_ELEMENT: [IN_http] AEL_MSG_CMD_RESUME,state:1
I (3853) ESP_AUDIO_CTRL: Func:_ctrl_play, Line:542, MEM Total:161556 Bytes

E (3897) esp-tls: couldn't get hostname for :dl.spressif.com:
E (3897) esp-tls: Failed to open new connection
E (3897) TRANS_SSL: Failed to open a new connection
E (3903) HTTP_CLIENT: Connection failed, sock < 0
E (3908) HTTP_STREAM: Failed to open http stream
E (3914) AUDIO_ELEMENT: [IN_http] AEL_STATUS_ERROR_OPEN
E (3920) ESP_AUDIO_TASK: Got element[IN_http] error:ESP_ERR_AUDIO_OPEN, stop pipeline
W (3928) AUDIO_PIPELINE: audio_pipeline state is not running, 1
W (3935) AUDIO_PIPELINE: audio_pipeline state is not running, 1
I (3941) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:1, Available:5, que:0x3ffc7e30
I (3949) HTTP_MP3_EXAMPLE: ESP_AUDIO_CALLBACK_FUNC, st: 5, err: 81100, src: 0
W (3954) AUDIO_ELEMENT: IN-[DEC_mp3] AEL_IO_ABORT
I (3962) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_PLAY, status:5
E (3967) STAGEFRIGHTMP3_DECODER: failed to read audio data (line 122)
E (3950) AUDIO_ELEMENT: [IN_http] AEL_MSG_CMD_ERROR
E (3980) AUDIO_ELEMENT: [DEC_mp3] AEL_STATUS_ERROR_OPEN
I (3992) STAGEFRIGHTMP3_DECODER: Closed
First play: [0x81100]
I (3996) ESP_AUDIO_TASK: Received last pos: 0 bytes
I (3999) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5, que:0x3ffc7e30
E (3996) AUDIO_ELEMENT: [DEC_mp3] AEL_MSG_CMD_ERROR
I (4012) ESP_AUDIO_TASK: It's a decoder
I (4012) ESP_AUDIO_CTRL: Func:_ctrl_play, Line:542, MEM Total:161232 Bytes

I (4022) ESP_AUDIO_TASK: 1.CUR IN:[IN_http],CODEC:[DEC_mp3],RESAMPLE:[48000],OUT:[OUT_iis]
I (4038) ESP_AUDIO_TASK: 2.Handles,IN:0x3ffc98d4,CODEC:0x3ffc9c40,FILTER:0x3ffcb464,OUT:0x3ffcb05c
W (4052) AUDIO_PIPELINE: audio_pipeline state is not running, 1
W (4058) AUDIO_PIPELINE: audio_pipeline state is not running, 1
W (4065) AUDIO_PIPELINE: audio_pipeline state is not running, 1
W (4072) AUDIO_PIPELINE: audio_pipeline state is not running, 1
I (4078) AUDIO_ELEMENT: [IN_http] Element task destroyed
I (4084) AUDIO_ELEMENT: [DEC_mp3] Element task destroyed
I (4090) AUDIO_ELEMENT: [OUT_iis] Element task destroyed
I (4096) AUDIO_ELEMENT: [resample] Element task destroyed
I (4102) AUDIO_PIPELINE: audio_pipeline_unlinked
W (4107) AUDIO_PIPELINE: There are no listener registered
W (4114) ESP_AUDIO_TASK: Destroy the old pipeline
I (4120) ESP_AUDIO_TASK: 2.2 Update all pipeline
I (4124) ESP_AUDIO_TASK: 2.3 Linked new pipeline
I (4130) AUDIO_PIPELINE: link el->rb, el:0x3ffc98d4, tag:IN_http, rb:0x3ffda10c
I (4138) AUDIO_PIPELINE: link el->rb, el:0x3ffc9c40, tag:DEC_mp3, rb:0x3ffd9d70
I (4146) AUDIO_PIPELINE: link el->rb, el:0x3ffcb464, tag:resample, rb:0x3ffd9ea8
I (4154) ESP_AUDIO_TASK: 3. Previous starting...
I (4159) AUDIO_ELEMENT: [IN_http] Element task created
I (4165) AUDIO_ELEMENT: [DEC_mp3] Element task created
I (4170) AUDIO_ELEMENT: [DEC_mp3] AEL_MSG_CMD_RESUME,state:1
I (4171) AUDIO_ELEMENT: [resample] Element task created
I (4177) STAGEFRIGHTMP3_DECODER: MP3 opened
I (4183) AUDIO_ELEMENT: [OUT_iis] Element task created
I (4171) AUDIO_ELEMENT: [IN_http] AEL_MSG_CMD_RESUME,state:1
I (5136) HTTP_STREAM: total_bytes=2994349
I (5144) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:1, Available:5, que:0x3ffc7e30
I (5145) ESP_AUDIO_TASK: Received muisc info, then send MEDIA_CTRL_EVT_PLAY
I (5149) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_PLAY, status:0
I (5155) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:358, MEM Total:85840 Bytes

I (5163) AUDIO_ELEMENT: [OUT_iis] AEL_MSG_CMD_RESUME,state:1
I (5170) I2S_STREAM: AUDIO_STREAM_WRITER
I (5195) AUDIO_PIPELINE: Pipeline started
Second play: [0x81100]
I (5196) AUDIO_ELEMENT: [resample] AEL_MSG_CMD_RESUME,state:1
I (5198) RSP_FILTER: sample rate of source data : 44100, channel of source data : 2, sample rate of destination data : 48000, channel of destination data : 2
I (5212) ESP_AUDIO_TASK: ESP_AUDIO status is AEL_STATUS_STATE_RUNNING
I (5219) HTTP_MP3_EXAMPLE: ESP_AUDIO_CALLBACK_FUNC, st: 1, err: 0, src: 0
```

jason-mao, 2020-02-12:
@xiruilin Thank you for your report.

jason-mao, 2020-02-21:
@xiruilin Bug fixed with `esp_audio` v1.5.3  on commit 37fd3281ea490af055e18474304537ac834cc2d9

closed 2020-02-21 by jason-mao

xiruilin, 2020-02-24:
Thanks for all you hard work.