espressif/esp-adf #334: ESP32 Client registration problem - Qualify in Asterisk

Patsjemoe, 2020-01-28:
Hi,
When using a softphone on my PC, registration to Asterisk (on my laptop) registration works fine.
(also calls work).
When using ESP32, it registers but says Unreachable when I have put Qualify=yes in Asterisk.
When I change  qualify to no, reload asterisk, exit, then change again qualify=yes and reload asterisk, then my ESP32 shows OK/ms, and can call android/linphone. (only voice if I call from the phone to esp32, not from esp32 to linphone...but that is the next problem to solve). 
Could this be a bug in ESP32-VOIP SW ? Or how can I get around this problem ?
Regards
Ludo

Patsjemoe, 2020-01-30:
Hi,
It seems this may be a problem with the "OPTIONS" message that Asterisk is sending out.
Does the VOIP SW on ESP32 not respond  to this ? (it would be good to have the source, so we can verify what the ESP32 side is doing, now we can only guess what is or what is not implemented)
regards

ahhfzhang, 2020-04-09:
Hi @Patsjemoe ,

Yes, seems the new Asterisk will send this, We've solved this problem(https://github.com/espressif/esp-adf/commit/20b0827), hope you can try it.

thanks.

Patsjemoe, 2020-04-09:
> 20b0827 <https://github.com/espressif/esp-adf/commit/20b0827>), hope you

Hi, thanks.
Been out of it for some time, but will try in the near future.
You refer to asterisk.... Do I need to install a new version, or do I need
to update some esp files? This is not clear in your message.
Or do I need to instasl esp-adf 2. 0?
Regards
Ludo

RogerLaw, 2020-04-19:
I tried Asterisk 13.1.0~dfsg-1.1ubuntu4.1.
No problem in the register.
From remote to ESP32, most of the time is OK.
from ESP32 to remote, it seems the state is not going into SIP_STATE_CALLING, then unable to create session.