espressif/esp-adf #335: examples/speech_recognition/asr 关键字唤醒后，无响应？

xuhongv, 2020-01-29:
- demo：examples/speech_recognition/asr/
- des：我的开发板是 lyar-v4.3，对它说了 “Hi 乐鑫”，之后串口打印有输出，但之后说 关键字无法唤醒，也没提示？

```
I (0) cpu_start: Starting scheduler on APP CPU.
I (122) gpio: GPIO[22]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (131) example_asr_keywords: Initialize SR handle
Quantized wakeNet4: wakeNet4_v1_hilexin_6_0.90_0.85, mode: 0 (Nov 20 2019 14:34:44)
I (144) example_asr_keywords: keywords: hilexin (index = 1)
Fail: index is out of range, the min index is 1, the max index is 1I (150) asr_event: keywords_num = 1, threshold = 0.000000,
sample_rate = 16000, chunksize = 480, sizeof_uint16 = 2
I (167) asr_event: [ 1 ] Start codec chip
I (191) asr_event: [ 2.0 ] Create audio pipeline for recording
I (191) asr_event: [ 2.1 ] Create i2s stream to read audio data from codec chip
I (197) asr_event: [ 2.2 ] Create filter to resample audio data
I (201) asr_event: [ 2.3 ] Create raw to receive data
I (207) asr_event: [ 3 ] Register all elements to audio pipeline
I (214) asr_event: [ 4 ] Link elements together [codec_chip]-->i2s_stream-->filter-->raw-->[SR]
I (223) asr_event: [ 5 ] Start audio_pipeline
I (2807) example_asr_keywords: Wake up
I (9353) example_asr_keywords: Wake up
I (46695) example_asr_keywords: Wake up
I (60866) example_asr_keywords: Wake up
I (69347) example_asr_keywords: Wake up
I (110948) example_asr_keywords: Wake up
I (126097) example_asr_keywords: Wake up
I (201366) example_asr_keywords: Wake up

```

jason-mao, 2020-02-12:
@xuhongv 这个example不支持关键词的识别，只能识别唤醒词。

xuhongv, 2020-02-12:
我看 readme 上，是识别开灯关灯的？那你们应该更新下 readme文档

closed 2020-02-25 by xuhongv