espressif/esp-adf #336: esp-adf-v2.0-beta2 , voip_app.elf not found

Patsjemoe, 2020-01-30:
Hi,
I installed the newest esp-adf-v2.0-beta2 version to find out if my registration problem would be solved.

After menuconfig, compilation is OK, but get following error when starting up : 

 c:/users/me/.espressif/tools/xtensa-esp32-elf/1.22.0-80-g6c4433a5-5.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-addr2line -pfiaC -e C:/msys32/home/me/esp-adf-v2.0-beta2/exa
mples/advanced_examples/voip/build/voip_app.elf 0x40078000: [Error 2] Het systeem kan het opgegeven bestand niet vinden (the systems cannot find the given file).
Note : I checked in the given directory, and the file "exists"

Any ideas on this ?
thanks in advance
best regards
Ludo Kustermans

jason-mao, 2020-02-12:
@Patsjemoe Could you test with latest master?

Patsjemoe, 2020-02-12:
hi,
where can I download this ?
what is the difference of master vs beta vs standaard release ?
regards

jason-mao, 2020-02-21:
@Patsjemoe You can follow the ADF [quick start](https://github.com/espressif/esp-adf#quick-start)clone the latest code and test. We are fix found bugs on master.

Patsjemoe, 2020-04-24:
Installed v2. 0 final release, problem gone. Still issues with calling android phone, but that is another issue. This one can be closed. 
Thank you

closed 2020-04-24 by Patsjemoe