espressif/esp-adf #337: example pipeline_raw_http fail

guolxtank, 2020-01-30:
when I try to run example pipeline_raw_http,it always show as below, I am trying my best to solve the problem but failed.Please give me a little help! Thank you!
my board is **ESP32-LyraT-Mini_V1.2** with 8M PSRAM and 8M flash.
```c
rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6892
load:0x40078000,len:12072
load:0x40080400,len:6756
entry 0x4008077c
I (71) boot: Chip Revision: 1
I (72) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot: ESP-IDF v3.3.1-dirty 2nd stage bootloader
I (39) boot: compile time 07:30:30
I (39) boot: Enabling RNG early entropy source...
I (44) boot: SPI Speed      : 40MHz
I (48) boot: SPI Mode       : DIO
I (52) boot: SPI Flash Size : 8MB
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00180000
I (90) boot: End of partition table
I (94) boot_comm: chip revision: 1, min. application chip revision: 0
I (101) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x2dfb0 (188336) map
I (176) esp_image: segment 1: paddr=0x0003dfd8 vaddr=0x3ffb0000 size=0x02038 (  8248) load
I (180) esp_image: segment 2: paddr=0x00040018 vaddr=0x400d0018 size=0xa3a5c (670300) map
0x400d0018: _flash_cache_start at ??:?

I (418) esp_image: segment 3: paddr=0x000e3a7c vaddr=0x3ffb2038 size=0x01268 (  4712) load
I (420) esp_image: segment 4: paddr=0x000e4cec vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/osboxes/esp/esp-adf/esp-idf/components/freertos/xtensa_vectors.S:1779

I (425) esp_image: segment 5: paddr=0x000e50f4 vaddr=0x40080400 size=0x169cc ( 92620) load
I (485) boot: Loaded app from partition at offset 0x10000
I (485) boot: Disabling RNG early entropy source...
I (486) psram: This chip is ESP32-D0WD
I (492) spiram: Found 64MBit SPI RAM device
I (495) spiram: SPI RAM mode: flash 40m sram 40m
I (500) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (507) cpu_start: Pro cpu up.
I (511) cpu_start: Application information:
I (516) cpu_start: Project name:     record_raw_http
I (522) cpu_start: App version:      1
I (526) cpu_start: Compile time:     Jan 30 2020 07:30:27
I (532) cpu_start: ELF file SHA256:  f17f9316d876abf8...
I (538) cpu_start: ESP-IDF:          v3.3.1-dirty
I (544) cpu_start: Starting app cpu, entry point is 0x400814b4
0x400814b4: call_start_cpu1 at /home/osboxes/esp/esp-adf/esp-idf/components/esp32/cpu_start.c:269

I (0) cpu_start: App cpu up.
I (1435) spiram: SPI SRAM memory test OK
I (1435) heap_init: Initializing. RAM available for dynamic allocation:
I (1435) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1441) heap_init: At 3FFB94C8 len 00026B38 (154 KiB): DRAM
I (1448) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1454) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1461) heap_init: At 40096DCC len 00009234 (36 KiB): IRAM
I (1467) cpu_start: Pro cpu start user code
I (1472) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (151) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (152) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (182) REC_RAW_HTTP: [ 1 ] Initialize Button Peripheral & Connect to wifi network
I (312) phy: phy_version: 4102, 2fa7a43, Jul 15 2019, 13:06:06, 0, 0
I (4682) REC_RAW_HTTP: [ 2 ] Start codec chip
E (4702) DRV8311: Es8311 Write Reg error
[... 174 lines cut ...]
I (41612) REC_RAW_HTTP: [ * ] Resuming pipeline
I (41632) REC_RAW_HTTP: [ + ] HTTP client HTTP_STREAM_PRE_REQUEST, lenght=0
I (41682) REC_RAW_HTTP: [ * ] Stop pipeline
W (41712) AUDIO_PIPELINE: audio_pipeline state is not running, 1
W (41712) AUDIO_PIPELINE: audio_pipeline state is not running, 1
I (43722) REC_RAW_HTTP: [ * ] Resuming pipeline
I (43742) REC_RAW_HTTP: [ * ] Stop pipeline
W (43762) AUDIO_PIPELINE: audio_pipeline state is not running, 1
W (43762) AUDIO_PIPELINE: audio_pipeline state is not running, 1
I (45762) REC_RAW_HTTP: [ * ] Resuming pipeline
I (45762) REC_RAW_HTTP: [ * ] Resuming pipeline
W (45762) AUDIO_PIPELINE: Pipeline already started, state:2
I (45762) REC_RAW_HTTP: [ * ] Stop pipeline
W (45812) AUDIO_PIPELINE: audio_pipeline state is not running, 1
W (45812) AUDIO_PIPELINE: audio_pipeline state is not running, 1
E (45812) AUDIO_ELEMENT: [http] Element destroy CMD failed
I (45822) REC_RAW_HTTP: [ * ] Stop pipeline
W (45822) AUDIO_PIPELINE: audio_pipeline state is not running, 1
W (45832) AUDIO_PIPELINE: audio_pipeline state is not running, 1
W (45842) AUDIO_PIPELINE: audio_pipeline state is not running, 1
W (45842) AUDIO_PIPELINE: audio_pipeline state is not running, 1
W (45852) AUDIO_ELEMENT: [i2s] Element has not create when AUDIO_ELEMENT_TERMINATE
E (45862) AUDIO_ELEMENT: [http] Element destroy CMD failed
E (45872) REC_RAW_HTTP: [ * ] Event interface error : -1
```

app-z, 2020-02-04:
Recently my project based on GoogleTranslateDevice stop working. Look like those issues connected

```
W (8013) GOOGLE_TRANSLATION_EXAMPLE: Start speaking now
I (8943) GOOGLE_TRANSLATION_EXAMPLE: [ * ] Event received: src_type:1048577, source:0x3ffbb2ac cmd:3, data:0x4, data_len:0
I (26133) GOOGLE_TRANSLATION_EXAMPLE: [ * ] Event received: src_type:1048577, source:0x3ffbb2ac cmd:4, data:0x4, data_len:0
I (26133) GOOGLE_TRANSLATION_EXAMPLE: [ * ] Stop pipeline
W (26143) AUDIO_ELEMENT: IN-[sr_http] AEL_IO_ABORT
I (26143) GOOGLE_SR: [ + ] HTTP client HTTP_STREAM_POST_REQUEST, write end chunked marker
W (26163) AUDIO_ELEMENT: OUT-[sr_i2s] AEL_IO_ABORT
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4016f067  PS      : 0x00060330  A0      : 0x800ed924  A1      : 0x3ffd4710  
0x4016f067: audio_element_getdata at /home/appz/esp/esp-adf/components/audio_pipeline/audio_element.c:483

A2      : 0x00000000  A3      : 0x3ffb3d2c  A4      : 0x3f41ea30  A5      : 0x0000001f  
A6      : 0x00000001  A7      : 0x00000003  A8      : 0x80082bc9  A9      : 0x3ffd46c0  
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x3ffafc78  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000003  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x000000b4  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff  

ELF file SHA256: 5d190d3fde264e0955abdfea4a9c1a6aa518de86d8e2046959a6999190c3c36d

Backtrace: 0x4016f067:0x3ffd4710 0x400ed921:0x3ffd4730 0x400edd35:0x3ffd4770 0x400eb9c2:0x3ffd47d0 0x400ebf07:0x3ffd4800 0x400ead8f:0x3ffd4830 0x400ec8d2:0x3ffd4870 0x4008e065:0x3ffd48a0
0x4016f067: audio_element_getdata at /home/appz/esp/esp-adf/components/audio_pipeline/audio_element.c:483

0x400ed921: dispatch_hook at /home/appz/esp/esp-adf/components/audio_stream/http_stream.c:135

0x400edd35: _http_close at /home/appz/esp/esp-adf/components/audio_stream/http_stream.c:628

0x400eb9c2: audio_element_process_deinit at /home/appz/esp/esp-adf/components/audio_pipeline/audio_element.c:472

0x400ebf07: audio_element_on_cmd at /home/appz/esp/esp-adf/components/audio_pipeline/audio_element.c:472

0x400ead8f: audio_event_iface_waiting_cmd_msg at /home/appz/esp/esp-adf/components/audio_pipeline/audio_event_iface.c:212

0x400ec8d2: audio_element_task at /home/appz/esp/esp-adf/components/audio_pipeline/audio_element.c:472

0x4008e065: vPortTaskWrapper at /home/appz/esp/esp-adf/esp-idf/components/freertos/port.c:403

Rebooting...

```

app-z, 2020-02-10:
pipeline_raw_http doesn't work via the I2S driver when using the “built-in DAC mode”. its what I noticed.

However, pipeline_passthru working with built-in DAC

jason-mao, 2020-02-12:
@guolxtank How about your problem now? `E (4702) DRV8311: Es8311 Write Reg error` The I2C communicate with codec ES8311 fail.

jason-mao, 2020-02-12:
@app-z How's it going your issue?

app-z, 2020-02-12:
I use Analog preamplifier connected to ADC. Set 
[image]

_I use initialization_

```
static const int i2s_num = 0; // i2s port number

static const i2s_config_t i2s_config = {
    .mode = I2S_MODE_MASTER | I2S_MODE_TX,
    .sample_rate = 32000,
    .bits_per_sample = 16,
    .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
    .communication_format = I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_MSB, //  I2S_COMM_FORMAT_I2S_LSB
    .intr_alloc_flags = 0, // default interrupt priority
    .dma_buf_count = 4,
    .dma_buf_len = 256,
};

i2s_driver_install(I2S_NUM, &i2s_config, 0, NULL);

// Somewhere here is stream config

........................................................................................

    adc1_config_width(ADC_WIDTH_BIT_12);
    adc1_config_channel_atten(ADC1_CHANNEL_0,ADC_ATTEN_DB_0);

```
Before my code is working. Now NOT

I try even build ESP-ADF Pre-release v2.0-beta2

Still dosent work

app-z, 2020-02-26:
I found working commit!
[Mic + Amplifier]-->[BUILT IN ADC]-->[HTTP stream]-->[Google Speech Recognition]-->[HTTP stream]-->[Google TTS]->[BUILT IN DAC]-->[Amplifier]
Its working well again
My board is here [https://github.com/app-z/esp32-oled-128x64](https://github.com/app-z/esp32-oled-128x64) still in develop...
Look like there was many significant changes in i2s_stream after September

HimbeersaftLP, 2020-07-31:
@app-z I'm having the same problem, what did you do to fix it?

jason-mao, 2022-02-08:
This topic has become inactive so I'm going to close the issue. Please reopen this if you have any questions or need any further assistance.

closed 2022-02-08 by jason-mao

patrickkeenan, 2023-10-09:
This looks like the same issue now in the example for pipeline_raw_http, any update?