espressif/esp-adf #339: ringing an mp3 in voip example

ottaviane, 2020-01-31:
Hi,
 i'm trying to run an mp3 in _sip_event_handler when SIP_EVENT_RINGING: is on.
I made it:
`/* VoIP Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event_loop.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "audio_element.h"
#include "audio_pipeline.h"
#include "audio_event_iface.h"
#include "audio_common.h"
#include "i2s_stream.h"
#include "esp_peripherals.h"
#include "periph_wifi.h"
#include "board.h"
#include "input_key_service.h"

#include "audio_mem.h"
#include "raw_stream.h"
#include "filter_resample.h"
#include "esp_sip.h"
#include "g711.h"
//**************************riproduttore MP3

#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

#include "esp_log.h"
#include "audio_element.h"
#include "audio_pipeline.h"
#include "audio_event_iface.h"
#include "audio_mem.h"
#include "audio_common.h"
#include "i2s_stream.h"
#include "mp3_decoder.h"
#include "filter_resample.h"
#include "board.h"

    #include "audio_mem.h"
    #include "mp3_decoder.h"
    #include "filter_resample.h"
    extern const uint8_t adf_music_mp3_start[] asm("_binary_adf_music_mp3_start");
    extern const uint8_t adf_music_mp3_end[]   asm("_binary_adf_music_mp3_end");    
    audio_pipeline_handle_t pipeline;
    audio_element_handle_t i2s_stream_writer, mp3_decoder;
//*******************************************

static const char *TAG = "VOIP_EXAMPLE";

#define RTP_HEADER_LEN 12
#define AUDIO_FRAME_SIZE (160)

#define I2S_SAMPLE_RATE     48000
#define I2S_CHANNEL         2
#define I2S_BITS            16
[... 422 lines cut ...]
        .acodec_type = SIP_ACODEC_G711A,
#else
        .acodec_type = SIP_ACODEC_G711U,
#endif
    };
    sip = esp_sip_init(&sip_cfg);
    esp_sip_start(sip);

    ESP_LOGI(TAG, "[ 5 ] Create decoder and encoder pipelines");
    g711enc_pipeline_open();
    g711dec_pipeline_open();
}

`

but when I do make all I have this error:
`
Generating esp32.project.ld
LD build/voip_app.elf
/home/ottaviane/esp/citoVoipSlave/build/main/libmain.a(voip_app.o):(.literal.mp3_music_read_cb+0x0): undefined reference to `_binary_adf_music_mp3_end'
/home/ottaviane/esp/citoVoipSlave/build/main/libmain.a(voip_app.o):(.literal.mp3_music_read_cb+0x4): undefined reference to `_binary_adf_music_mp3_start'
collect2: error: ld returned 1 exit status
make: *** [/home/ottaviane/esp/esp-idf/make/project.mk:483: /home/ottaviane/esp/citoVoipSlave/build/voip_app.elf] Error 1`
what appen?
bye and thanks

ahhfzhang, 2020-02-03:
Hi @ottaviane 
Did you add `COMPONENT_EMBED_TXTFILES := adf_music.mp3` to your example component.mk ?

ottaviane, 2020-02-05:
thankyou , now it works but audio is very distorted.
I'd like to create a VOIP intercom, but I've problems working with two pipeline:
one is for voip and other is for my mp3(ring) and it seems they conflict!
New code is it:
```

#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event_loop.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "audio_element.h"
#include "audio_pipeline.h"
#include "audio_event_iface.h"
#include "audio_common.h"
#include "i2s_stream.h"
#include "esp_peripherals.h"
#include "periph_wifi.h"
#include "board.h"
#include "input_key_service.h"

#include "mp3_decoder.h"  //inserito il 5/2/2020*    
#include "periph_touch.h"//                     *
#include "periph_adc_button.h" //               *
#include "periph_button.h" //********************

#include "audio_mem.h"
#include "raw_stream.h"
#include "filter_resample.h"
#include "esp_sip.h"
#include "g711.h"

static const char *TAG = "CITOVOIP_SLAVE";

#define RTP_HEADER_LEN 12
#define AUDIO_FRAME_SIZE (160)

#define I2S_SAMPLE_RATE     48000
#define I2S_CHANNEL         2
#define I2S_BITS            16

#define G711_SAMPLE_RATE    8000
#define G711_CHANNEL        1

static sip_handle_t sip;
static audio_element_handle_t raw_read;
static audio_element_handle_t raw_write;

bool semaforo;

//*******************************************************************************************************************************************************
//inserito il 5/2/2020                                                                                                                                  *
extern const uint8_t adf_music_mp3_start[] asm("_binary_adf_music_mp3_start");//                                                                        *
extern const uint8_t adf_music_mp3_end[]   asm("_binary_adf_music_mp3_end");//                                                                          *
static int adf_music_mp3_pos;//                                                                                                                         *
audio_element_handle_t  mp3_decoder; 
audio_element_handle_t my_i2s_stream_writer;//                                                                                                          *
audio_pipeline_handle_t mypipeline;   //                                                                                                                *

int mp3_music_read_cb(audio_element_handle_t el, char *buf, int len, TickType_t wait_time, void *ctx)
{
    int read_size = adf_music_mp3_end - adf_music_mp3_start - adf_music_mp3_pos;
    if (read_size == 0) {
        return AEL_IO_DONE;
    } else if (len < read_size) {
        read_size = len;
[... 353 lines cut ...]

    ESP_LOGI(TAG, "[ 4 ] Create SIP Service");
    sip_config_t sip_cfg = {
        .uri = CONFIG_SIP_URI,
        .event_handler = _sip_event_handler,
#ifdef CONFIG_SIP_CODEC_G711A
        .acodec_type = SIP_ACODEC_G711A,
#else
        .acodec_type = SIP_ACODEC_G711U,
#endif
    };
    sip = esp_sip_init(&sip_cfg);
    esp_sip_start(sip);

    ESP_LOGI(TAG, "[ 5 ] Create decoder and encoder pipelines");
    semaforo=false;
    mp3_pipeline_open();
    g711enc_pipeline_open();
    g711dec_pipeline_open();
    
}

```
what is wrong?
bye