espressif/esp-adf #341: Add esp32-a1s ac101 driver

xuhongv, 2020-02-01:
Hello , I am a Software Engineer at the Ai-Thinker , We will publish an ESP32 audo kit.The codec of the board is AC101.We plan to add our codes of AC101 to ESP-ADF and be compatible with your program.I test our audio kit and ESP32 LYRAT Board.The program works well. We will help you to develop ESP-ADF.

嗨，我是安信可的一名嵌入式工程师。我们已发布ESP32 audio音频开发板套件。开发板的编解码器为AC101。我们计划将AC101的编码添加到ESP-ADF中，并与您的程序兼容。我测试音频套件程序运行良好，我们将帮助您开发和完善ESP-ADF框架。

janchri, 2020-02-03:
Hi, nice to see that there will be official support from ai thinker side. So far I needed to implement the support myself. I think in your current version there is an update with all the details to the a1s board missing in file /components/audio_hal/CMakeLists.txt

xuhongv, 2020-02-03:
@janchri 
thanks，i just add ac101 driver to ```/components/audio_hal/CMakeLists.txt```.

- https://github.com/xuhongv/esp-adf/commit/10df9536431a3b57d920714a770b92a9ffc608d9

pilnikov, 2020-02-10:
look for my variant
https://github.com/pilnikov/adf/tree/master/esp-adf/components/audio_hal/driver/ac101

jason-mao, 2020-06-19:
@xuhongv 现在ADF支持了custom board，建议`esp32-a1s`的驱动和 board config 通过这个实现。

tcpipchip, 2020-06-30:
working!!!!!!!!!!

closed 2020-10-15 by jason-mao

herostrat, 2020-12-31:
@jason-mao: The user @othiman did a very good job with porting the AI Tinker board to the official esp-adf.

Is it worth the hustle to prepare a PR for it? Or will the board definition and codec not be merged into esp-adf?

tcpipchip, 2020-12-31:
Yes! Good question!!!!