espressif/esp-adf #343: some errors on compile player example

pilnikov, 2020-02-10:
1.
[image]
because in esp-adf/components/bluetooth_service/CmakeLists.txt component link missing 
 in original 
set(COMPONENT_REQUIRES bt audio_sal audio_pipeline esp_peripherals)
should be
set(COMPONENT_REQUIRES bt audio_sal audio_pipeline esp_peripherals **audio_stream**)

2.
[image]
because ????? 
esp-adf/components/esp-adf-libs/esp_codec/lib/libesp-mp3.a is precompile and not open

eclipsevl, 2020-02-11:
Tonight I've got same issue with pow, log10 functions, after workaround for these two linker failing on dct_16_opt. Seems like some library is not linked, but I can't find which one.

pilnikov, 2020-02-11:
https://github.com/pilnikov/adf/tree/master/esp-adf/components/esp-adf-libs/esp_codec/lib
these working - checked

eclipsevl, 2020-02-11:
> https://github.com/pilnikov/adf/tree/master/esp-adf/components/esp-adf-libs/esp_codec/lib
> these working - checked

Thanks! Indeed, these libs are compilable.
However I still missing log10 and pow functions - had to define it myself, and after that II'm getting following errors after pipeline start:

E (4117) STAGEFRIGHTMP3_DECODER: Encountered error reading when MP3 init
E (4127) AUDIO_ELEMENT: [mp3] AEL_STATUS_ERROR_OPEN
W (4137) AUDIO_ELEMENT: IN-[i2s] AEL_IO_ABORT

Idk if it is connected somehow to this issue.

jason-mao, 2020-02-12:
@pilnikov Issue 1 is incoming fix. Issue 2,  are you used 5.2.0 toolchain works fine?

jason-mao, 2020-02-12:
@eclipsevl 
> However I still missing log10 and pow functions - had to define it myself, and after that II'm getting following errors after pipeline start:
>
> E (4117) STAGEFRIGHTMP3_DECODER: Encountered error [...]

Could you provide your detail logs?

pilnikov, 2020-02-12:
> @pilnikov Issue 1 is incoming fix. Issue 2, are you used 5.2.0 toolchain works fine?

Yes, everything works except the above. I changed some things for this.

eclipsevl, 2020-02-12:
> @eclipsevl
>
>
> Could you provide your detail logs?

Sure.

This is compile log without pow, log10 defined in main.c:
https://pastebin.com/Q5djNzGh

My workaround:
https://pastebin.com/BwT7daMt

App log:
https://pastebin.com/mwEf0L9d

And audio init code. (Board initialization is ommited)
https://pastebin.com/05etDKEr

pilnikov, 2020-02-17:
[image]
decision:
added line
#include "driver/gpio.h"
in file esp_peripherals.c

jason-mao, 2020-02-17:
@pilnikov 

> [image]
> decision:
> added line
> #include "driver/gpio.h"
> in file esp_peripherals.c

ADF support IDF v3.3 release only. Did you used that?

pilnikov, 2020-02-17:
But does it really matter which version of IDF to use? Or in the v3.3 links organized differently?  Yes, I use version not 3.3 but 4.1. But someone promised up to version 4  in December.

pilnikov, 2020-02-17:
Well, if #include "periph_gpio_isr.h"
  it is not right. Then maybe #include "periph_gpio_isr.h" instead? Because
     // TODO: Should we uninstall gpio isr service ??
     // TODO: Because gpio need for sdcard and gpio, then install isr here

jason-mao, 2020-02-21:
@pilnikov 
> But does it really matter which version of IDF to use? Or in the v3.3 links organized differently? Yes, I use version not 3.3 but 4.1. But someone promised up to version 4 in December.

Sorry for late. It can be support IDF v4.x start 2a95e1f2c603d07f98735518295243cbdb2958ed

pilnikov, 2020-02-21:
tnx for support

https://github.com/espressif/esp-adf/commit/9c06d63f38cddbef420af3aae78ba15709729b0c#commitcomment-37420385

&& with libesp-mp3.a questions remained

jason-mao, 2020-02-21:
@pilnikov How to reproduce your mp3 issue?

> tnx for support
> 
> [9c06d63#commitcomment-37420385](https://github.com/espressif/esp-adf/commit/9c06d63f38cddbef420af3aae78ba15709729b0c#commitcomment-37420385)
> 
> && with libesp-mp3.a questions remained
Yes, we will fix that later.

pilnikov, 2020-02-21:
i check it now - 
https://github.com/espressif/esp-adf-libs/commit/c7c8398e35cbe9949bd427ab42830ff6b6cdb3c1#commitcomment-37420679

closed 2020-02-21 by pilnikov