espressif/esp-adf #350: component_wrapper.mk:292: periph_wifi.o] Error 1

HoneyPot2020, 2020-02-20:
Board: ESP32 LyraT Mini V1.2
Toolchain version: esp-2019r2
Compiler version: 8.2.0
Python requirements from /esp/esp-idf/requirements.txt are satisfied.

Tried to run the getting started mp3 example. Using Mingw32 ran the 
Make Flash Monitor
Error Message:
make[1]: *** [/esp/esp-idf/make/component_wrapper.mk:292: periph_wifi.o] Error 1
make: *** [/esp/esp-idf/make/project.mk:604: component-esp_peripherals-build] Error 2

Please Note:
The Hello World, Blink examples are running correctly.

jason-mao, 2020-02-20:
@HoneyPot2020 Please note that ADF only support IDF v3.3.1.

jason-mao, 2020-02-21:
@HoneyPot2020 And now after 37fd3281ea490af055e18474304537ac834cc2d9 ADF could be support IDF v4.0 and toolchain v8.2.0. Thank you.

closed 2020-02-21 by jason-mao