espressif/esp-adf #351: In "esp_audio  v1.5.3", filter_resample.c cause a error

xiruilin, 2020-02-24:
Hi,
In the new update, ``esp_adf_libs`` commit [4d90bdf](https://github.com/espressif/esp-adf-libs/commit/4d90bdff983a7504142ea96d8fde99e532c51142), when last sample rate is ``48000hz``, 
and now need resample from ``44100hz`` to ``48000hz``, cause a error, the log below:
```
I (15532) DCS_AUDIO: Playing audio, offset: 2925462 url: http://duer-music.cdn.bcebos.com/audio/128/de77dfaffbaea4b6b9a3f4d4b22f0d54.mp3?authorization=bce-auth-v1/d3af5aaafadb4603b04c30e2acb194a6/2020-02-24T08:35:31Z/86400//c840a2a2cdac0f2a943c25e5207b2d7e157550cd05dfbbb55f8621bfb4470ecb
I (15549) DISPATCHER: EXE IN, cmd type:2, index:4001, data:0x3f8101d4, len:12
I (15555) DCS_WRAPPER: duer_dcs_action_audio_play, 0x3f8101d4, dˁ?��,, 12
I (15563) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5, que:0x3ffe0f5c
I (15571) ESP_AUDIO_CTRL: Func:_ctrl_play, Line:542, MEM Total:3920416 Bytes, Inter:62504 Bytes, Dram:59976 Bytes

I (15582) ESP_AUDIO_TASK: It's a decoder
I (15586) ESP_AUDIO_TASK: 1.CUR IN:[IN_http],CODEC:[DEC_mp3],RESAMPLE:[48000],OUT:[OUT_iis]
I (15596) ESP_AUDIO_TASK: 2.Handles,IN:0x3f80bf44,CODEC:0x3f80ca80,FILTER:0x3f812980,OUT:0x3f80d014
W (15605) AUDIO_PIPELINE: audio_pipeline state is not running, 1
W (15612) AUDIO_PIPELINE: audio_pipeline state is not running, 1
W (15618) AUDIO_PIPELINE: audio_pipeline state is not running, 1
W (15625) AUDIO_PIPELINE: audio_pipeline state is not running, 1
I (15637) AUDIO_ELEMENT: [IN_flash] Element task destroyed
I (15644) AUDIO_ELEMENT: [DEC_mp3] Element task destroyed
I (15656) AUDIO_ELEMENT: [OUT_iis] Element task destroyed
I (15663) AUDIO_ELEMENT: [resample] Element task destroyed
I (15669) AUDIO_PIPELINE: audio_pipeline_unlinked
W (15674) AUDIO_PIPELINE: There are no listener registered
W (15680) ESP_AUDIO_TASK: Destroy the old pipeline
I (15685) ESP_AUDIO_TASK: 2.2 Update all pipeline
I (15692) ESP_AUDIO_TASK: 2.3 Linked new pipeline
I (15697) AUDIO_PIPELINE: link el->rb, el:0x3f80bf44, tag:IN_http, rb:0x3f81d584
I (15705) AUDIO_PIPELINE: link el->rb, el:0x3f80ca80, tag:DEC_mp3, rb:0x3f81da2c
I (15713) AUDIO_PIPELINE: link el->rb, el:0x3f812980, tag:resample, rb:0x3f81d674
I (15721) ESP_AUDIO_TASK: 3. Previous starting...
I (15734) AUDIO_ELEMENT: [IN_http] Element task created
I (15735) AUDIO_ELEMENT: [DEC_mp3] Element task created
I (15737) AUDIO_ELEMENT: [DEC_mp3] AEL_MSG_CMD_RESUME,state:1
I (15738) AUDIO_ELEMENT: [resample] Element task created
I (15744) STAGEFRIGHTMP3_DECODER: MP3 opened
I (15738) AUDIO_ELEMENT: [IN_http] AEL_MSG_CMD_RESUME,state:1
I (15755) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:1, Available:5, que:0x3ffe0f5c
I (15769) ESP_AUDIO_TASK: Received muisc info, then send MEDIA_CTRL_EVT_PLAY
I (15777) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_PLAY, status:0
I (15783) AUDIO_ELEMENT: [OUT_iis] Element task created
I (15788) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:358, MEM Total:3919580 Bytes, Inter:62408 Bytes, Dram:59880 Bytes

I (15800) AUDIO_ELEMENT: [OUT_iis] AEL_MSG_CMD_RESUME,state:1
I (15806) I2S_STREAM: AUDIO_STREAM_WRITER
I (15831) AUDIO_PIPELINE: Pipeline started
I (15834) AUDIO_ELEMENT: [resample] AEL_MSG_CMD_RESUME,state:1
I (15834) RSP_FILTER: sample rate of source data : 48000, channel of source data : 2, sample rate of destination data : 48000, channel of destination data : 2
I (15851) DISPATCHER: EXE OUT,result type:2, index:4001, ret:0, data:0x0, len:0
I (15854) DCS_AUDIO: Audio ret:0
I (15907) HTTP_STREAM: total_bytes=1950615
I (15909) AUDIO_ELEMENT: [IN_http] AEL_MSG_CMD_RESUME,state:2
I (15917) ESP_AUDIO_TASK: ESP_AUDIO status is AEL_STATUS_STATE_RUNNING
I (15917) DUEROS_CPP: ESP_AUDIO_CALLBACK_FUNC, st: 1, err: 0, src: 801
I (15920) AUDIO_ELEMENT: [DEC_mp3] AEL_MSG_CMD_RESUME,state:2
I (15922) DUEROS_CPP: duer_dcs_audio_on_running
I (15936) DISPATCHER: EXE IN, cmd type:2, index:300d, data:0x0, len:0
I (15943) DIS_ACTION: display_action_music_on
I (15950) DISPATCHER: EXE OUT,result type:2, index:300d, ret:0, data:0x0, len:0
I (15956) RSP_FILTER: reset sample rate of source data : 44100, reset channel of source data : 2
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4009bdec  PS      : 0x00060e30  A0      : 0x8014fa66  A1      : 0x3ffed7d0
0x4009bdec: memcpy at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:176

A2      : 0x00000000  A3      : 0x3ffebe90  A4      : 0x00000030  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x000000ff
A10     : 0x3ffef1f0  A11     : 0x3ffef1f0  A12     : 0x3ffefa00  A13     : 0x00000000
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x00000004  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x4009bde0  LEND    : 0x4009be0e  LCOUNT  : 0x00000002
0x4009bde0: memcpy at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:168

0x4009be0e: memcpy at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:201

ELF file SHA256: 049d1d4fec160489b456befb37f24671b797470db2da2929e0fc3d8ecdf40a2f

Backtrace: 0x4009bde9:0x3ffed7d0 0x4014fa63:0x3ffed7e0 0x4014af7e:0x3ffed850 0x40147f91:0x3ffed870 0x40149146:0x3ffed890
0x4009bde9: memcpy at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:173

0x4014fa63: li_resampling_dec at /Users/maojianxin/duros/esp-adf-libs-source/esp_processing/esp-resample/src_li/li_resample.c:407
 (inlined by) li_resample_process at /Users/maojianxin/duros/esp-adf-libs-source/esp_processing/esp-resample/src_li/li_resample.c:605

0x4014af7e: esp_resample_run at /Users/maojianxin/duros/esp-adf-libs-source/esp_processing/esp-resample/src/esp_resample.c:46

0x40147f91: rsp_filter_process at E:/esp-adf/components/esp-adf-libs/esp_codec/filter_resample.c:148

0x40149146: audio_element_process_running at E:/esp-adf/components/audio_pipeline/audio_element.c:287
 (inlined by) audio_element_task at E:/esp-adf/components/audio_pipeline/audio_element.c:438
```

jason-mao, 2020-02-24:
@xiruilin I have not reproduced you issue. Could you provide more info? BTW, did you clean you environment?

xiruilin, 2020-02-26:
@jason-mao 
code snippet:
```
    esp_audio_sync_play(handle, "flash://tone/65_dueros_microphone_on.mp3", 0);
    esp_audio_play(handle, AUDIO_CODEC_TYPE_DECODER, "https://dl.espressif.com/dl/audio/gs-16b-2c-44100hz.mp3", 4096);
```
The ``esp_audio_sync_play`` url is a 48000hz mp3 file, you can use other mp3 file with 48000hz.

The log:
```
I (3708) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5, que:0x3ffd14e8
D (3716) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_SETUP,data:0x0,size:0,stack:2152
I (3722) ESP_AUDIO_TASK: It's a decoder
D (3727) ESP_AUDIO_TASK: URI:flash://tone/65_dueros_microphone_on.mp3

I (3734) ESP_AUDIO_TASK: 1.CUR IN:[IN_flash],CODEC:[DEC_mp3],RESAMPLE:[48000],OUT:[OUT_iis]
D (3743) ESP_AUDIO_TASK: 1.1 Found the element
I (3748) ESP_AUDIO_TASK: 2.Handles,IN:0x3f805c98,CODEC:0x3f806000,FILTER:0x3f806334,OUT:0x3f806200
I (3757) ESP_AUDIO_TASK: 2.2 Update all pipeline
I (3762) ESP_AUDIO_TASK: 2.3 Linked new pipeline
I (3768) AUDIO_PIPELINE: link el->rb, el:0x3f805c98, tag:IN_flash, rb:0x3f806448
I (3776) AUDIO_PIPELINE: link el->rb, el:0x3f806000, tag:DEC_mp3, rb:0x3f808c88
I (3784) AUDIO_PIPELINE: link el->rb, el:0x3f806334, tag:resample, rb:0x3f80acc8
I (3792) ESP_AUDIO_TASK: 3. Previous starting...
I (3797) AUDIO_ELEMENT: [IN_flash] Element task created
I (3803) AUDIO_ELEMENT: [DEC_mp3] Element task created
I (3809) AUDIO_ELEMENT: [DEC_mp3] AEL_MSG_CMD_RESUME,state:1
I (3809) AUDIO_ELEMENT: [resample] Element task created
I (3815) STAGEFRIGHTMP3_DECODER: MP3 opened
D (3821) ESP_AUDIO_TASK: Waiting MEDIA_CTRL_EVT_PLAY sent by AEL_MSG_CMD_REPORT_MUSIC_INFO
I (3809) AUDIO_ELEMENT: [IN_flash] AEL_MSG_CMD_RESUME,state:1
I (3841) flashPartition: 133: labe:flash_tone
I (3846) TONE_STREAM: Wanted read flash tone index is 65
I (3852) TONE_STREAM: Tone offset:000ad760, Tone length:14829, total num:77  pos:65

D (3860) ESP_AUDIO_TASK: Recv Element[IN_flash-0x3f805c98] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_RUNNING
I (3809) ESP_AUDIO_CTRL: Func:_ctrl_play, Line:542, MEM Total:4305224 Bytes, Inter:210048 Bytes, Dram:194520 Bytes

D (3877) ESP_AUDIO_TASK: Recv Element[DEC_mp3-0x3f806000] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_RUNNING
D (3893) ESP_AUDIO_TASK: Recv MUSIC INFO, msg[DEC_mp3] cmd:9 arrived,rate:48000,ch:2,bit:16,pos:384,bps:64000,duration:0,size:0,fmt:3
I (3910) RSP_FILTER: reset sample rate of source data : 48000, reset channel of source data : 2
I (3923) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:1, Available:5, que:0x3ffd14e8
I (3931) ESP_AUDIO_TASK: Received muisc info, then send MEDIA_CTRL_EVT_PLAY
I (3938) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_PLAY, status:0
I (3944) AUDIO_ELEMENT: [OUT_iis] Element task created
I (3950) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:358, MEM Total:4264600 Bytes, Inter:206108 Bytes, Dram:190580 Bytes

I (3962) AUDIO_ELEMENT: [OUT_iis] AEL_MSG_CMD_RESUME,state:1
I (3968) I2S_STREAM: AUDIO_STREAM_WRITER
I (3994) AUDIO_PIPELINE: Pipeline started
D (3994) ESP_AUDIO_TASK: Recv Element[OUT_iis-0x3f806200] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_RUNNING
I (3998) AUDIO_ELEMENT: [resample] AEL_MSG_CMD_RESUME,state:1
I (4005) RSP_FILTER: sample rate of source data : 48000, channel of source data : 2, sample rate of destination data : 48000, channel of destination data : 2
D (4019) ESP_AUDIO_TASK: Recv Element[resample-0x3f806334] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_RUNNING
I (4030) ESP_AUDIO_TASK: ESP_AUDIO status is AEL_STATUS_STATE_RUNNING
W (4231) TONE_STREAM: No more data,ret:0 ,info.byte_pos:14829
I (4232) AUDIO_ELEMENT: IN-[IN_flash] AEL_IO_DONE,0
D (4233) ESP_AUDIO_TASK: Recv Element[IN_flash-0x3f805c98] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_FINISHED
I (5531) AUDIO_ELEMENT: IN-[DEC_mp3] AEL_IO_DONE,-2
I (5784) STAGEFRIGHTMP3_DECODER: Closed
D (5784) ESP_AUDIO_TASK: Recv POSITION INFO, msg[DEC_mp3] cmd:11 arrived,rate:48000,ch:2,bit:16,pos:14784,size:0,fmt:3
I (5788) ESP_AUDIO_TASK: Received last pos: 14784 bytes
D (5794) ESP_AUDIO_TASK: Recv Element[DEC_mp3-0x3f806000] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_FINISHED
I (5840) AUDIO_ELEMENT: IN-[resample] AEL_IO_DONE,-2
D (5840) ESP_AUDIO_TASK: Recv Element[resample-0x3f806334] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_FINISHED
I (5883) AUDIO_ELEMENT: IN-[OUT_iis] AEL_IO_DONE,-2
D (5921) ESP_AUDIO_TASK: Recv POSITION INFO, msg[OUT_iis] cmd:11 arrived,rate:48000,ch:2,bit:16,pos:363008,size:0,fmt:0
I (5921) ESP_AUDIO_TASK: Received last time: 1890 ms
D (5926) ESP_AUDIO_TASK: Recv Element[OUT_iis-0x3f806200] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_FINISHED
I (5937) ESP_AUDIO_TASK: ESP_AUDIO status is AEL_STATUS_STATE_FINISHED
[... 65 lines cut ...]
A10     : 0x3ffd5de8  A11     : 0x00000000  A12     : 0x00000014  A13     : 0x000000a5
A14     : 0x00000014  A15     : 0x00060023  SAR     : 0x00000004  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x4009ab88  LEND    : 0x4009abb6  LCOUNT  : 0x00000002
0x4009ab88: memcpy at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:168

0x4009abb6: memcpy at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:201

ELF file SHA256: 2faa012db1943d6dd675c77e02a84f9825fb3129bde5dabaddac866e4bf784f5

[... cut ...]
0x400dfcea: esp_resample_run at /Users/maojianxin/duros/esp-adf-libs-source/esp_processing/esp-resample/src/esp_resample.c:46

0x400df996: rsp_filter_process at E:/esp-adf/components/esp-adf-libs/esp_codec/filter_resample.c:148

0x400ed419: audio_element_process_running at E:/esp-adf/components/audio_pipeline/audio_element.c:287

0x400ed5c3: audio_element_task at E:/esp-adf/components/audio_pipeline/audio_element.c:438

0x400970bd: vPortTaskWrapper at E:/esp-idf-v3.2/components/freertos/xtensa/port.c:143
```

jason-mao, 2020-02-26:
@xiruilin Thank you.  I have reproduced this bug. It will be fixed ASAP.  Did this bug not occur before `4d90bdf`?

xiruilin, 2020-02-26:
@jason-mao 
Yes, it's not occur before ``4d90bdf``.
Here is our temporary solution:
```
diff --git a/esp_codec/filter_resample.c b/esp_codec/filter_resample.c
index 308650b..671982d 100644
--- a/esp_codec/filter_resample.c
+++ b/esp_codec/filter_resample.c
@@ -26,6 +26,9 @@ typedef struct rsp_filter {
     int8_t flag;  //the flag must be 0 or 1. 1 means the parameter in `resample_info_t` changed. 0 means the parameter in `resample_info_t` is original.
 } rsp_filter_t;
 
+static int new_rate;
+static int new_ch;
+
 static esp_err_t is_valid_rsp_filter_samplerate(int samplerate)
 {
     if (samplerate < 8000
@@ -56,11 +59,17 @@ esp_err_t rsp_filter_set_src_info(audio_element_handle_t self, int src_rate, int
         || is_valid_rsp_filter_channel(src_ch) != ESP_OK) {
         return ESP_ERR_INVALID_ARG;
     } else {
-        filter->flag = 1;
-        filter->resample_info->src_rate = src_rate;
-        filter->resample_info->src_ch = src_ch;
+        if (filter->rsp_hd != NULL) {
+            new_rate = src_rate;
+            new_ch = src_ch;
+            filter->flag = 1;
+        } else {
+            filter->flag = 1;
+            filter->resample_info->src_rate = src_rate;
+            filter->resample_info->src_ch = src_ch;
+        }
         ESP_LOGI(TAG, "reset sample rate of source data : %d, reset channel of source data : %d",
-                 filter->resample_info->src_rate, filter->resample_info->src_ch);
+                 src_rate, src_ch);
     }
     return ESP_OK;
 }
@@ -120,6 +129,7 @@ static esp_err_t rsp_filter_close(audio_element_handle_t self)
     rsp_filter_t *filter = (rsp_filter_t *)audio_element_getdata(self);
     if (filter->rsp_hd != NULL) {
         esp_resample_destroy(filter->rsp_hd);
+        filter->rsp_hd = NULL;
     }
     return ESP_OK;
 }
@@ -136,6 +146,8 @@ static int rsp_filter_process(audio_element_handle_t self, char *in_buffer, int
         if (ret != ESP_OK) {
             return AEL_PROCESS_FAIL;
         }
+        filter->resample_info->src_rate = new_rate;
+        filter->resample_info->src_ch = new_ch;
         ret = rsp_filter_open(self);
         if (ret != ESP_OK) {
             return AEL_PROCESS_FAIL;
```

closed 2020-03-03 by espressif-bot

xiruilin, 2020-03-04:
@jason-mao 
Thanks for update.

In new [commit](https://github.com/espressif/esp-adf/commit/d4c36b4ba8d20352205ebaab0e8ddf7d99351044),  ``esp-adf/CMakeLists.txt`` lines 6 :
```
    if(NOT DEFINED $ENV{IDF_PATH})
```
should changed to:
```
    if(NOT DEFINED ENV{IDF_PATH})
```
because [this](https://cmake.org/pipermail/cmake/2011-October/046706.html).