espressif/esp-adf #352: DuerOS例程语音识别成功率太低

liurk104, 2020-02-25:
使用ESP32-LyraTD-MSC，编译下载Dueros后，hi lexin唤醒词识别率很高, 但是之后再说的话识别率很低，按下REC说话失败的打印如下
对比正常识别的log，失败时总是出现
I (1189356) REC_ENG: Wakeup time is out
E (1189363) REC_ENG: State FETCH_DATA, wakeup_time_out:1, vad_speech_on:0, vad_time_mode:0，识别不到是因为没有录音成功？每次录音时间是多少呢？

I (1188338) DUEROS: Periph Event received: src_type:10000c, source:0x3ffc8564 cmd:1, data:0x2, data_len:3
I (1188338) DUEROS: PERIPH_NOTIFY_KEY_REC
I (1188342) REC_ENG: Recorder trigger start
I (1188356) REC_ENG: WAKEUP timer started
I (1188357) DUEROS: rec_engine_cb - REC_EVENT_WAKEUP_START
I (1188357) DUEROS: rec_engine_cb - Play tone
I (1188363) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5, que:0x3ffe255c
I (1188371) ESP_AUDIO_CTRL: Func:_ctrl_play, Line:542, MEM Total:4145268 Bytes, Inter:121180 Bytes, Dram:82484 Bytes

I (1188372) ESP_AUDIO_TASK: It's a decoder
I (1188387) ESP_AUDIO_TASK: 1.CUR IN:[IN_file],CODEC:[DEC_wav],RESAMPLE:[48000],OUT:[OUT_iis]
I (1188396) ESP_AUDIO_TASK: 2.Handles,IN:0x3f80c890,CODEC:0x3f80d010,FILTER:0x3f81785c,OUT:0x3f80d578
W (1188406) AUDIO_PIPELINE: audio_pipeline state is not running, 1
W (1188413) AUDIO_PIPELINE: audio_pipeline state is not running, 1
W (1188420) AUDIO_PIPELINE: audio_pipeline state is not running, 1
W (1188426) AUDIO_PIPELINE: audio_pipeline state is not running, 1
I (1188434) AUDIO_ELEMENT: [IN_file] Element task destroyed
I (1188440) AUDIO_ELEMENT: [DEC_wav] Element task destroyed
I (1188446) AUDIO_ELEMENT: [OUT_iis] Element task destroyed
I (1188452) AUDIO_ELEMENT: [resample] Element task destroyed
I (1188459) AUDIO_PIPELINE: audio_pipeline_unlinked
W (1188464) AUDIO_PIPELINE: There are no listener registered
W (1188470) ESP_AUDIO_TASK: Destroy the old pipeline
I (1188476) ESP_AUDIO_TASK: 2.2 Update all pipeline
I (1188482) ESP_AUDIO_TASK: 2.3 Linked new pipeline
I (1188488) AUDIO_PIPELINE: link el->rb, el:0x3f80c890, tag:IN_file, rb:0x3f80dd80
I (1188496) AUDIO_PIPELINE: link el->rb, el:0x3f80d010, tag:DEC_wav, rb:0x3f818164
I (1188504) AUDIO_PIPELINE: link el->rb, el:0x3f81785c, tag:resample, rb:0x3f80de00
I (1188512) ESP_AUDIO_TASK: 3. Previous starting...
I (1188519) AUDIO_ELEMENT: [IN_file] Element task created
I (1188524) AUDIO_ELEMENT: [DEC_wav] Element task created
I (1188530) AUDIO_ELEMENT: [DEC_wav] AEL_MSG_CMD_RESUME,state:1
I (1188531) AUDIO_ELEMENT: [resample] Element task created
I (1188542) AUDIO_ELEMENT: [OUT_iis] Element task created
I (1188549) AUDIO_ELEMENT: [IN_file] AEL_MSG_CMD_RESUME,state:1
I (1188611) FATFS_STREAM: File size is 12024 byte,pos:0
I (1188615) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:1, Available:5, que:0x3ffe255c
I (1188615) ESP_AUDIO_TASK: Received muisc info, then send MEDIA_CTRL_EVT_PLAY
I (1188622) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_PLAY, status:0
I (1188628) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:358, MEM Total:4146036 Bytes, Inter:120972 Bytes, Dram:82276 Bytes

I (1188640) AUDIO_ELEMENT: [OUT_iis] AEL_MSG_CMD_RESUME,state:1
I (1188647) I2S_STREAM: AUDIO_STREAM_WRITER
I (1188652) AUDIO_PIPELINE: Pipeline started
I (1188657) AUDIO_ELEMENT: [resample] AEL_MSG_CMD_RESUME,state:1
I (1188664) RSP_FILTER: sample rate of source data : 16000, channel of source data : 1, sample rate of destination data : 48000, channel of destination data : 2
I (1188678) ESP_AUDIO_TASK: ESP_AUDIO status is AEL_STATUS_STATE_RUNNING
W (1188707) FATFS_STREAM: No more data,ret:0
I (1188708) AUDIO_ELEMENT: IN-[IN_file] AEL_IO_DONE,0
I (1188737) ESP_AUDIO_TASK: Received last pos: 11874 bytes
I (1189006) AUDIO_ELEMENT: IN-[resample] AEL_IO_DONE,-2
I (1189006) RSP_FILTER: rsp_filter_close
I (1189062) AUDIO_ELEMENT: IN-[OUT_iis] AEL_IO_DONE,-2
I (1189099) ESP_AUDIO_TASK: Received last time: 415 ms
I (1189100) ESP_AUDIO_TASK: ESP_AUDIO status is AEL_STATUS_STATE_FINISHED
I (1189356) REC_ENG: Wakeup time is out  
E (1189363) REC_ENG: State FETCH_DATA, wakeup_time_out:1, vad_speech_on:0, vad_time_mode:0
W (1189363) REC_ENG: State WAKEUP_END
I (1189365) DUEROS: rec_engine_cb - REC_EVENT_WAKEUP_END
I (1189371) REC_ENG: state idle
I (1191267) DUEROS: Periph Event received: src_type:10000c, source:0x3ffc8564 cmd:2, data:0x2, data_len:3

liurk104, 2020-02-25:
成功率也就20%，但是板子自带的固件语音成功率还是可以的

espressif-zhanghu, 2020-05-27:
@liurk104  录音结束是根据 VAD 检测，没有声音就结束了。这个语音识别是在 dueros 服务端做的，可能需要做的是确保语音数据有没有传输至云端。

jason-mao, 2020-06-19:
建议使用这个[ESP32-Korvo-Du1906 example](https://github.com/espressif/esp-adf/tree/master/examples/korvo_du1906)

closed 2020-06-19 by jason-mao

mingchau, 2021-10-16:
试用hi lexin可以唤醒，之后按`REC`键一直没有响应。log如下：
> W (219883) REC_ENG: State VAD silence_time >= vad_off_window
W (220697) REC_ENG: State VAD silence_time >= vad_off_window
I (221348) DUEROS: PERIPH_NOTIFY_KEY_REC
I (221348) REC_ENG: Recorder trigger start
W (221348) AUDIO_EVT: There is no space in external queue
W (221503) REC_ENG: State VAD silence_time >= vad_off_window
I (222408) DUEROS: rec_engine_cb - REC_EVENT_VAD_START
I (222409) DUEROS: Recv Que DUER_CMD_START
I (222411) AUDIO_WRAPPER: duer_dcs_get_speaker_state
I (220953,tid:3ffc7cf0) lightduer_dcs_local.c( 193):Current dialog id: 218e000000000a502959d800035f1900000009
W (222425) DUEROS: duer_callback: type:0, source:0x3f80b090 data:4, data_len:0
I (221041,tid:3ffddf40) lightduer_dcs_router.c( 459):Directive name: ThrowException
E (221041,tid:3ffddf40) lightduer_dcs_system.c( 314):error code: INTERNAL_SERVICE_EXCEPTION
E (221045,tid:3ffddf40) lightduer_dcs_system.c( 315):error: didp product not exsists
I (221219,tid:3ffddf40) lightduer_dcs_router.c( 459):Directive name: ThrowException
E (221219,tid:3ffddf40) lightduer_dcs_system.c( 314):error code: INTERNAL_SERVICE_EXCEPTION
E (221225,tid:3ffddf40) lightduer_dcs_system.c( 315):error: didp product not exsists
I (221387,tid:3ffddf40) lightduer_dcs_router.c( 459):Directive name: ThrowException
E (221387,tid:3ffddf40) lightduer_dcs_system.c( 314):error code: INTERNAL_SERVICE_EXCEPTION
E (221392,tid:3ffddf40) lightduer_dcs_system.c( 315):error: didp product not exsists
I (221566,tid:3ffddf40) lightduer_dcs_router.c( 459):Directive name: ThrowException
E (221566,tid:3ffddf40) lightduer_dcs_system.c( 314):error code: INTERNAL_SERVICE_EXCEPTION
E (221571,tid:3ffddf40) lightduer_dcs_system.c( 315):error: didp product not exsists
I (221868,tid:3ffddf40) lightduer_dcs_router.c( 459):Directive name: ThrowException
E (221868,tid:3ffddf40) lightduer_dcs_system.c( 314):error code: INTERNAL_SERVICE_EXCEPTION
E (221872,tid:3ffddf40) lightduer_dcs_system.c( 315):error: didp product not exsists
I (221905,tid:3ffddf40) lightduer_dcs_router.c( 459):Directive name: ThrowException
W (223368) AUDIO_EVT: There is no space in external queue
E (221907,tid:3ffddf40) lightduer_dcs_system.c( 314):error code: INTERNAL_SERVICE_EXCEPTION
E (221917,tid:3ffddf40) lightduer_dcs_system.c( 315):error: didp product not exsists
W (221933,tid:3ffddf40) lightduer_engine.c( 697):data cache has not sent, pending..., dcache_len:1, qcache_len:1
I (222105,tid:3ffddf40) lightduer_dcs_router.c( 459):Directive name: ThrowException
E (222105,tid:3ffddf40) lightduer_dcs_system.c( 314):error code: INTERNAL_SERVICE_EXCEPTION
E (222109,tid:3ffddf40) lightduer_dcs_system.c( 315):error: didp product not exsists
I (222236,tid:3ffddf40) lightduer_dcs_router.c( 459):Directive name: ThrowException
E (222239,tid:3ffddf40) lightduer_dcs_system.c( 314):error code: INTERNAL_SERVICE_EXCEPTION
E (222243,tid:3ffddf40) lightduer_dcs_system.c( 315):error: didp product not exsists
W (223806) AUDIO_EVT: There is no space in external queue
I (222429,tid:3ffddf40) lightduer_dcs_router.c( 459):Directive name: ThrowException
E (222429,tid:3ffddf40) lightduer_dcs_system.c( 314):error code: INTERNAL_SERVICE_EXCEPTION
E (222433,tid:3ffddf40) lightduer_dcs_system.c( 315):error: didp product not exsists
I (222601,tid:3ffddf40) lightduer_dcs_router.c( 459):Directive name: ThrowException
E (222602,tid:3ffddf40) lightduer_dcs_system.c( 314):error code: INTERNAL_SERVICE_EXCEPTION
E (222607,tid:3ffddf40) lightduer_dcs_system.c( 315):error: didp product not exsists
I (222735,tid:3ffddf40) lightduer_dcs_router.c( 459):Directive name: ThrowException
E (222737,tid:3ffddf40) lightduer_dcs_system.c( 314):error code: INTERNAL_SERVICE_EXCEPTION
E (222742,tid:3ffddf40) lightduer_dcs_system.c( 315):error: didp product not exsists
I (222915,tid:3ffddf40) lightduer_dcs_router.c( 459):Directive name: ThrowException
E (222915,tid:3ffddf40) lightduer_dcs_system.c( 314):error code: INTERNAL_SERVICE_EXCEPTION
E (222919,tid:3ffddf40) lightduer_dcs_system.c( 315):error: didp product not exsists
I (223099,tid:3ffddf40) lightduer_dcs_router.c( 459):Directive name: ThrowException
E (223101,tid:3ffddf40) lightduer_dcs_system.c( 314):error code: INTERNAL_SERVICE_EXCEPTION
W (224567) REC_ENG: State VAD silence_time >= vad_off_window
I (224573) DUEROS: rec_engine_cb - REC_EVENT_VAD_STOP, state:4
E (223119,tid:3ffddf40) lightduer_dcs_system.c( 315):error: didp product not exsists
I (224605) DUEROS: Dueros DUER_CMD_STOP
W (224611) DUEROS: duer_callback: type:0, source:0x3f80b090 data:5, data_len:0
I (224615) DUEROS: Func:sys_monitor_task, Line:379, MEM Total:4156804 Bytes, Inter:124924 Bytes, Dram:110584 Bytes

这个是什么情况呢？