espressif/esp-adf #353: Bluetooth A2DP streaming from macbook Air at 48000hz not working flawlessly

rsm123, 2020-02-26:
I'm using a custom esp-wroom32 board with pipeline_bt_sink example. A2DP sink works fine with android phones and iphones but when I use macbook Air, the audio is choppy and not working flawlessly. I noticed macbook air streams at 48000hz and seems no way to specify 44100 rate. Smartphones use 44100 and works fine. Doesn't esp-adf support 48000hz sampling rate?

beckmx, 2020-03-05:
do you happen to have your code uploaded?

rsm123, 2020-03-23:
Code is the same as ‘pipeline_bt_sink’ in adf.

https://github.com/espressif/esp-adf/tree/master/examples/player/pipeline_bt_sink

jason-mao, 2020-03-25:
@rsm123 Did you help to check the I2S LRCK frequence? How is it ?

rsm123, 2020-03-26:
My apologies, I don't know how to check the LRCK frequency. The i2s DAC i'm using is PCM5102 which supports sampling rate upto 384 khz. Android and IOS A2DP streaming works fine at 44100, so it seems to be an issue with handling 48000 hz sampling rate by Macbook. There is no option to select 44100 in Macbook audio settings.

rsm123, 2021-06-15:
Is there any way to force the A2DP sample rate to 44100 so that when OSX negotiates sample rate, 44100 would be selected instead of 48000hz?