espressif/esp-adf #356: play_mp3 example  on LyraT board

embelgit, 2020-02-29:
hello, 
i am trying to run the example of play_mp3 while i am doing make menuconfig i got  issue 
esp\esp-adf\examples\get-started\play_mp3>make menuconfig
0: cd: C:UsersHpDesktopespesp-adf: No such file or directory
esp\esp-idf/make/project.mk:114: *** IDF_PATH cannot contain colons. If overridi
ws, use MSYS Unix-style /c/dir instead of C:/dir.  Stop.
please help me out  I am beginner

vaderj, 2020-03-08:
I was having the same issue and was able to resolve it by updating the ADF_PATH global variable

[image]

vaderj, 2020-03-08:
I was able to fix these issues by :

1.) updating both the ADF_PATH and IDF_PATH global variables
2.) make clean

[image]

closed 2020-03-18 by jason-mao