espressif/esp-adf #361: Memory leak when using RAW, Filter and I2S

iscapla, 2020-03-10:
Good morning.

Starting from the VoIP example, I tried to communicate 2 devices using UDP using 2 ESP32 LyraT v4.3 boards.
All the scenario works well (despite of some noise in the channel, but this could be another topic that I would like to analyze first).
Due to the limited memory of RAM that the module has and a lot of functionalities that this product has to use (MQTT + TLS + BLE + MP3 + RAW stream), I was force to use all the init and deinit functions available. To focus in the minimal resources that the module has to use:
- MQTT (with TLS and WIFI) or BLE
- MP3 or RAW stream (note that I2S could be always enabled)
- 1 TCP connection (when RAW stream is used, a UDP socket is required)

All the time, a timer is monitoring the free heap memory with "heap_caps_get_free_size(MALLOC_CAP_8BIT)" function.

When the program begins, the total HEAP memory is 228KB.
Then, I initialize the common audio components (I2S). 203KB remaining.
After WIFI get an IP, only 166KB of memory is accesible.
When MQTT with TLS are totally connected with the broker, only 110KB of can be used.

Here is where the problem appears.
1.- After a single play of a MP3 track, a ~1KB of memory is lost.
2.- After a session of RAW stream, a ~25KB of memory is lost.

So, after some usage of the application a restart is needed because there is enough memory to be allocated with a malloc call.

All the init and deinit functions are executed and the sockets (UDP in this case) is properly finished. You can find attached the library that I've being used to init and deinit all the components.

Is there any problem with the memory allocation?
Is there any possibility to remove all the malloc calls?
How can I contain the memory usage of all this issue?
What do you recommend me?

[code.zip](https://github.com/espressif/esp-adf/files/4311053/code.zip)

Thanks so much.

--
Original post in: https://www.esp32.com/viewtopic.php?f=20&t=14563&sid=7e798f573f31d13237db455c0a8f7957

ahhfzhang, 2020-06-02:
Thanks for your report.

1. about I2S stream element deinit driver crash, i think you can set `i2s_cfg.uninstall_drv = false` 
then you can use one API `audio_pipeline_deinit` to release pipeline containing all elements.

2. you can use` AUDIO_MEM_SHOW(TAG) `  to track memory usage.

jason-mao, 2020-10-15:
@iscapla Do you have any update?