espressif/esp-adf #362: IDF 4.0 Support

pilnikov, 2020-03-13:
Please add in file components\wifi_service\src\wifi_service.c  compatible with IDF 4.0 wifi callback mechanism.
&& add in components\esp_peripherals\periph_ws2812.c line #include "freertos/semphr.h"

pilnikov, 2020-03-13:
[79/131] Building C object esp-idf/wifi_service/CMakeFiles/__idf_wifi_service.dir/src/wifi_service.c.obj
In file included from Z:/esp-adf/components/wifi_service/src/wifi_service.c:39:
Z:/esp-idf/components/esp_event/include/esp_event_loop.h:2:2: warning: #warning "esp_event_loop.h is deprecated, please include esp_event.h instead" [-Wcpp]
 #warning "esp_event_loop.h is deprecated, please include esp_event.h instead"
  ^~~~~~~
In file included from Z:/esp-adf/components/wifi_service/src/wifi_service.c:36:
Z:/esp-adf/components/wifi_service/src/wifi_service.c: In function 'wifi_event_cb':
Z:/esp-adf/components/wifi_service/src/wifi_service.c:135:35: warning: passing argument 1 of 'ip4addr_ntoa' from incompatible pointer type [-Wincompatible-pointer-types]
                      ip4addr_ntoa(&event->event_info.got_ip.ip_info.ip));
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Z:/esp-idf/components/log/include/esp_log.h:320:137: note: in definition of macro 'ESP_LOG_LEVEL'
         if (level==ESP_LOG_ERROR )          { esp_log_write(ESP_LOG_ERROR,      tag, LOG_FORMAT(E, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
                                                                                                                                         ^~~~~~~~~~~
Z:/esp-idf/components/log/include/esp_log.h:287:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
 #define ESP_LOGI( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_INFO,    tag, format, ##__VA_ARGS__)
                                      ^~~~~~~~~~~~~~~~~~~
Z:/esp-adf/components/wifi_service/src/wifi_service.c:134:13: note: in expansion of macro 'ESP_LOGI'
             ESP_LOGI(TAG, "Got ip:%s",
             ^~~~~~~~
In file included from Z:/esp-idf/components/lwip/lwip/src/include/lwip/ip_addr.h:43,
                 from Z:/esp-idf/components/tcpip_adapter/include/tcpip_adapter_types.h:18,
                 from Z:/esp-idf/components/tcpip_adapter/include/tcpip_adapter.h:24,
                 from Z:/esp-idf/components/esp_netif/include/esp_netif.h:35,
                 from Z:/esp-idf/components/esp_event/include/esp_event_legacy.h:22,
                 from Z:/esp-idf/components/esp_event/include/esp_event.h:26,
                 from Z:/esp-adf/components/wifi_service/src/wifi_service.c:37:
Z:/esp-idf/components/lwip/lwip/src/include/lwip/ip4_addr.h:207:7: note: expected 'const ip4_addr_t *' {aka 'const struct ip4_addr *'} but argument is of type 'esp_ip4_addr_t *' {aka 'struct esp_ip4_addr *'}
 char *ip4addr_ntoa(const ip4_addr_t *addr);
       ^~~~~~~~~~~~
In file included from Z:/esp-adf/components/wifi_service/src/wifi_service.c:36:
Z:/esp-adf/components/wifi_service/src/wifi_service.c:135:35: warning: passing argument 1 of 'ip4addr_ntoa' from incompatible pointer type [-Wincompatible-pointer-types]
                      ip4addr_ntoa(&event->event_info.got_ip.ip_info.ip));
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Z:/esp-idf/components/log/include/esp_log.h:321:137: note: in definition of macro 'ESP_LOG_LEVEL'
         else if (level==ESP_LOG_WARN )      { esp_log_write(ESP_LOG_WARN,       tag, LOG_FORMAT(W, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
                                                                                                                                         ^~~~~~~~~~~
Z:/esp-idf/components/log/include/esp_log.h:287:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
 #define ESP_LOGI( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_INFO,    tag, format, ##__VA_ARGS__)
                                      ^~~~~~~~~~~~~~~~~~~
Z:/esp-adf/components/wifi_service/src/wifi_service.c:134:13: note: in expansion of macro 'ESP_LOGI'
             ESP_LOGI(TAG, "Got ip:%s",
             ^~~~~~~~
In file included from Z:/esp-idf/components/lwip/lwip/src/include/lwip/ip_addr.h:43,
                 from Z:/esp-idf/components/tcpip_adapter/include/tcpip_adapter_types.h:18,
                 from Z:/esp-idf/components/tcpip_adapter/include/tcpip_adapter.h:24,
                 from Z:/esp-idf/components/esp_netif/include/esp_netif.h:35,
                 from Z:/esp-idf/components/esp_event/include/esp_event_legacy.h:22,
                 from Z:/esp-idf/components/esp_event/include/esp_event.h:26,
                 from Z:/esp-adf/components/wifi_service/src/wifi_service.c:37:
Z:/esp-idf/components/lwip/lwip/src/include/lwip/ip4_addr.h:207:7: note: expected 'const ip4_addr_t *' {aka 'const struct ip4_addr *'} but argument is of type 'esp_ip4_addr_t *' {aka 'struct esp_ip4_addr *'}
 char *ip4addr_ntoa(const ip4_addr_t *addr);
       ^~~~~~~~~~~~
In file included from Z:/esp-adf/components/wifi_service/src/wifi_service.c:36:
Z:/esp-adf/components/wifi_service/src/wifi_service.c:135:35: warning: passing argument 1 of 'ip4addr_ntoa' from incompatible pointer type [-Wincompatible-pointer-types]
                      ip4addr_ntoa(&event->event_info.got_ip.ip_info.ip));
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Z:/esp-idf/components/log/include/esp_log.h:322:137: note: in definition of macro 'ESP_LOG_LEVEL'
         else if (level==ESP_LOG_DEBUG )     { esp_log_write(ESP_LOG_DEBUG,      tag, LOG_FORMAT(D, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
                                                                                                                                         ^~~~~~~~~~~
[... cut ...]
Z:/esp-adf/components/wifi_service/src/wifi_service.c:191:5: warning: 'esp_event_loop_init' is deprecated [-Wdeprecated-declarations]
     ESP_ERROR_CHECK(esp_event_loop_init(wifi_event_cb, para));
     ^~~~~~~~~~~~~~~
In file included from Z:/esp-idf/components/esp_event/include/esp_event.h:26,
                 from Z:/esp-adf/components/wifi_service/src/wifi_service.c:37:
Z:/esp-idf/components/esp_event/include/esp_event_legacy.h:227:11: note: declared here
 esp_err_t esp_event_loop_init(system_event_cb_t cb, void *ctx) __attribute__ ((deprecated));
           ^~~~~~~~~~~~~~~~~~~

pilnikov, 2020-03-13:
In file included from Z:/esp-adf/components/esp_peripherals/periph_ws2812.c:25:
Z:/esp-idf/components/freertos/include/freertos/FreeRTOS.h:831:27: error: unknown type name 'SemaphoreHandle_t'
  #define xSemaphoreHandle SemaphoreHandle_t
                           ^~~~~~~~~~~~~~~~~
Z:/esp-adf/components/esp_peripherals/periph_ws2812.c:83:5: note: in expansion of macro 'xSemaphoreHandle'
     xSemaphoreHandle          sem;
     ^~~~~~~~~~~~~~~~
Z:/esp-adf/components/esp_peripherals/periph_ws2812.c: In function 'ws2812_handle_interrupt':
Z:/esp-adf/components/esp_peripherals/periph_ws2812.c:167:9: error: implicit declaration of function 'xSemaphoreGiveFromISR'; did you mean 'xQueueGiveFromISR'? [-Werror=implicit-function-declaration]
         xSemaphoreGiveFromISR(ws->sem, &taskAwoken);
         ^~~~~~~~~~~~~~~~~~~~~
         xQueueGiveFromISR
Z:/esp-adf/components/esp_peripherals/periph_ws2812.c: In function 'ws2812_set_colors':
Z:/esp-adf/components/esp_peripherals/periph_ws2812.c:199:5: error: implicit declaration of function 'xSemaphoreTake'; did you mean 'xSemaphoreHandle'? [-Werror=implicit-function-declaration]
     xSemaphoreTake(ws->sem, portMAX_DELAY);
     ^~~~~~~~~~~~~~
     xSemaphoreHandle
Z:/esp-adf/components/esp_peripherals/periph_ws2812.c: In function '_ws2812_destroy':
Z:/esp-adf/components/esp_peripherals/periph_ws2812.c:348:9: error: implicit declaration of function 'vSemaphoreDelete'; did you mean 'xSemaphoreHandle'? [-Werror=implicit-function-declaration]
         vSemaphoreDelete(periph_ws2812->sem);
         ^~~~~~~~~~~~~~~~
         xSemaphoreHandle
Z:/esp-adf/components/esp_peripherals/periph_ws2812.c: In function 'periph_ws2812_init':
Z:/esp-adf/components/esp_peripherals/periph_ws2812.c:368:26: error: implicit declaration of function 'xSemaphoreCreateBinary'; did you mean 'xSemaphoreHandle'? [-Werror=implicit-function-declaration]
     periph_ws2812->sem = xSemaphoreCreateBinary();
                          ^~~~~~~~~~~~~~~~~~~~~~
                          xSemaphoreHandle
Z:/esp-adf/components/esp_peripherals/periph_ws2812.c:393:28: warning: assignment to 'int' from 'void *' makes integer from pointer without a cast [-Wint-conversion]
         periph_ws2812->sem = NULL;
                            ^
cc1.exe: some warnings being treated as errors

pilnikov, 2020-03-13:
warning: the default selection ESP32_CORVO_DU1906_CODEC_TAS5805M (undefined) of <choice ESP32_CORVO_DU1906_DAC> 
(defined at Z:/esp-adf/components/audio_board/Kconfig.projbuild:27) is not contained in the choice

because 

    default ESP32_CORVO_DU1906_**CODEC**_TAS5805M

   config ESP32_CORVO_DU1906_**DAC**_TAS5805M
       bool "ESP32_CORVO_DU1906_**DAC**_TAS5805M"

espressif-zhanghu, 2020-03-13:
Hello, thank you  very much. are you using the latest master branch code of ADF? and using "release/v4.0" of IDF?

pilnikov, 2020-03-13:
ADF from master branch 
$ git describe --tags
v2.0-beta3-21-g5d92078

IDF
$ git describe --tags
v4.2-dev-674-g640c7c510

espressif-zhanghu, 2020-03-16:
We have fixed this issue locally, and will soon fix in the github. thank you very much.

closed 2020-03-18 by espressif-bot