espressif/esp-adf #365: Unsure how to access raw i2s stream data

drewandre, 2020-03-17:
Hi,

I'm looking to run an fft on the esp32 lyrat 4.3 board but I'm totally lost on how to access the buffer from the a2dp stream in my main application loop. I'm fairly new to esp-idf and coming from an arduino background (love esp-idf so far).

I'm using the esp-dsp library's fft which prefers a [float type buffer](https://github.com/espressif/esp-dsp/blob/master/modules/fft/float/dsps_fft2r_fc32_ansi.c#L76) but can use [fixed inputs](https://github.com/espressif/esp-dsp/blob/master/modules/fft/fixed/dsps_fft2r_sc16_ansi.c#L117) as well. I've tried several approaches including:
- Creating an i2s write callback function which I would then use to memcopy the buffer to a global buffer which would be read and analyzed in the main loop. I could never get the callback function to fire. Not sure why - I just used `audio_element_set_write_cb(i2s_stream_writer, i2s_input_write_cb, NULL);` (see linked repo for full example)
- Creating an audio element as part of my pipeline which performed a similar task of copying the buffer into a global buffer. Not only did this not work, but I feel since that because the fft isn't influencing the audio buffer, it shouldn't be a pipeline element anyway. I'm using the fft data in led strip animations, so I just need the fft updated at a reasonable frame rate. In other words the fft analysis could probably just live in the main application loop.
- Reading the i2s data using `audio_element_getdata` which seemed to just return junk data

What are best practices for doing something like this? Especially since the fft analysis doesn't necessarily need to be part of the audio pipeline.

[Here](https://github.com/drewandre/a2dp_gatts_coex) is the repo as-is if anyone wants to look at it, but it's basically the [a2dp_gatts_coex](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/coex/a2dp_gatts_coex) esp-idf example. Lots of commented out code and failed attempts so for reference it may be easier to look at the official example.

Tarun-Narain, 2020-03-20:
Hi guys, 
Im stuck on the same problem.

jason-mao, 2020-03-20:
@drewandre @Tarun-Narain Hi guys, I think the `audio_element_multi_output` could be useful . Here is  example :https://github.com/espressif/esp-adf/blob/f6b00f3eb92ffa9aa2f1130839923166ca4fc2cd/components/audio_stream/i2s_stream.c#L257.

drewandre, 2020-03-20:
I actually did end up using the a2dp date callback function but the data is looking kind of funky. For example if I play a simple 440hz sine wave, I get amplitude data for the entire spectrum. A visual representation of the fft can be viewed [here](https://gist.github.com/drewandre/189637a3705bc41021981aef11ce02e3). Here is the full 4096 sample buffer in a [gist](https://gist.github.com/drewandre/71726ae32c88ec7d4db6f24da2f18c9e). This is apparently PCM decoded data according to the `esp_a2d_sink_data_cb_t` [type definition](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/bluetooth/esp_a2dp.html#_CPPv422esp_a2d_sink_data_cb_t).

Does anyone know if this is an appropriate way to access the a2dp i2s stream for use in an fft analysis? I feel like I _don't_ want PCM decoded data.

For reference I'm running ESP-IDF 3.3.1m ESP-ADF v2.0-beta3, and ESP-DSP v1.0. When I feed ESP-DSP's sine wave generator function, the fft is accurate. So it's not the FFT - it looks like it's the buffer data from the a2dp stream I'm passing to the fft that is not correct. 

Here is my current approach for running an esp-dsp FFT on an a2dp bluetooth audio stream:

1. Initialize fft
```
#define BUF_SIZE 4096
void init_fft()
{
  esp_err_t ret = dsps_fft2r_init_fc32(NULL, BUF_SIZE);
  if (ret != ESP_OK) { ... }
}
```
2. Initialize A2DP sink with default `bt_app_a2d_cb` config and `bt_app_a2d_data_cb` data callback
```
esp_a2d_register_callback(&bt_app_a2d_cb);
esp_a2d_sink_register_data_callback(bt_app_a2d_data_cb);
esp_a2d_sink_init();
```
3. Connect an fft data handler function in `bt_app_a2d_data_cb`
```
void bt_app_a2d_data_cb(const uint8_t *data, uint32_t len)
{
  size_t bytes_written;
  i2s_write(0, data, len, &bytes_written, portMAX_DELAY);
  copy_a2dp_buffer_for_fft(&data, len); // len = 4096
}
```
4. Copy data over to a new buffer in `copy_a2dp_buffer_for_fft`
```
static uint8_t fft_buffer[BUF_SIZE];
void copy_a2dp_buffer_for_fft(const uint8_t *data, uint32_t len)
{
  memcpy(fft_buffer, data, len);
}
```
5. Run `calculate_fft` function which can be viewed in this [gist](https://gist.github.com/drewandre/3f4ecedc07b85fd77a1b3f156e9e43ff)

drewandre, 2020-03-20:
@jason-mao lives in `i2s_stream_init`'s `_i2s_process` [callback function](https://github.com/espressif/esp-adf/blob/master/components/audio_stream/i2s_stream.c#L257) which to my understanding can't be easily overwritten. I tried adding a callback with `audio_element_set_write_cb` but that overwrites the `_i2s_process` function which kills the audio output. I could implement an entirely custom i2s handler but there must be an easier way

drewandre, 2020-03-20:
[Here](https://gist.github.com/drewandre/bb2db03d3068a42345d6b2f360638787) is a float type buffer that works well with esp-dsp's [dsps_fft2r_fc32_ansi](https://docs.espressif.com/projects/esp-dsp/en/latest/esp-dsp-apis.html#_CPPv220dsps_fft2r_fc32_ansiPfi) (which is just the output of the library's [dsps_tone_gen_f32](https://docs.espressif.com/projects/esp-dsp/en/latest/esp-dsp-apis.html#_CPPv217dsps_tone_gen_f32Pfifff)), and [here](https://gist.github.com/drewandre/42698cb4ab2cbeae63836942a8e62563) is the uint8_t type buffer that I receive in the a2dp data callback.

Totally different. I tried converting to a float buffer but didn't help.

So my question is what is the best approach for running an fft on a bluetooth audio stream? And I mean independent of any library. Is this a2dp data callback the correct buffer to analyze?

Tarun-Narain, 2020-03-21:
@drewandre @jason-mao thanks for the support,  as output of  `audio_element_set_write_cb` blocked the audio, i used i2s_write in the callback function of `audio_element_set_write_cb` and disabled the i2s_stream_writer pipeline. i didn't wanted to complicate stuff.

drewandre, 2020-03-22:
@Tarun-Narain could you provide a code sample?

Tarun-Narain, 2020-03-23:
`audio_element_set_write_cb` callback function : 
```
 int printData(audio_element_handle_t el, char *buf, int len, TickType_t wait_time, void *ctx){
	
 size_t bytes_written;
	//whatever you want to do with buffer.
 i2s_write(0, buf, len, &bytes_written, portMAX_DELAY);   
	return len;
}
```
The stream data i used : in app_main
```
audio_pipeline_register(pipeline_e, raw_read, "raw"); 
	audio_element_set_write_cb(bt_stream_reader, printData, NULL);
```

Dont direct audio_pipeline_register to i2s_stream_writer as shown.

drewandre, 2020-03-30:
This week I learned a lot about PCM audio streams and figured everything out. For future readers, [this](https://stackoverflow.com/questions/32128206/what-does-interleaved-stereo-pcm-linear-int16-big-endian-audio-look-like/32129233#32129233) SO post on PCM audio was very helpful, and [this](https://hackaday.io/project/166122/instructions) hackaday post helped figure out how to reformat the PCM stream for an fft.

closed 2020-03-30 by drewandre

oliverjc, 2020-03-30:
I'm just chiming in here to say I'm working on something similar right now. I also saw you commented on my post over [here](https://esp32.com/viewtopic.php?f=13&t=11849&p=57011#p57011) and i've been going around in circles for a few hours. I thought I found the answer, but instead, it was a forum post I made back in august 😆

The raw data I'm getting from the consists of unexpectedly small values. Considering I'm playing a 16 bit mp3 file I expected numbers in the 10s of thousands, but instead, I'm just getting values in the hundreds. 

I'd be Interested to hear your findings.