espressif/esp-adf #367: StoreProhibited Error when adding GATT code to pipeline_a2dp_sink example

Tarun-Narain, 2020-03-23:
Hi guys, 

i successfully set up dual mode, A2DP is working fine but whenever i try to connect to BLE from android Phone (gatt browser) the following error crashes esp32, i used the GATT code from esp-idf `a2dp_gatt_coex` example and tried to merge it in ADF.
BLE is advertising but it cant connect.  
Please help!
```
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400903d0  PS      : 0x00060d33  A0      : 0x8008e785  A1      : 0x3ffd67d0
0x400903d0: uxPortCompareSet at C:/Users/Tarun/Desktop/esp-adf/esp-idf/components/freertos/tasks.c:3507
 (inlined by) vPortCPUAcquireMutexIntsDisabledInternal at C:/Users/Tarun/Desktop/esp-adf/esp-idf/components/freertos/portmux_impl.inc.h:86
 (inlined by) vPortCPUAcquireMutexIntsDisabled at C:/Users/Tarun/Desktop/esp-adf/esp-idf/components/freertos/portmux_impl.h:98
 (inlined by) vTaskEnterCritical at C:/Users/Tarun/Desktop/esp-adf/esp-idf/components/freertos/tasks.c:4201

A2      : 0x0000ff09  A3      : 0x0000000b  A4      : 0x3ffcfe58  A5      : 0x3ffd84c4
A6      : 0x3ffb93cc  A7      : 0x00000000  A8      : 0x0000cdcd  A9      : 0x0000cdcd
A10     : 0xb33fffff  A11     : 0x0000abab  A12     : 0x4008336c  A13     : 0x00000001
0x4008336c: _calloc_r at C:/Users/Tarun/Desktop/esp-adf/esp-idf/components/newlib/syscalls.c:51

A14     : 0x00060d20  A15     : 0x00000001  SAR     : 0x00000003  EXCCAUSE: 0x0000001d
EXCVADDR: 0x0000ff09  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff

ELF file SHA256: f68ae8c684f085b2e7acca0cb66886997ff25d3e9085941293c1215309240209

Backtrace: 0x400903d0:0x3ffd67d0 0x4008e782:0x3ffd67f0 0x400829f1:0x3ffd6810 0x40082a22:0x3ffd6830 0x40083345:0x3ffd6850 0x4000beaf:0x3ffd6870 0x40083383:0x3ffd6890 0x4000bef5:0x3ffd68b0 0x4015a103:0x3ffd68d0 0x40179bc6:0x3ffd68f0 0x4016ced2:0x3ffd6910 0x4001791d:0x3ffd6930 0x4016d619:0x3ffd6950 0x4016d6f6:0x3ffd6990 0x400188f5:0x3ffd69c0 0x4016d0ab:0x3ffd6a00 0x40044ea7:0x3ffd6a20 0x40042cee:0x3ffd6a40 0x40044712:0x3ffd6a70 0x40045a8b:0x3ffd6aa0 0x4001a637:0x3ffd6ac0 0x40019d11:0x3ffd6af0 0x40055b4d:0x3ffd6b10 0x4016bc87:0x3ffd6b30 0x4016c25d:0x3ffd6b50 0x4008ef15:0x3ffd6b80
0x400903d0: uxPortCompareSet at C:/Users/Tarun/Desktop/esp-adf/esp-idf/components/freertos/tasks.c:3507
 (inlined by) vPortCPUAcquireMutexIntsDisabledInternal at C:/Users/Tarun/Desktop/esp-adf/esp-idf/components/freertos/portmux_impl.inc.h:86
 (inlined by) vPortCPUAcquireMutexIntsDisabled at C:/Users/Tarun/Desktop/esp-adf/esp-idf/components/freertos/portmux_impl.h:98
 (inlined by) vTaskEnterCritical at C:/Users/Tarun/Desktop/esp-adf/esp-idf/components/freertos/tasks.c:4201

0x4008e782: multi_heap_internal_lock at C:/Users/Tarun/Desktop/esp-adf/esp-idf/components/heap/multi_heap.c:380
 (inlined by) multi_heap_malloc_impl at C:/Users/Tarun/Desktop/esp-adf/esp-idf/components/heap/multi_heap.c:416

0x400829f1: heap_caps_malloc at C:/Users/Tarun/Desktop/esp-adf/esp-idf/components/heap/heap_caps.c:232

0x40082a22: heap_caps_malloc_default at C:/Users/Tarun/Desktop/esp-adf/esp-idf/components/heap/heap_caps.c:232

0x40083345: _malloc_r at C:/Users/Tarun/Desktop/esp-adf/esp-idf/components/newlib/syscalls.c:37

0x40083383: _calloc_r at C:/Users/Tarun/Desktop/esp-adf/esp-idf/components/newlib/syscalls.c:58

0x4015a103: host_recv_pkt_cb at C:/Users/Tarun/Desktop/esp-adf/esp-idf/components/bt/bluedroid/hci/hci_hal_h4.c:360

0x40179bc6: vhci_send at ??:?

0x4016ced2: r_eif_send at ??:?

0x4016d619: hci_tx_start at hci_tl.c:?

0x4016d6f6: r_hci_tl_send at ??:?

0x4016d0ab: r_hci_send_2_host_hack at ??:?

0x4016bc87: r_rw_schedule at ??:?

0x4016c25d: btdm_controller_task at ??:?

0x4008ef15: vPortTaskWrapper at C:/Users/Tarun/Desktop/esp-adf/esp-idf/components/freertos/port.c:403
```

jason-mao, 2020-03-25:
@Tarun-Narain How I can reproduced your issue?

Tarun-Narain, 2020-03-25:
@jason-mao ,
The zip file contains
1. sdkconfig
2. code (a2dp_sink_and_hfp_example.c)
3. code (gatts_demo.h) taken from a2dp_gatt_coex
4. my partition file (partitions_bt_sink_example.csv)
5. error log in a txt

[pipeline_a2dp_sink_and_hfp.zip](https://github.com/espressif/esp-adf/files/4381430/pipeline_a2dp_sink_and_hfp.zip)

Tarun-Narain, 2020-03-31:
@jason-mao i tried again merging GATTS code with player/pipeline_bt_sink but the error still persists can you check Compability of ADF(bluetooth_service.h) with GATTS, i'm approaching last stage of my prototype.

Tarun-Narain, 2020-04-02:
solved the issue, if someone is facing the same error, you need to comment the following line in bluetooth_service.h
```
ESP_ERROR_CHECK(esp_bt_controller_mem_release(ESP_BT_MODE_BLE));
```

Tarun-Narain, 2020-04-13:
@jason-mao i'm closing the issue, can you look at the pull request, is everything fine with it?
#374

closed 2020-04-13 by Tarun-Narain