espressif/esp-adf #368: dueros编译失败

Kevincoooool, 2020-03-25:
我使用windows环境和linux环境下，IDF和ADF都更新到了最新，都出了同一个错，请问如何解决？
错误如下：
CC build/wpa_supplicant/src/utils/wpa_debug.o
CC build/wpa_supplicant/src/utils/common.o
CC build/wpa_supplicant/src/utils/base64.o
CC build/wpa_supplicant/src/utils/ext_password.o
CC build/wpa_supplicant/src/utils/wpabuf.o
CC build/wpa_supplicant/src/utils/uuid.o
CC build/wpa_supplicant/src/esp_supplicant/esp_wpas_glue.o
CC build/wpa_supplicant/src/esp_supplicant/esp_wps.o
CC build/wpa_supplicant/src/esp_supplicant/esp_hostap.o
CC build/wpa_supplicant/src/esp_supplicant/esp_wpa2.o
CC build/wpa_supplicant/src/esp_supplicant/esp_wpa_main.o
CC build/wpa_supplicant/src/wps/wps_attr_process.o
CC build/wpa_supplicant/src/wps/wps_validate.o
CC build/wpa_supplicant/src/wps/wps_attr_parse.o
CC build/wpa_supplicant/src/wps/wps_dev_attr.o
CC build/wpa_supplicant/src/wps/wps.o
CC build/wpa_supplicant/src/wps/wps_common.o
CC build/wpa_supplicant/src/wps/wps_registrar.o
CC build/wpa_supplicant/src/wps/wps_attr_build.o
CC build/wpa_supplicant/src/wps/wps_enrollee.o
AR build/wpa_supplicant/libwpa_supplicant.a
CC build/xtensa/debug_helpers.o
CC build/xtensa/trax.o
AR build/xtensa/libxtensa.a
Generating esp32.project.ld
LD build/esp_dueros.elf
/home/kevin/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/kevin/esp/esp-adf/components/clouds/dueros/lightduer/libduer-device.a(lightduer_http_client_ops.c.o):(.literal+0x24): undefined reference to `lwip_connect_r'
/home/kevin/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/kevin/esp/esp-adf/components/clouds/dueros/lightduer/libduer-device.a(lightduer_http_client_ops.c.o):(.literal+0x38): undefined reference to `lwip_send_r'
/home/kevin/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/kevin/esp/esp-adf/components/clouds/dueros/lightduer/libduer-device.a(lightduer_http_client_ops.c.o):(.literal+0x40): undefined reference to `lwip_recv_r'
/home/kevin/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/kevin/esp/esp-adf/components/clouds/dueros/lightduer/libduer-device.a(lightduer_http_client_ops.c.o): in function `connect':
/Users/maojianxin/duros/esp-adf-internal/esp-idf/components/lwip/include/lwip/lwip/sockets.h:577: undefined reference to `lwip_connect_r'
/home/kevin/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/kevin/esp/esp-adf/components/clouds/dueros/lightduer/libduer-device.a(lightduer_http_client_ops.c.o): in function `socket_send':
/Users/maojianxin/duros/lightduer/libduer-device_3.0.7/modules/HTTP/lightduer_http_client_ops.c:196: undefined reference to `lwip_send_r'
/home/kevin/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/kevin/esp/esp-adf/components/clouds/dueros/lightduer/libduer-device.a(lightduer_http_client_ops.c.o): in function `socket_recv':
/Users/maojianxin/duros/lightduer/libduer-device_3.0.7/modules/HTTP/lightduer_http_client_ops.c:177: undefined reference to `lwip_recv_r'
/home/kevin/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/kevin/esp/esp-adf/components/clouds/dueros/lightduer/libduer-device.a(baidu_ca_socket_adp.c.o):(.literal+0x58): undefined reference to `lwip_connect_r'
/home/kevin/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/kevin/esp/esp-adf/components/clouds/dueros/lightduer/libduer-device.a(baidu_ca_socket_adp.c.o):(.literal+0x60): undefined reference to `lwip_send_r'
/home/kevin/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/kevin/esp/esp-adf/components/clouds/dueros/lightduer/libduer-device.a(baidu_ca_socket_adp.c.o):(.literal+0x6c): undefined reference to `lwip_recv_r'
/home/kevin/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/kevin/esp/esp-adf/components/clouds/dueros/lightduer/libduer-device.a(baidu_ca_socket_adp.c.o): in function `bcasoc_parse_ip':
/Users/maojianxin/duros/lightduer/libduer-device_3.0.7/platform/source-freertos/baidu_ca_socket_adp.c:126: undefined reference to `lwip_connect_r'
/home/kevin/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/kevin/esp/esp-adf/components/clouds/dueros/lightduer/libduer-device.a(baidu_ca_socket_adp.c.o): in function `bcasoc_send':
/Users/maojianxin/duros/lightduer/libduer-device_3.0.7/platform/source-freertos/baidu_ca_socket_adp.c:345: undefined reference to `lwip_send_r'
/home/kevin/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/kevin/esp/esp-adf/components/clouds/dueros/lightduer/libduer-device.a(baidu_ca_socket_adp.c.o): in function `bcasoc_recv':
/Users/maojianxin/duros/lightduer/libduer-device_3.0.7/platform/source-freertos/baidu_ca_socket_adp.c:380: undefined reference to `lwip_recv_r'
collect2: error: ld returned 1 exit status
/home/kevin/esp/esp-idf/make/project.mk:528: recipe for target '/home/kevin/esp/dueros/build/esp_dueros.elf' failed
make: *** [/home/kevin/esp/dueros/build/esp_dueros.elf] Error 1
kevin@ubuntu1804:~/esp/dueros$

jason-mao, 2020-03-27:
@Kevincoooool  2e820278180d4d1d209ff22bba5cdd820d90b9f1，修复了这个问题。但是当前`dueros` example还只能用make 编译，Cmake编译还会报错。

Kevincoooool, 2020-03-27:
你好，刚刚已经编译成功并且下载进去了，从百度dueros获得的profile是之前的，现在来看百度开发平台已经更新似乎不支持了，请问有最新的解决方案吗

jason-mao, 2020-03-27:
@Kevincoooool 是有些调整。您这边是个人，还是公司呢？

Kevincoooool, 2020-03-27:
我是个人，是想体验下dueros的。。结果今天就更新了

jason-mao, 2020-06-19:
建议使用这个[ESP32-Korvo-Du1906 example](https://github.com/espressif/esp-adf/tree/master/examples/korvo_du1906)

closed 2020-06-19 by jason-mao