espressif/esp-adf #369: “http_stream" cause memory leaks

xiruilin, 2020-03-28:
Hi,
In commit 20a2933, “http_stream" cause memory leaks.
Log below:
```
I (24326) HTTP_STREAM: total_bytes=757
I (24326) HTTP_STREAM: Func:_resolve_playlist, Line:235, MEM Total:3867404 Bytes, Inter:64156 Bytes, Dram:62440 Bytes

I (24332) HTTP_STREAM: Live stream URI. Need to be fetched again!
I (24399) HTTP_STREAM: total_bytes=71252
I (24401) AAC_DECODER: new audio song
I (24401) AAC_DECODER: this audio is TS AAC
I (24426) RSP_FILTER: reset sample rate of source data : 24000, reset channel of source data : 2
I (24427) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:1, Available:5, que:0x3ffe0a78
I (24433) ESP_AUDIO_TASK: Received muisc info, then send MEDIA_CTRL_EVT_PLAY
I (24440) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_PLAY, status:0
I (24447) AUDIO_ELEMENT: [OUT_iis] Element task created
I (24452) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:358, MEM Total:3729456 Bytes, Inter:38836 Bytes, Dram:37120 Bytes

I (24464) AUDIO_ELEMENT: [OUT_iis] AEL_MSG_CMD_RESUME,state:1
I (24470) I2S_STREAM: AUDIO_STREAM_WRITER
I (24495) AUDIO_PIPELINE: Pipeline started
I (24497) AUDIO_ELEMENT: [resample] AEL_MSG_CMD_RESUME,state:1
I (24497) RSP_FILTER: sample rate of source data : 24000, channel of source data : 2, sample rate of destination data : 48000, channel of destination data : 2
I (24510) ESP_AUDIO_TASK: ESP_AUDIO status is AEL_STATUS_STATE_RUNNING

......

W (59195) HTTP_STREAM: No more data,errno:0, total_bytes:72004, rlen = 0
I (59195) AUDIO_ELEMENT: IN-[IN_http] AEL_IO_DONE,0
I (59197) AUDIO_ELEMENT: [IN_http] AEL_MSG_CMD_RESUME,state:1
I (59205) HTTP_STREAM: Fetching again...
I (59575) HTTP_STREAM: total_bytes=757
I (59575) HTTP_STREAM: Func:_resolve_playlist, Line:235, MEM Total:3726844 Bytes, Inter:28408 Bytes, Dram:26692 Bytes

W (59581) HLS_PLAYLIST: URI exist
I (59586) HTTP_STREAM: Live stream URI. Need to be fetched again!
I (59763) HTTP_STREAM: total_bytes=70124
W (66330) HTTP_STREAM: No more data,errno:0, total_bytes:70124, rlen = 0

......

I (87950) HTTP_STREAM: Fetching again...
I (88213) HTTP_STREAM: total_bytes=757
I (88214) HTTP_STREAM: Func:_resolve_playlist, Line:235, MEM Total:3722016 Bytes, Inter:24124 Bytes, Dram:22408 Bytes

I (88219) HTTP_STREAM: Live stream URI. Need to be fetched again!
I (88276) HTTP_STREAM: total_bytes=70688
W (95070) HTTP_STREAM: No more data,errno:0, total_bytes:70688, rlen = 0

......

I (123929) HTTP_STREAM: Fetching again...
I (124204) HTTP_STREAM: total_bytes=757
I (124204) HTTP_STREAM: Func:_resolve_playlist, Line:235, MEM Total:3717068 Bytes, Inter:19740 Bytes, Dram:18024 Bytes

W (124209) HLS_PLAYLIST: URI exist
I (124214) HTTP_STREAM: Live stream URI. Need to be fetched again!
I (124275) HTTP_STREAM: total_bytes=70688
W (131085) HTTP_STREAM: No more data,errno:0, total_bytes:70688, rlen = 0

......

I (152712) HTTP_STREAM: Fetching again...
I (152927) HTTP_STREAM: total_bytes=757
I (152927) HTTP_STREAM: Func:_resolve_playlist, Line:235, MEM Total:3712448 Bytes, Inter:15456 Bytes, Dram:13740 Bytes

W (152932) HLS_PLAYLIST: URI exist
I (152937) HTTP_STREAM: Live stream URI. Need to be fetched again!
I (152983) HTTP_STREAM: total_bytes=72756
W (159914) HTTP_STREAM: No more data,errno:0, total_bytes:72756, rlen = 0

......

I (181463) HTTP_STREAM: Fetching again...
I (181749) HTTP_STREAM: total_bytes=757
I (181749) HTTP_STREAM: Func:_resolve_playlist, Line:235, MEM Total:3708492 Bytes, Inter:11568 Bytes, Dram:9852 Bytes

W (181756) HLS_PLAYLIST: URI exist
I (181759) HTTP_STREAM: Live stream URI. Need to be fetched again!
I (181802) HTTP_STREAM: total_bytes=71440
W (188658) HTTP_STREAM: No more data,errno:0, total_bytes:71440, rlen = 0

......

I (210166) HTTP_STREAM: Fetching again...
I (210298) HTTP_STREAM: total_bytes=757
I (210299) HTTP_STREAM: Func:_resolve_playlist, Line:235, MEM Total:3704536 Bytes, Inter:7676 Bytes, Dram:5960 Bytes

I (210305) HTTP_STREAM: Live stream URI. Need to be fetched again!
I (210371) HTTP_STREAM: total_bytes=70500
W (217372) HTTP_STREAM: No more data,errno:0, total_bytes:70500, rlen = 0

......

I (246307) HTTP_STREAM: Fetching again...
I (246550) HTTP_STREAM: total_bytes=757
I (246550) HTTP_STREAM: Func:_resolve_playlist, Line:235, MEM Total:3700592 Bytes, Inter:3788 Bytes, Dram:2072 Bytes

W (246556) HLS_PLAYLIST: URI exist
I (246561) HTTP_STREAM: Live stream URI. Need to be fetched again!
I (246607) HTTP_STREAM: total_bytes=70124
W (253389) HTTP_STREAM: No more data,errno:0, total_bytes:70124, rlen = 0

......

I (275022) HTTP_STREAM: Fetching again...
I (275242) HTTP_STREAM: total_bytes=757
I (275242) HTTP_STREAM: Func:_resolve_playlist, Line:235, MEM Total:3696652 Bytes, Inter:1960 Bytes, Dram:244 Bytes

W (275248) HLS_PLAYLIST: URI exist
I (275252) HTTP_STREAM: Live stream URI. Need to be fetched again!
I (275294) HTTP_STREAM: total_bytes=71064
E (276770) wifi:alloc eb len=24 type=3 fail, heap:3686216

E (276770) wifi:m f null
```

jason-mao, 2020-03-28:
@xiruilin Thank you for your report. Which URL are you use?

xiruilin, 2020-03-28:
@jason-mao 

url: http://open.ls.qingting.fm/live/332/64k.m3u8?format=ts

closed 2020-03-30 by espressif-bot