espressif/esp-adf #371: "play_http_mp3" continuous loop play, can't end

xiruilin, 2020-03-31:
Hi,
When play http mp3, It's not over.
Code snippet:
```
    esp_audio_play(handle, AUDIO_CODEC_TYPE_DECODER, "https://dl.espressif.com/dl/audio/gs-16b-2c-44100hz.mp3", 0);
```
Log below:
```
I (3888) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5, que:0x3ffd19a4
I (3896) ESP_AUDIO_TASK: It's a decoder
I (3900) ESP_AUDIO_TASK: 1.CUR IN:[IN_http],CODEC:[DEC_mp3],RESAMPLE:[48000],OUT:[OUT_iis],rate:0, ch:0
I (3910) ESP_AUDIO_TASK: 2.Handles,IN:0x3f805df8,CODEC:0x3f806048,FILTER:0x3f806378,OUT:0x3f806258
I (3920) ESP_AUDIO_TASK: 2.2 Update all pipeline
I (3925) ESP_AUDIO_TASK: 2.3 Linked new pipeline
I (3931) AUDIO_PIPELINE: link el->rb, el:0x3f805df8, tag:IN_http, rb:0x3f806498
I (3939) AUDIO_PIPELINE: link el->rb, el:0x3f806048, tag:DEC_mp3, rb:0x3f808cd8
I (3947) AUDIO_PIPELINE: link el->rb, el:0x3f806378, tag:resample, rb:0x3f80ad18
I (3955) ESP_AUDIO_TASK: 3. Previous starting...
I (3960) AUDIO_ELEMENT: [IN_http] Element task created
I (3966) AUDIO_ELEMENT: [DEC_mp3] Element task created
I (3971) AUDIO_ELEMENT: [DEC_mp3] AEL_MSG_CMD_RESUME,state:1
I (3971) AUDIO_ELEMENT: [resample] Element task created
I (3978) STAGEFRIGHTMP3_DECODER: MP3 opened
I (3972) AUDIO_ELEMENT: [IN_http] AEL_MSG_CMD_RESUME,state:1
I (3972) ESP_AUDIO_CTRL: Func:_ctrl_play, Line:657, MEM Total:4301728 Bytes, Inter:201068 Bytes, Dram:190080 Bytes

I (4852) HTTP_STREAM: total_bytes=254415
I (4861) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:1, Available:5, que:0x3ffd19a4
I (4862) ESP_AUDIO_TASK: Received muisc info, then send MEDIA_CTRL_EVT_PLAY
I (4866) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_PLAY, status:0
I (4872) AUDIO_ELEMENT: [OUT_iis] Element task created
I (4878) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:358, MEM Total:4227384 Bytes, Inter:163268 Bytes, Dram:152280 Bytes

I (4890) AUDIO_ELEMENT: [OUT_iis] AEL_MSG_CMD_RESUME,state:1
I (4896) I2S_STREAM: AUDIO_STREAM_WRITER
I (4920) AUDIO_PIPELINE: Pipeline started
I (4920) AUDIO_ELEMENT: [resample] AEL_MSG_CMD_RESUME,state:1
I (4920) RSP_FILTER: sample rate of source data : 44100, channel of source data : 2, sample rate of destination data : 48000, channel of destination data : 2
I (4935) ESP_AUDIO_TASK: ESP_AUDIO status is AEL_STATUS_STATE_RUNNING
I (4942) HTTP_MP3_EXAMPLE: ESP_AUDIO_CALLBACK_FUNC, st: 1, err: 0, src: 0
W (19939) HTTP_STREAM: No more data,errno:0, total_bytes:254415, rlen = 0
I (19939) AUDIO_ELEMENT: IN-[IN_http] AEL_IO_DONE,0
I (19941) AUDIO_ELEMENT: [IN_http] AEL_MSG_CMD_RESUME,state:1
I (20066) HTTP_STREAM: total_bytes=254415
W (35857) HTTP_STREAM: No more data,errno:0, total_bytes:254415, rlen = 0
I (35858) AUDIO_ELEMENT: IN-[IN_http] AEL_IO_DONE,0
I (35859) AUDIO_ELEMENT: [IN_http] AEL_MSG_CMD_RESUME,state:1
I (35991) HTTP_STREAM: total_bytes=254415
W (51757) HTTP_STREAM: No more data,errno:0, total_bytes:254415, rlen = 0
I (51758) AUDIO_ELEMENT: IN-[IN_http] AEL_IO_DONE,0
I (51759) AUDIO_ELEMENT: [IN_http] AEL_MSG_CMD_RESUME,state:1
I (51888) HTTP_STREAM: total_bytes=254415
W (67676) HTTP_STREAM: No more data,errno:0, total_bytes:254415, rlen = 0
I (67676) AUDIO_ELEMENT: IN-[IN_http] AEL_IO_DONE,0
I (67678) AUDIO_ELEMENT: [IN_http] AEL_MSG_CMD_RESUME,state:1
I (67824) HTTP_STREAM: total_bytes=254415
W (83576) HTTP_STREAM: No more data,errno:0, total_bytes:254415, rlen = 0
I (83576) AUDIO_ELEMENT: IN-[IN_http] AEL_IO_DONE,0
I (83578) AUDIO_ELEMENT: [IN_http] AEL_MSG_CMD_RESUME,state:1
I (83727) HTTP_STREAM: total_bytes=254415
W (99495) HTTP_STREAM: No more data,errno:0, total_bytes:254415, rlen = 0
I (99496) AUDIO_ELEMENT: IN-[IN_http] AEL_IO_DONE,0
I (99498) AUDIO_ELEMENT: [IN_http] AEL_MSG_CMD_RESUME,state:1
I (99619) HTTP_STREAM: total_bytes=254415
W (115395) HTTP_STREAM: No more data,errno:0, total_bytes:254415, rlen = 0
I (115395) AUDIO_ELEMENT: IN-[IN_http] AEL_IO_DONE,0
I (115397) AUDIO_ELEMENT: [IN_http] AEL_MSG_CMD_RESUME,state:1
I (115533) HTTP_STREAM: total_bytes=254415
W (131301) HTTP_STREAM: No more data,errno:0, total_bytes:254415, rlen = 0
I (131302) AUDIO_ELEMENT: IN-[IN_http] AEL_IO_DONE,0
I (131304) AUDIO_ELEMENT: [IN_http] AEL_MSG_CMD_RESUME,state:1
I (131475) HTTP_STREAM: total_bytes=254415
W (147215) HTTP_STREAM: No more data,errno:0, total_bytes:254415, rlen = 0
I (147215) AUDIO_ELEMENT: IN-[IN_http] AEL_IO_DONE,0
I (147217) AUDIO_ELEMENT: [IN_http] AEL_MSG_CMD_RESUME,state:1
I (147350) HTTP_STREAM: total_bytes=254415

......

```

jason-mao, 2020-03-31:
@xiruilin Thank you for your report.

vikramdattu, 2020-03-31:
Hi @xiruilin can you please share some more info:
1. IDF version
2. If issue is always visible OR is random?

xiruilin, 2020-03-31:
Hi,
@vikramdattu 
1. IDF version: v4.2-dev-910-g4fe04f115-dirty
2. The issue is always visible

If use esp_adf_libs, there are usually the following callback:
```
int _http_stream_event_handle(http_stream_event_msg_t *msg)
{
    if (msg->event_id == HTTP_STREAM_RESOLVE_ALL_TRACKS) {
        return ESP_OK;
    }

    if (msg->event_id == HTTP_STREAM_FINISH_TRACK) {
        return http_stream_next_track(msg->el);
    }
    if (msg->event_id == HTTP_STREAM_FINISH_PLAYLIST) {
        return http_stream_fetch_again(msg->el);
    }
    return ESP_OK;
}

```
In latest commit ``http_stream.c``, ``http_stream_next_track`` changed to:
```
esp_err_t http_stream_next_track(audio_element_handle_t el)
{
    http_stream_t *http = (http_stream_t *)audio_element_getdata(el);
    audio_element_reset_state(el);
    audio_element_info_t info;
    audio_element_getinfo(el, &info);
    info.byte_pos = 0;
    info.total_bytes = 0;
    audio_element_setinfo(el, &info);
    http->is_open = false;
    return ESP_OK;
}
```
This is usually true for live stream, but not single music.
Our interim solution is as follows:
```
esp_err_t http_stream_next_track(audio_element_handle_t el)
{
    http_stream_t *http = (http_stream_t *)audio_element_getdata(el);
    if (!(http->enable_playlist_parser && http->is_playlist_resolved)) {
        ESP_LOGW(TAG, "there are no track");
        return ESP_OK;
    }
    audio_element_reset_state(el);
    audio_element_info_t info;
    audio_element_getinfo(el, &info);
    info.byte_pos = 0;
    info.total_bytes = 0;
    audio_element_setinfo(el, &info);
    http->is_open = false;
    return ESP_OK;
}

```
Thanks!

vikramdattu, 2020-04-01:
@xiruilin was thinking on the similar lines, since this is the function which resets the element state.

Thanks for the insight.

closed 2020-04-01 by espressif-bot