espressif/esp-adf #376: Error when `make deploy ...`

sasilva1998, 2020-04-04:
Hi, I am trying to get this into my ESPLyrat, but haven't been able to do it, followed all instructions with no succes, I am getting the following:

```
CC ../../drivers/bus/softspi.c
CC ../../drivers/dht/dht.c
CC build-GENERIC_SPIRAM/frozen_content.c
LINK build-GENERIC_SPIRAM/application.elf
xtensa-esp32-elf-ld: cannot find -lesp_dlna
Makefile:746: recipe for target 'build-GENERIC_SPIRAM/application.elf' failed
make: *** [build-GENERIC_SPIRAM/application.elf] Error 1
```
Can anyone help with this please?

xocoatzin, 2020-05-14:
Hi @sasilva1998 , were you able to find a solution?

sasilva1998, 2020-05-14:
Hi @xocoatzin not yet, are you having that same error?

xocoatzin, 2020-05-14:
Yeah, the same, but digging into it, I found that `esp_dlna` was removed in [this commit](https://github.com/espressif/esp-adf-libs/commit/c8a229a0bf4ebc57ed761e62fd79e379e4055b27), so it seems to be a mismatch between versions of `esp-adf` and `esp-adf-libs`

xocoatzin, 2020-05-14:
Yeah, that was it. The two commits below compile just fine. Should use that instead of `master`.

```
cd $ESPADF
git checkout 588b493d4d88183e349c96dae137697ba3f7a9f1
cd $ESPADF/components/esp-adf-libs/
git checkout 8391235c6bda889c09eb1caf49acefd444528bbf
```

closed 2020-06-02 by espressif-bot