espressif/esp-adf #377: i2s_stream.c does not support 32bit on I2S

rpi-Receiver, 2020-04-05:
Trying to implement voip example with spk0415hm4h mems microphone. For the microphone WS must be BCLK/64, so i set I2S_SAMPLE_RATE 16000, I2S_CHANNEL 2, I2S_BITS 32 which did not work.
So i changed
static int _i2s_read(audio_element_handle_t self, char *buffer, int len, TickType_t ticks_to_wait, void *context)
{
    i2s_stream_t *i2s = (i2s_stream_t *)audio_element_getdata(self);
    size_t bytes_read = 0;
    audio_element_info_t info;
    audio_element_getinfo(self, &info);
    if (info.bits == 32) {
        int32_t *temp_buf = audio_malloc(sizeof(int32_t)/sizeof(int16_t) * len);
        if(temp_buf == NULL){
            ESP_LOGE(TAG, "audio_malloc failed for temp_buf. (line %d)", __LINE__);
            return ESP_ERR_NO_MEM;
        }
        i2s_read(i2s->config.i2s_port, temp_buf, sizeof(int32_t)/sizeof(int16_t) * len, &bytes_read, ticks_to_wait);
        int16_t *buf16 = (int16_t *)buffer;
        int i;
        for (i = 0; i < (bytes_read / sizeof(int32_t)); i++) {
            buf16[i] = temp_buf[i] / 0x4000;
        }
        if (temp_buf) {
            free(temp_buf);
        }
        bytes_read = bytes_read / (sizeof(int32_t)/sizeof(int16_t));
    } else {
        i2s_read(i2s->config.i2s_port, buffer, len, &bytes_read, ticks_to_wait);
    }
    if (bytes_read > 0) {
        if (info.channels == 1) {
            i2s_mono_fix(info.bits, (uint8_t *)buffer, bytes_read);
        }
        info.byte_pos += bytes_read;
        audio_element_setinfo(self, &info);
    }
    return bytes_read;
}

and
static int _i2s_write(audio_element_handle_t self, char *buffer, int len, TickType_t ticks_to_wait, void *context)
{
    i2s_stream_t *i2s = (i2s_stream_t *)audio_element_getdata(self);
    size_t bytes_written = 0;
    audio_element_info_t info;
    audio_element_getinfo(self, &info);
    if (info.bits == 32) {
        int32_t *temp_buf = audio_calloc(1, sizeof(int32_t)/sizeof(int16_t) * len);
        if(temp_buf == NULL){
            ESP_LOGE(TAG, "audio_calloc failed for temp_buf. (line %d)", __LINE__);
            return ESP_ERR_NO_MEM;
        }
        int16_t *buf16 = (int16_t *)buffer;
        for (int i = 0; i < (len / sizeof(int16_t)); i++) {
            temp_buf[i] = buf16[i] * 0x4000;
        }
        i2s_write(i2s->config.i2s_port, (char *)temp_buf, len * sizeof(int32_t)/sizeof(int16_t), &bytes_written, ticks_to_wait);
        if (temp_buf) {
            free(temp_buf);
        }
        bytes_written = bytes_written / 2;
    } else
    i2s_write(i2s->config.i2s_port, buffer, len, &bytes_written, ticks_to_wait);
    return bytes_written;
}

and now it works for my microphone.
Is there a chance to bring this to the master tree - as I think streams are always 16bit?

espressif-zhanghu, 2020-05-27:
@rpi-Receiver   i2s always support 32bit.

rpi-Receiver, 2020-05-27:
yes,  i2s hardware always supports 32bit, but as I see stream implementation does not.
In _i2s_open() -> Line 153 -> i2s_info.bits = 16;

rpi-Receiver, 2020-05-27:
I modified the voip example according to my hardware, I did not try i2s_stream_set_clk() to modify clk, bits, etc as I did think that this would not help at all. Am I wrong?

jason-mao, 2020-10-15:
@rpi-Receiver For the _i2s_write , I think the https://github.com/espressif/esp-adf/issues/318#issuecomment-572047168  is better. I think above _i2s_read implementation is not a general usage, 32bit and 16bit align at 4 byte, i2s_read is ok for them.

closed 2020-10-15 by jason-mao