espressif/esp-adf #380: Icecast/Shoutcast support in HTTP stream

alxvasilev, 2020-04-15:
Hello,
I am wondering if Shoutcast/Icecast support is planned, and if yes, what is the timeline. If not, I may be able to contribute a patch - in that case, is there interest in accepting the patch?

Greetings
Alex

vikramdattu, 2020-04-16:
HTTP steam already supports ICY URIs in a very basic way.
Can you please check with your example URI as a HTTP URI?

alxvasilev, 2020-04-16:
Do you mean .m3u and .m3u8 playlists? That part works indeed, but what I am interested in, is receiving meta data in the stream, like song name, etc. Looking at the source code of http_stream.c, there is no support for that. Do you plan to support it, are patches welcome?

vikramdattu, 2020-04-16:
@alxvasilev I was talking more of this stuff:
https://github.com/espressif/esp-idf/commit/742d4689dd728cb3bd5f06747b04b7d8930b7d02

One can use ICY uri as a normal http uri and use that in http stream

As far as other intricate details, it's not implemented yet

closed 2021-02-23 by jason-mao