espressif/esp-adf #381: "stacks_in_external_mem" needs to be improved

xiruilin, 2020-04-18:
Hi,
The feature of ``stacks_in_external_mem`` free up a lot of internal memory, is great.

There are several questions to explore:

1. File ``/esp-adf/components/audio_pipeline/audio_element.c``, line 456, in function
``audio_element_task()``,
```cpp
audio_thread_delete_task(&el->audio_thread);
```
should change to:
```cpp
vTaskDelete(NULL);
```
Because it's in task inner, the task must be deleted unconditionally.

2. File ``/esp-adf/components/audio_pipeline/audio_element.c``, line 1011, in function
``audio_element_run()``,
```cpp
        int ret = audio_thread_create(&el->audio_thread, el->tag, audio_element_task, el, el->task_stack,
                                      el->task_prio, el->stack_in_ext, el->task_core);
```
should change to:
```cpp
        ret = audio_thread_create(&el->audio_thread, el->tag, audio_element_task, el, el->task_stack,
                                  el->task_prio, el->stack_in_ext, el->task_core);
```
Because the ``ret`` variable defined here will cover the ``ret`` variable of line 1001.

3. File ``/esp-adf/components/audio_sal/audio_thread.c``, line 84, in function
``audio_thread_create()``,
```cpp
    *p_handle = handle;
```
should move to line 52, because if has some error occured, jump to line 87:
```cpp
audio_thread_create_error:
```
``handle`` no assignment to ``*p_handle``, maybe memory leak.

4. If enable ``stack_in_ext`` of ``filter_resample``, will be all sorts of very strange mistakes, for example:
```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4009be84  PS      : 0x00060b33  A0      : 0x8009aa3d  A1      : 0x3f832e90
0x4009be84: vListInsert at E:/esp-idf/components/freertos/list.c:189 (discriminator 1)

A2      : 0x3ffb09e8  A3      : 0x3ffedc98  A4      : 0x00000004  A5      : 0xb33fffff
A6      : 0x00000001  A7      : 0x00000001  A8      : 0x3ffbb288  A9      : 0x00000014
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3f833300  A13     : 0x0000000c
A14     : 0x3ffb65fc  A15     : 0x00000005  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x4009e6c1  LEND    : 0x4009e6d1  LCOUNT  : 0xffffffff
0x4009e6c1: strlen at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S:84

0x4009e6d1: strlen at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S:96

Backtrace:0x4009be81:0x3f832e90 0x4009aa3a:0x3f832eb0 0x40099bcb:0x3f832ed0 0x40099c81:0x3f832f10 0x400842be:0x3f832f30 0x40084399:0x3f832f70 0x4022508e:0x3f832f90 0x40228463:0x3f8332a0 0x4023e69b:0x3f8332d0 0x400977fd:0x3f833300 0x40149146:0x3f833350 0x40149649:0x3f833400 0x4013dde5:0x3f833420 0x4013e1d8:0x3f833440 0x4013f0e1:0x3f833470 0x4013e97b:0x3f8334b0
0x4009be81: vListInsert at E:/esp-idf/components/freertos/list.c:206

0x4009aa3a: vTaskPlaceOnEventList at E:/esp-idf/components/freertos/tasks.c:2904 (discriminator 2)

0x40099bcb: xQueueGenericReceive at E:/esp-idf/components/freertos/queue.c:1595

0x40099c81: xQueueTakeMutexRecursive at E:/esp-idf/components/freertos/queue.c:634

0x400842be: lock_acquire_generic at E:/esp-idf/components/newlib/locks.c:156

0x40084399: _lock_acquire_recursive at E:/esp-idf/components/newlib/locks.c:170

0x4022508e: _vfprintf_r at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:853 (discriminator 2)

0x40228463: vprintf at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vprintf.c:34

0x4023e69b: esp_log_writev at E:/esp-idf/components/log/log.c:189

0x400977fd: esp_log_write at E:/esp-idf/components/log/log.c:199

0x40149146: mp3_decoder_open at /home/houhaiyan/esp-audio-adf-projects/esp-adf-libs-source/esp_processing/esp-wrapper/mp3_decoder.c:150

0x40149649: _mp3_decoder_open at /home/houhaiyan/esp-audio-adf-projects/esp-adf-libs-source/esp_processing/esp-wrapper/mp3_decoder.c:496

0x4013dde5: audio_element_process_init at E:/esp-adf/components/audio_pipeline/audio_element.c:170

0x4013e1d8: audio_element_on_cmd at E:/esp-adf/components/audio_pipeline/audio_element.c:266

0x4013f0e1: audio_event_iface_waiting_cmd_msg at E:/esp-adf/components/audio_pipeline/audio_event_iface.c:243 (discriminator 1)

0x4013e97b: audio_element_task at E:/esp-adf/components/audio_pipeline/audio_element.c:437

ELF file SHA256: 8846a7981affab44

Rebooting...
```
...

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4009be99  PS      : 0x00060833  A0      : 0x8009ac1d  A1      : 0x3ffe1ff0
0x4009be99: uxListRemove at E:/esp-idf/components/freertos/list.c:219

A2      : 0x3ffecf3c  A3      : 0x3ffc4614  A4      : 0x00000000  A5      : 0xb33fffff
A6      : 0x00000001  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x3ffcf330
A10     : 0x3ffecc28  A11     : 0x3ffc4614  A12     : 0x00000000  A13     : 0xb33fffff
A14     : 0x00000001  A15     : 0x00000001  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000004  LBEG    : 0x4009d6e0  LEND    : 0x4009d6fc  LCOUNT  : 0x00000000
0x4009d6e0: memcpy at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:175

0x4009d6fc: memcpy at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:197

Backtrace:0x4009be96:0x3ffe1ff0 0x4009ac1a:0x3ffe2010 0x40099863:0x3ffe2030 0x4013f103:0x3ffe2070 0x4013db81:0x3ffe2090 0x4013e892:0x3ffe20d0 0x4013f9a2:0x3ffe2110 0x4013fb2d:0x3ffe2130 0x401433e1:0x3ffe2150
0x4009be96: uxListRemove at E:/esp-idf/components/freertos/list.c:215

0x4009ac1a: xTaskRemoveFromEventList at E:/esp-idf/components/freertos/tasks.c:3120

0x40099863: xQueueGenericSend at E:/esp-idf/components/freertos/queue.c:772

0x4013f103: audio_event_iface_cmd at E:/esp-adf/components/audio_pipeline/audio_event_iface.c:252 (discriminator 1)

0x4013db81: audio_element_cmd_send at E:/esp-adf/components/audio_pipeline/audio_element.c:149

0x4013e892: audio_element_resume at E:/esp-adf/components/audio_pipeline/audio_element.c:1121

0x4013f9a2: audio_pipeline_resume at E:/esp-adf/components/audio_pipeline/audio_pipeline.c:320

0x4013fb2d: audio_pipeline_run at E:/esp-adf/components/audio_pipeline/audio_pipeline.c:360

0x401433e1: media_ctrl_task at /Users/maojianxin/duros/esp-adf-libs-source/esp_audio/src/media_ctrl_task.c:585

ELF file SHA256: 8846a7981affab44

Rebooting...
```

jason-mao, 2020-04-21:
@xiruilin Thank you for your great suggestions.

xiruilin, 2020-04-21:
@jason-mao 
Hi
If enable ``stack_in_ext`` of ``aac_decoder``, will be an occasional fault during playback. This fault may appear after a while, or it may not appear for more than 10 consecutive hours.
url: ``http://cnlive.cnr.cn/hls/bjyygb.m3u8``
Log:
```
W (55319) HTTP_STREAM: No more data,errno:0, total_bytes:99640, rlen = 0
I (55319) AUDIO_ELEMENT: IN-[IN_http] AEL_IO_DONE,0
I (55321) AUDIO_ELEMENT: [IN_http] AEL_MSG_CMD_RESUME,state:1
I (55609) HTTP_STREAM: total_bytes=98888
W (65575) HTTP_STREAM: No more data,errno:0, total_bytes:98888, rlen = 0
I (65575) AUDIO_ELEMENT: IN-[IN_http] AEL_IO_DONE,0
I (65577) AUDIO_ELEMENT: [IN_http] AEL_MSG_CMD_RESUME,state:1
I (65846) HTTP_STREAM: total_bytes=99828
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x401662e0  PS      : 0x00060430  A0      : 0x80165e78  A1      : 0x3f838ba0
0x401662e0: huffspec_fxp at /Users/maojianxin/duros/esp-adf-libs-source/esp_codec/esp-aac/src/huffspec_fxp.c:481

A2      : 0x00000000  A3      : 0x40167f08  A4      : 0x00000000  A5      : 0x3fff3640
0x40167f08: decode_huff_cw_tab10 at /Users/maojianxin/duros/esp-adf-libs-source/esp_codec/esp-aac/src/decode_huff_cw_binary.c:569

A6      : 0x3f42b8f0  A7      : 0x00000099  A8      : 0x3fff3e48  A9      : 0x00000008
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000031  A13     : 0x3f4283a8
A14     : 0x00000135  A15     : 0x00000002  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000099  LBEG    : 0x4009ca2a  LEND    : 0x4009ca35  LCOUNT  : 0x00000000
0x4009ca2a: memset at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:150

0x4009ca35: memset at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:160

Backtrace:0x401662dd:0x3f838ba0 0x40165e75:0x3f838c10 0x40160081:0x3f838c60 0x40157ee2:0x3f838cb0 0x40152af9:0x3f838d40 0x40152d9d:0x3f838de0 0x4014651a:0x3f838e00
0x401662dd: huffspec_fxp at /Users/maojianxin/duros/esp-adf-libs-source/esp_codec/esp-aac/src/huffspec_fxp.c:479

0x40165e75: getics at /Users/maojianxin/duros/esp-adf-libs-source/esp_codec/esp-aac/src/getics.c:662

0x40160081: huffdecode at /Users/maojianxin/duros/esp-adf-libs-source/esp_codec/esp-aac/src/huffdecode.c:505

0x40157ee2: PVMP4AudioDecodeFrame at /Users/maojianxin/duros/esp-adf-libs-source/esp_codec/esp-aac/src/pvmp4audiodecoderframe.c:753

0x40152af9: aac_decoder_process at /home/houhaiyan/esp-audio-adf-projects/esp-adf-libs-source/esp_processing/esp-wrapper/aac_decoder.c:888

0x40152d9d: _aac_decoder_process at /home/houhaiyan/esp-audio-adf-projects/esp-adf-libs-source/esp_processing/esp-wrapper/aac_decoder.c:1001

0x4014651a: audio_element_process_running at E:/esp-adf/components/audio_pipeline/audio_element.c:290
 (inlined by) audio_element_task at E:/esp-adf/components/audio_pipeline/audio_element.c:441

ELF file SHA256: 68e12e74f2cc43fe

Rebooting...
```

Thanks!

closed 2020-04-30 by espressif-bot

xiruilin, 2020-05-02:
@jason-mao 
Hi,
Nice to see the update on this issue, thanks!

``stack_in_ext`` of ``filter_resample`` mistake has been solved,
but issue ``stack_in_ext`` of ``aac_decoder`` still exists.