espressif/esp-adf #383: Does esp_audio can play with raw_writer?

younghyunjo, 2020-04-21:
I want to play mp3 using the esp_audio_play API.
MP3 is in memory and I want to use raw_writer.
How can I pass the uri format to the esp_audio_play() API?

espressif-zhanghu, 2020-05-25:
@younghyunjo   Such as "file://sdcard/test.wav" or "http://iot.espressif.com/file/example.mp3".
do help you?

jason-mao, 2020-05-31:
@younghyunjo Sure.
- Add the a raw stream writer instance to esp_audio, like:
```c
    raw_stream_cfg_t raw_cfg = RAW_STREAM_CFG_DEFAULT();
    raw_cfg.type = AUDIO_STREAM_WRITER;
    raw_play_handle = raw_stream_init(&raw_cfg);
    esp_audio_input_stream_add(handle, raw_play_handle);
```
- Fill music data by ` raw_stream_write(raw_play_handle, buff, buffer_length);`

- Then you can start `esp_audio_play`by`raw://sdcard/test/test.mp3`, `//sdcard/test/test.mp3` is mock path, no need really file.`.mp3` used to select your added decoder.

jason-mao, 2020-06-19:
The Higher level API `audio_player` integrate all supported URLs, please refer to `https://github.com/espressif/esp-adf/tree/master/examples/korvo_du1906/components/audio_player`

closed 2020-06-19 by jason-mao