espressif/esp-adf #384: SIP destroy task not working, causing slow leak in internal heap memory

sosaucily, 2020-04-22:
Hello,

When using esp-adf v2.0.0-beta3 I notice that sometimes the call to `esp_sip_destroy(sip_client)` does not always seem to properly kill the sip task. It is rare, but sometimes I continue to get the following SIP logs repeating even after successfully esp_sip_destroy. 

```

OPTIONS sip:phoneindemo1@91.121.30.149:5060 SIP/2.0
Via: SIP/2.0/UDP 10.27.99.238:5060;branch=z9hG4bK--339532111;rport
From: <sip:phoneindemo1@91.121.30.149:5060>;tag=-390163123
To: <sip:phoneindemo1@91.121.30.149:5060>
Contact: <sip:phoneindemo1@10.27.99.238:5060>
Max-Forwards: 70
Call-ID: A17ED83692E938F8720B082282B7F70BF43007612155
CSeq: 6 OPTIONS
Expires: 3600
User-Agent: ESP32 SIP/2.0
Content-Length: 0
Allow: INVITE, ACK, CANCEL, BYE, REFER, MESSAGE, OPTIONS, INFO, SUBSCRIBE
Supported: replaces, norefersub, extended-refer, timer, X-cisco-serviceuri
Allow-Events: presence, kpml
Authorization: Digest username="phoneindemo1", realm="sip.antisip.com", nonce="XqBWMV6gVQXEKjD2iHvKLCY1HuOOdIJNU0SJOVKhhustiA1uUKFRNg==", uri="sip:91.121.30.149:5060", response="106d63554bc513dfdf4aeb4bee9454da", algorithm=MD5, nc=00000001, cnonce="01f0ff2bab136a2d", qop="auth"

W (350894) SIP: [2020-04-22/16:31:01]=======================>>
W (350944) SIP: [2020-04-22/16:31:01]<<=====READ 0554 bytes==
I (350944) SIP: 

SIP/2.0 407 Proxy Authentication Required
Via: SIP/2.0/UDP 10.27.99.238:5060;branch=z9hG4bK--339532111;rport=5060;received=195.228.147.122
From: <sip:phoneindemo1@91.121.30.149:5060>;tag=-390163123
To: <sip:phoneindemo1@91.121.30.149:5060>;tag=3fff72771323d080125777a3d95c3d38.8e7c5c8c
Call-ID: A17ED83692E938F8720B082282B7F70BF43007612155
CSeq: 6 OPTIONS
Proxy-Authenticate: Digest realm="sip.antisip.com", nonce="XqBWUl6gVSZAdkKYsGTB1Q97QCWPKIyQU0SJOVKhhustiA1uUKFRNg==", qop="auth"
Server: kamailio (5.3.3 (x86_64/linux))
Content-Length: 0
```

I know esp_sip_destroy was called because I see the following in the logs: `I (335854) SIP: SIP destroy done`

As these logs loop around and around every 20-30 seconds, or so, I see the INTERNAL HEAP slowly shrink, by about 400-600 (bytes?)

starting around
```
I (348874) HEAP_DEBUG: ======== INTERNAL HEAP INFO ========
I (348874) HEAP_DEBUG: Total allocated bytes:   283124
I (348884) HEAP_DEBUG: Total free bytes:         76652
I (348884) HEAP_DEBUG: Minimum free bytes:       29824
I (348904) HEAP_DEBUG: Largest free block:       27220
I (348904) HEAP_DEBUG: Alloceted blocks:           460
I (348914) HEAP_DEBUG: Free blocks:                 19
I (348914) HEAP_DEBUG: Total blocks:               479
```

and getting to about 
```
I (3569965) HEAP_DEBUG: ======== INTERNAL HEAP INFO ========
I (3569965) HEAP_DEBUG: Total allocated bytes:   338848
I (3569995) HEAP_DEBUG: Total free bytes:         18556
I (3569995) HEAP_DEBUG: Minimum free bytes:       18256
I (3569995) HEAP_DEBUG: Largest free block:       16980
I (3570015) HEAP_DEBUG: Alloceted blocks:          1029
I (3570015) HEAP_DEBUG: Free blocks:                 43
I (3570025) HEAP_DEBUG: Total blocks:              1072
```

At which point the wifi will fail to allocate memory and we'll start getting the following type of memory symptom.

```
E (3563675) wifi:m f auth

I (3564755) wifi:state: auth -> init (200)
I (3564755) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (3564755) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (3564765) wifi:state: init -> auth (b0)
E (3564795) wifi:alloc eb len=32 type=3 fail, heap:509240

E (3564825) wifi:m f auth

I (3565855) wifi:state: auth -> init (200)
I (3565905) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (3566005) WIFI-PROV: wifi disconnected event
```

Any thoughts on why sip_destroy doesn't always work, maybe how we can access and kill the task ourselves? And is there a memory leak in there?

Thanks

sosaucily, 2020-05-25:
Follow up on this, I suspect what was happening is that we were calling `esp_sip_start` twice, which created two sip tasks. And our subsequent calls to `esp_sip_stop` and `esp_sip_destroy` only stopped and killed one of the task, leaving the other one in ghost mode. Perhaps something that can be improved in the internal code of the esp task.

ahhfzhang, 2020-06-01:
Thanks for your report.
1. The second call of `esp_sip_start` should return a failure and will get a warning: 
`W : SIP: SIP service was started`

2. About sending `OPTIONS` will cause mem leak issue, it has been fixed in the lastest esp_sip lib.

jason-mao, 2020-06-19:
@sosaucily How about your problem? I believe that the latest code  works fine.

sosaucily, 2020-08-06:
We haven't had any issues recently, so I would say it's safe to close. Thank you for following up!

closed 2020-08-06 by sosaucily