espressif/esp-adf #388: Warning on adf v2.0-beta3

szucsi23, 2020-05-05:
Hey!
We have encountered a Warning about a defined but not used function in esp-adf/components/esp_peripherals/periph_wifi.c, line 294.
The wifi_reconnect_timer() function is never used if there is an idf version under 4.0.

Putting the function definition between an if-else would solve the problem:

#if !defined(ESP_IDF_VERSION) || (ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(4, 0, 0))
static void wifi_reconnect_timer(xTimerHandle tmr)
{
    esp_periph_handle_t periph = (esp_periph_handle_t)pvTimerGetTimerID(tmr);
    esp_periph_stop_timer(periph);
    esp_wifi_connect();
}
#endif

espressif-zhanghu, 2020-05-25:
@szucsi23 Yes.We are found this issue too. Fix is coming.

szucsi23, 2020-06-03:
Great, thanks!
We also see an error now:

/root/esp/esp-adf/components/esp_peripherals/periph_console.c: In function 'console_get_line':
/root/esp/esp-adf/components/esp_peripherals/periph_console.c:110:9: error: 'nread' undeclared (first use in this function)
     if (nread <= 0) {
         ^
and:
/root/esp/esp-adf/components/esp_peripherals/periph_console.c:100:10: error: variable 'tx' set but not used [-Werror=unused-but-set-variable]
     char tx[3];
          ^

espressif-zhanghu, 2020-06-19:
thanks you, we have fixed it .
commit id：2e99eea6cb4dd034970989cfdda6e5f8d5d20998

closed 2020-06-19 by jason-mao

sosaucily, 2020-07-01:
Hi @jason-mao, do you know when this will be merged into the v2.0 release?

sosaucily, 2020-08-06:
I see this is merged into 2.1. Thanks!