espressif/esp-adf #392: ESP_LYRAT_MINI_V1_1_BOARD 连接dueros无法播放

Shaopus, 2020-05-09:
` ESP_LYRAT_MINI_V1_1_BOARD ` 已经连接上dueros，但是无法播放，查看了下 `log` 感觉是播放频率不对，板子初始的频率是 `16K` ，但播放的是 `48k` ，以下是播放音乐的 `log` ，请帮忙查看，谢谢
```shel
I (1362591) DUEROS: rec_engine_cb - REC_EVENT_WAKEUP_END
I (1362602) REC_ENG: state idle
I (1362597) AUDIO_WRAPPER: Playing speak: http://res.iot.baidu.com:80/api/v1/tts
/tLxlqU9wnCwvRxfXiqKLzCtY-XXnLPHegVdqM5VE1XzdBgKS2f5ZV73_ARGCsB2TFQrSlhkBJZ7de2L
H6yRTM1exOSF5jhtsA0Uq1AEm5zg.mp3
I (1362591) DUEROS: Dueros DUER_CMD_STOP
I (1362623) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5, que:0x3ffcb17c

I (1362635) ESP_AUDIO_CTRL: Func:_ctrl_play, Line:659, MEM Total:4083276 Bytes,
Inter:52524 Bytes, Dram:39124 Bytes

W (1362641) DUEROS: duer_callback: type:0, source:0x3f807b90 data:5, data_len:0
I (1362655) DUEROS: Dueros DUER_CMD_STOP
I (1362647) ESP_AUDIO_TASK: It's a decoder
I (1362665) ESP_AUDIO_TASK: 1.CUR IN:[IN_http],CODEC:[DEC_mp3],RESAMPLE:[48000],
OUT:[OUT_iis],rate:0, ch:0
W (1362665) DUEROS: duer_callback: type:0, source:0x3f807b90 data:5, data_len:0
I (1362675) ESP_AUDIO_TASK: 2.Handles,IN:0x3f807294,CODEC:0x3f80765c,FILTER:0x3f
8179b4,OUT:0x3f807aa0
W (1362692) AUDIO_PIPELINE: Without stop, st:1
W (1362697) AUDIO_PIPELINE: Without wait stop, st:1
W (1362703) AUDIO_PIPELINE: Without stop, st:1
W (1362708) AUDIO_PIPELINE: Without wait stop, st:1
W (1362714) AUDIO_ELEMENT: [IN_file] Element has not create when AUDIO_ELEMENT_T
ERMINATE
W (1362722) AUDIO_ELEMENT: [DEC_wav] Element has not create when AUDIO_ELEMENT_T
ERMINATE
W (1362731) AUDIO_ELEMENT: [OUT_iis] Element has not create when AUDIO_ELEMENT_T
ERMINATE
I (1362740) AUDIO_ELEMENT: [resample] Element task destroyed
I (1362747) AUDIO_PIPELINE: audio_pipeline_unlinked
W (1362752) AUDIO_PIPELINE: There are no listener registered
W (1362758) ESP_AUDIO_TASK: Destroy the old pipeline
I (1362764) ESP_AUDIO_TASK: 2.2 Update all pipeline
I (1362769) ESP_AUDIO_TASK: 2.3 Linked new pipeline
I (1362775) AUDIO_PIPELINE: link el->rb, el:0x3f807294, tag:IN_http, rb:0x3f8178
38
I (1362784) AUDIO_PIPELINE: link el->rb, el:0x3f80765c, tag:DEC_mp3, rb:0x3f81f0
ac
I (1362792) AUDIO_PIPELINE: link el->rb, el:0x3f8179b4, tag:resample, rb:0x3f817
714
I (1362800) ESP_AUDIO_TASK: 3. Previous starting...
I (1362827) AUDIO_ELEMENT: [IN_http] Element task created
I (1362828) AUDIO_THREAD: The DEC_mp3 task allocate stack on external memory
I (1362831) AUDIO_ELEMENT: [DEC_mp3] Element task created
I (1362836) AUDIO_ELEMENT: [DEC_mp3] AEL_MSG_CMD_RESUME,state:1
I (1362837) AUDIO_THREAD: The resample task allocate stack on external memory
I (1362843) STAGEFRIGHTMP3_DECODER: MP3 opened
I (1362851) AUDIO_ELEMENT: [resample] Element task created
I (1362840) AUDIO_ELEMENT: [IN_http] AEL_MSG_CMD_RESUME,state:1
I (1362969) HTTP_STREAM: total_bytes=0
I (1362985) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:1, Available:5, que:0x3ffcb17c

I (1362986) ESP_AUDIO_TASK: Received muisc info, then send MEDIA_CTRL_EVT_PLAY
I (1362991) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_PLAY, status:0
I (1362998) AUDIO_ELEMENT: [OUT_iis] Element task created
I (1363003) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:358, MEM Total:4028716
 Bytes, Inter:36708 Bytes, Dram:23308 Bytes

I (1363015) AUDIO_ELEMENT: [OUT_iis] AEL_MSG_CMD_RESUME,state:1
I (1363022) I2S_STREAM: AUDIO_STREAM_WRITER
I (1363027) AUDIO_PIPELINE: Pipeline started
I (1363027) AUDIO_ELEMENT: [resample] AEL_MSG_CMD_RESUME,state:1
I (1363039) RSP_FILTER: sample rate of source data : 16000, channel of source da
ta : 1, sample rate of destination data : 48000, channel of destination data : 2

I (1363054) ESP_AUDIO_TASK: ESP_AUDIO status is AEL_STATUS_STATE_RUNNING
I (1363061) AUDIO_WRAPPER: esp_auido status:1,err:0,state:1
```

espressif-zhanghu, 2020-05-25:
请试一下： 使用 git submodule update --init 更新一下 esp-adf-libs 。

jason-mao, 2020-06-19:
建议使用这个[ESP32-Korvo-Du1906 example](https://github.com/espressif/esp-adf/tree/master/examples/korvo_du1906)

closed 2020-06-19 by jason-mao