espressif/esp-adf #393: ESP32-LyraT-Mini: Blink example doesn't start unless pressing RESET button

menway, 2020-05-10:
[ copied from esp-idf repository, because I don't really know whether this is caused by idf or adf :-( ]

Hello,
I just bought an ESP32-LyraT-Mini board and flashed the blink example program under esp-idf repository. When the board is connected to the host computer and running by "make flash monitor" command, the blinking function works fine. But when I only connect the power cable, the board is "idle" and doesn't blink. After manually pressing the RESET button, it starts blinking.

video: https://drive.google.com/file/d/1OFcvq2AEqKQzyI-GyT05DoX1OxfXq2OJ/view?usp=sharing

My question is: is this a correct behavior, or did I miss anything? How to make it running as soon as I plug in a power cable? Thanks.

espressif-zhanghu, 2020-05-27:
@menway  , we will fix it ,
thank you

jason-mao, 2020-06-01:
Closed by same issue #398

closed 2020-06-01 by jason-mao