espressif/esp-adf #397: LyraT 4.3 OTA Example - sd card error

shotkee, 2020-05-18:
> I (2100) wifi:AP's beacon interval = 102400 us, DTIM period = 1
> [0;32mI (3310) event: sta ip: 192.168.1.66, mask: 255.255.255.0, gw: 192.168.1.1[0m
> [0;32mI (3310) HTTPS_OTA_EXAMPLE: [1.2] Mount SDCard[0m
> [0;32mI (3320) SDCARD: Trying to mount with base path=/sdcard[0m
> [0;32mI (3360) SDCARD: CID name ASTC!
> [0m
> [0;32mI (3810) HTTPS_OTA_EXAMPLE: [2.0] Create OTA service[0m
> [0;32mI (3810) HTTPS_OTA_EXAMPLE: [2.1] Set upgrade list[0m
> [0;32mI (3810) HTTPS_OTA_EXAMPLE: [2.2] Start OTA service[0m
> [0;32mI (3810) OTA_SERVICE: set list  2[0m
> [0;32mI (3810) HTTPS_OTA_EXAMPLE: Func:app_main, Line:154, MEM Total:191456 Bytes
> 
> [0m
> [0;32mI (3820) OTA_DEFAULT: data upgrade uri file://sdcard/flash_tone.bin[0m
> [0;32mI (3830) FATFS_STREAM: File size is 0 byte,pos:0[0m
> [0;31mE (3840) FATFS_STREAM: Failed to open file /sdcard/flash_tone.bin[0m
> [0;31mE (3840) AUDIO_ELEMENT: [file] AEL_STATUS_ERROR_OPEN[0m
> [0;31mE (3850) OTA_DEFAULT: reader stream init failed[0m
> [0;33mW (3850) AUDIO_ELEMENT: [file] Element has not create when AUDIO_ELEMENT_TERMINATE[0m

sd card works fine with other examples (http_play_and_save_to_ file and etc.)

espressif-zhanghu, 2020-05-25:
@shotkee  your sd card have **flash_tone.bin** file ?

shotkee, 2020-05-25:
@espressif-zhanghu Yes, I tried to copy a random file named flash_tone.bin on the sd card. I tried with two different cards. But it was not a valid firmware image bin-file , but just an random file.
Can you explain the purpose of this file. I think, I have a misunderstanding about this.

jason-mao, 2020-06-19:
@shotkee As your said, the error message is right. `flash_tone.bin` created by `https://github.com/espressif/esp-adf/tree/master/tools/audio_tone`.

closed 2020-06-19 by jason-mao