espressif/esp-adf #398: ESP32-LyraT-Mini: Blink example doesn't start unless pressing RESET button (IDFGH-3265)

menway, 2020-05-20:
Hello,
I just bought an ESP32-LyraT-Mini board and flashed the blink example program under esp-idf repository. When the board is connected to the host computer and running by "make flash monitor" command, the blinking function works fine. But when I only connect the power cable, the board is "idle" and doesn't blink. After manually pressing the RESET button, it starts blinking.

video: https://drive.google.com/file/d/1OFcvq2AEqKQzyI-GyT05DoX1OxfXq2OJ/view?usp=sharing

My question is: is this a correct behavior, or did I miss anything? How to make it running as soon as I plug in a power cable? Thanks.

espressif-zhanghu, 2020-05-26:
@menway  thank you , we will fix it

Yohannfra, 2021-09-15:
Hello @jason-mao did you fix it ? I still have the exact same issue.
Thanks

HengYongChao, 2021-09-16:
Hi @Yohannfra 

I'v tried on ESP32-Lyrat-mini v1.2, set blink led GPIO22, then flash the get-started/blink example, it works fine.

IDF V3.3.5 and IDF V4.2.2 both work good. Have you changed the restart behavior in menuconfig?

Yohannfra, 2021-09-16:
No i didn't change the restard behavior.
When you flash the example if you disconnect the esp32 and close the idf.py monitor then reconnect the esp32 does it start blinking right away? 

Thanks

Yohannfra, 2021-09-16:
What variables in menuconfig change the restart behavior @HengYongChao ? 
Thanks

Yohannfra, 2021-09-16:
So the issue is caused by the C9 capacitor on the Lyrat-mini being too small (1uF). 
To fix this issue I had to solder a bigger (I used a 220uF) capacitor in parallel with the C9 capacitor.

HengYongChao, 2021-09-17:
Hi @Yohannfra 

After replacing a larger capacitor, your board can be reset by the software correctly? If yes, it seems that the capacitor of the reset circuit is aging.

jason-mao, 2023-04-25:
This topic has become inactive so I'm going to close the issue. Please reopen this if you have any questions or need any further assistance.

closed 2023-04-25 by jason-mao

RuvanMK, 2023-05-26:
Update, I can see the C9 cap when zooming in on the schematic PDF https://dl.espressif.com/dl/schematics/Layout_ESP32-LyraT-Mini_V1.2_20220317.pdf

I'm still wondering if this will be fixed in a future board revision and whether the PCB design files are available anywhere. Thanks!

RuvanMK, 2023-05-30:
Second update, while the larger cap in parallel with C9 does cause the board to run the main program at boot, it stops the board from being able to be put into flashing mode either by the python flasher or by the buttons on the board.

EyalSi, 2024-11-01:
Just got an ESP32-LyraT-Mini V1.2 with the exact same problem.
For me, it makes the board useless... 
Is there another solution for that other than soldering another capacitor to C9 (it is so tiny!)?