espressif/esp-adf #399: LyraT-v4.3官方例程asr播放声音错误abort

zxc312654, 2020-05-21:
第一次播放提示声音是可以的，第二次就会强制abort()复位
这是什么原因?

终端信息如下
I (4270) asr_keywords: wake up
W (4326) FATFS_STREAM: No more data,ret:0
phrase ID: 0, prob: 0.772613
I (7764) asr_keywords: turn on air-condition
W (7766) AUDIO_PIPELINE: Without stop, st:1
W (7766) AUDIO_PIPELINE: Without wait stop, st:1
W (7770) AUDIO_PIPELINE: Without stop, st:1
W (7774) AUDIO_PIPELINE: Without wait stop, st:1
W (7780) AUDIO_PIPELINE: There are no listener registered
W (7786) ESP_AUDIO_TASK: Destroy the old pipeline
E (7794) FreeRTOS: FreeRTOS Task "IN_file" should not return, Aborting now!
abort() was called at PC 0x4008f8af on core 0

zxc312654, 2020-05-21:
晕
把cfg.prefer_type = ESP_AUDIO_PREFER_MEM;
改成cfg.prefer_type = ESP_AUDIO_PREFER_SPEED;
就没问题了, 奇怪的BUG...

closed 2020-05-21 by zxc312654

xumanhua, 2020-05-30:
同样问题，关键esp_audio是库，很多具体的看不到，官方能否解释一下不
看代码解释esp_audio works on sepcific type, default memory is preferred. ，有建议用 ESP_AUDIO_PREFER_MEM;

jason-mao, 2020-05-31:
@xumanhua @zxc312654 你们出问题的ADF commit ID分别是多少？

zxc312654, 2020-05-31:
7217e9c603aafb4e1fbbe564e949782b0a7a9ce5

zxc312654, 2020-05-31:
@jason-mao 7217e9c603aafb4e1fbbe564e949782b0a7a9ce5

jason-mao, 2020-05-31:
@zxc312654 请用新0f2fa97c64ad9c39486a594d5894175c44226c8b测试一下

zxc312654, 2020-05-31:
@jason-mao 刚刚测试了还是同样的问题。

zxc312654, 2020-05-31:
已经测试了还是同样的问题。

zxc312654, 2020-05-31:
更新后原来的工程都出问题了，SPIRAM一直出错...