espressif/esp-adf #400: can not find lwip_sendto_r   hepme

weilaideai2008, 2020-05-21:
esptool.py v2.8
Generated D:/esp-v4.0/esp-adf/examples/advanced_examples/dlna/build/bootloader/bootloader.bin
[572/573] Linking CXX executable dlna_example.elf
FAILED: dlna_example.elf
cmd.exe /C "cd . && D:\esp-tool\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address  -nostdlib @CMakeFiles\dlna_example.elf.rsp  -o dlna_example.elf  && cd ."
d:/esp-tool/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: D:/esp-v4.0/esp-adf-v2.0/components/esp-adf-libs/esp_ssdp/lib\libesp_ssdp.a(esp_ssdp.o):(.literal.ssdp_response+0x20): undefined reference to `lwip_sendto_r'
d:/esp-tool/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: D:/esp-v4.0/esp-adf-v2.0/components/esp-adf-libs/esp_ssdp/lib\libesp_ssdp.a(esp_ssdp.o):(.literal.ssdp_create_sock+0x1c): undefined reference to `lwip_setsockopt_r'
d:/esp-tool/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: D:/esp-v4.0/esp-adf-v2.0/components/esp-adf-libs/esp_ssdp/lib\libesp_ssdp.a(esp_ssdp.o):(.literal.ssdp_create_sock+0x20): undefined reference to `lwip_bind_r'
d:/esp-tool/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: D:/esp-v4.0/esp-adf-v2.0/components/esp-adf-libs/esp_ssdp/lib\libesp_ssdp.a(esp_ssdp.o):(.literal.ssdp_read_sock+0xc): undefined reference to `lwip_recvfrom_r'
d:/esp-tool/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: D:/esp-v4.0/esp-adf-v2.0/components/esp-adf-libs/esp_ssdp/lib\libesp_ssdp.a(esp_ssdp.o): in function `ssdp_response':
/home/michael/work/esp/esp-adf-libs-source/esp_ssdp/esp_ssdp.c:405: undefined reference to `lwip_sendto_r'
d:/esp-tool/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: D:/esp-v4.0/esp-adf-v2.0/components/esp-adf-libs/esp_ssdp/lib\libesp_ssdp.a(esp_ssdp.o): in function `setsockopt':
/home/michael/work/esp/esp-adf-internal/esp-idf/components/lwip/lwip/src/include/lwip/sockets.h:571: undefined reference to `lwip_setsockopt_r'
d:/esp-tool/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: D:/esp-v4.0/esp-adf-v2.0/components/esp-adf-libs/esp_ssdp/lib\libesp_ssdp.a(esp_ssdp.o): in function `ssdp_create_sock':
/home/michael/work/esp/esp-adf-libs-source/esp_ssdp/esp_ssdp.c:249: undefined reference to `lwip_bind_r'
d:/esp-tool/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: /home/michael/work/esp/esp-adf-libs-source/esp_ssdp/esp_ssdp.c:260: undefined reference to `lwip_setsockopt_r'
d:/esp-tool/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: D:/esp-v4.0/esp-adf-v2.0/components/esp-adf-libs/esp_ssdp/lib\libesp_ssdp.a(esp_ssdp.o): in function `recvfrom':
/home/michael/work/esp/esp-adf-internal/esp-idf/components/lwip/lwip/src/include/lwip/sockets.h:585: undefined reference to `lwip_recvfrom_r'
d:/esp-tool/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: D:/esp-v4.0/esp-adf-v2.0/components/esp-adf-libs/esp_ssdp/lib\libesp_ssdp.a(esp_ssdp.o): in function `setsockopt':
/home/michael/work/esp/esp-adf-internal/esp-idf/components/lwip/lwip/src/include/lwip/sockets.h:571: undefined reference to `lwip_setsockopt_r'
d:/esp-tool/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: D:/esp-v4.0/esp-adf-v2.0/components/esp-adf-libs/esp_ssdp/lib\libesp_ssdp.a(esp_ssdp.o): in function `sendto':
/home/michael/work/esp/esp-adf-internal/esp-idf/components/lwip/lwip/src/include/lwip/sockets.h:591: undefined reference to `lwip_sendto_r'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

weilaideai2008, 2020-05-21:
I Used ESP-IDF V4.0 Release And ESP_ADF V2.0 Release

espressif-zhanghu, 2020-05-25:
you can update ESP_ADF with using **git submodule update --init** ,maybe that esp-adf-libs cause you error.

jason-mao, 2020-06-19:
@weilaideai2008 Please test with [latest master](code4bd5fa24a4f8afce801b95bb147fc7b46d0411cc). We have fixed this build error.

closed 2020-06-19 by jason-mao