espressif/esp-adf #401: speech_recognition asr example linking issue

jun66534, 2020-05-21:
The following error occurred while building the example.

/home/ajou/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/ajou/esp/esp-adf/components/esp-sr/lib/libwakenet.a(mfcc_runner_straight.o): in function `run_logfbank':
/home/sunxiangyu/esp/esp_sr_lib/components/wakenet/mfcc_runner_straight.c:70: undefined reference to `csf_preemphasis_step'
/home/ajou/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/sunxiangyu/esp/esp_sr_lib/components/wakenet/mfcc_runner_straight.c:72: undefined reference to `csf_logfbank_HTK_step'
/home/ajou/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/ajou/esp/esp-adf/components/esp-sr/lib/libwakenet.a(mfcc_runner_straight.o): in function `run':
/home/sunxiangyu/esp/esp_sr_lib/components/wakenet/mfcc_runner_straight.c:60: undefined reference to `csf_mfcc_HTK'
collect2: error: ld returned 1 exit status
CMakeFiles/speech_recognition_example.elf.dir/build.make:451: recipe for target 'speech_recognition_example.elf' failed
make[2]: *** [speech_recognition_example.elf] Error 1
CMakeFiles/Makefile2:2358: recipe for target 'CMakeFiles/speech_recognition_example.elf.dir/all' failed
make[1]: *** [CMakeFiles/speech_recognition_example.elf.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
make failed with exit code 2

I think there was an error in the linking process because there was a message like below before.

Scanning dependencies of target speech_recognition_example.elf
[ 99%] Building C object CMakeFiles/speech_recognition_example.elf.dir/project_elf_src.c.obj
[100%] Linking CXX executable speech_recognition_example.elf

Can anyone help me?

jun66534, 2020-05-21:
[](url)

espressif-zhanghu, 2020-05-22:
@jun66534  what is your IDF id ?

jun66534, 2020-05-22:
@espressif-zhanghu 
IDF id? I can't guess. I use LyraT Mini v1.1 in menuconfig

espressif-zhanghu, 2020-05-25:
@jun66534   
you can use **git submodule update --init** in path (esp-adf). and id is shown in more log.

closed 2020-10-15 by espressif-bot

jalakamkiranadspl, 2021-07-26:
I am getting this same issue. Can you please tell me the work around. My esp-idf version is 4.3