espressif/esp-adf #402: cli播放在线mp3的时候遭遇崩溃

coollofty, 2020-05-22:
以下是examples/cli程序崩溃时打出的Log。播放的是我本机在线的URL，这个MP3是用Nginx+PHP推送的，编码格式是chunked。在浏览器中，或其它PC平台的播放器上，播放相同的URL是没有任何问题的。但是esp32就会崩溃，而且是不论重复操作多少次都100%崩溃。请问这样的问题如何解决呢？
`
I (7487) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5
I (7497) ESP_AUDIO_TASK: It's a decoder.URI:http://192.168.86.136/test.php

I (7547) ESP_AUDIO_CTRL: Func:media_ctrl_play, Line:405, MEM Total:3998208 Bytes, Inter:49744 Bytes, Dram:39240 Bytes

I (7557) ESP_AUDIO_TASK: 1.IN:[http],CODEC:[auto],FILTER:[NULL],OUT:[iis],RESAMPLE:[0]
I (7567) ESP_AUDIO_TASK: 2.Handles,IN:0x3f808358,CODEC:0x3f808738,FILTER:0x0,OUT:0x3f808474
I (7747) ESP_AUDIO_TASK: 2.1 Update all pipelines
W (7747) ESP_AUDIO_TASK: Destroy the old pipeline
I (7747) AUDIO_PIPELINE: audio_pipeline_unlinked
W (7747) AUDIO_PIPELINE: There are no listener registered
I (7757) ESP_AUDIO_TASK: Func:media_ctrl_task, Line:314, MEM Total:4019472 Bytes, Inter:50340 Bytes, Dram:39836 Bytes

I (7857) ESP_AUDIO_TASK: in_rb:0x3f82fe54,filter_rb:0x0,out_rb:0x3f8352b4
I (7867) ESP_AUDIO_TASK: 3. Previous starting...
I (7937) AUDIO_ELEMENT: [http] Element task created
I (7937) AUDIO_ELEMENT: [http] AEL_MSG_CMD_RESUME,state:1
I (7937) AUDIO_ELEMENT: [auto] Element task created
I (7997) AUDIO_ELEMENT: [auto] AEL_MSG_CMD_RESUME,state:1
I (7997) ESP_DECODER: Ready to do audio type check (line 121)
I (7997) AUDIO_ELEMENT: [iis] Element task created
I (8007) AUDIO_ELEMENT: [iis] AEL_MSG_CMD_RESUME,state:1
I (8017) I2S_STREAM: AUDIO_STREAM_WRITER
I (8187) ESP_DECODER: Detect audio type is mp3
I (8187) STAGEFRIGHTMP3_DECODER: MP3 opened
I (8387) STAGEFRIGHTMP3_DECODER: I2S setup: sample_rate = 44100 : channels = 2
I (8617) AUDIO_ELEMENT: [iis] AEL_MSG_CMD_PAUSE

CORRUPT HEAP: multi_heap.c:428 detected at 0x3f8352f0
abort() was called at PC 0x40096f83 on core 0
0x40096f83: multi_heap_assert at E:/esp32/esp-adf/esp-idf/components/heap/multi_heap.c:696

Backtrace: 0x40097867:0x3ffd0140 0x400979bf:0x3ffd0160 0x40096f83:0x3ffd0180 0x40097306:0x3ffd01a0 0x400842b4:0x3ffd01d0
 0x40084353:0x3ffd0200 0x4008ba7d:0x3ffd0250 0x4019b172:0x3ffd0270 0x400d323c:0x3ffd0290
0x40097867: invoke_abort at E:/esp32/esp-adf/esp-idf/components/esp32/panic.c:649

0x400979bf: abort at E:/esp32/esp-adf/esp-idf/components/esp32/panic.c:649

0x40096f83: multi_heap_assert at E:/esp32/esp-adf/esp-idf/components/heap/multi_heap.c:696

0x40097306: multi_heap_malloc_impl at E:/esp32/esp-adf/esp-idf/components/heap/multi_heap.c:696

0x400842b4: heap_caps_malloc at E:/esp32/esp-adf/esp-idf/components/heap/heap_caps.c:103

0x40084353: heap_caps_malloc_prefer at E:/esp32/esp-adf/esp-idf/components/heap/heap_caps.c:181 (discriminator 7)

0x4008ba7d: wifi_malloc at E:/esp32/esp-adf/esp-idf/components/esp32/wifi_os_adapter.c:61

0x4019b172: lmac_stop_hw_txq at ??:?

0x400d323c: timer_process_alarm at E:/esp32/esp-adf/esp-idf/components/esp32/esp_timer.c:284
 (inlined by) timer_task at E:/esp32/esp-adf/esp-idf/components/esp32/esp_timer.c:311
`

espressif-zhanghu, 2020-05-22:
@coollofty   your url is " http://192.168.86.136/test.php " ？ is right?

coollofty, 2020-05-22:
> @coollofty your url is " http://192.168.86.136/test.php " ？ is right?

yes, is right, it is my local service, I tested with chrome & other players

closed 2020-05-22 by coollofty

espressif-zhanghu, 2020-05-25:
I can see more log? 
and what is the ID of components/esp-adf-libs ? 915fe10 ?

espressif-zhanghu, 2020-05-25:
@coollofty   
I can see more log?
and what is the ID of components/esp-adf-libs ? 915fe10 ?