espressif/esp-adf #404: rec_engine_data_read 返回值异常？

junbowu, 2020-05-24:
在通一次语音唤醒，读取数据时，当	rec_engine_data_read 返回值 翻转后(比如之前是连续的1~70)，然后下一次变成了0，此时其实并没有人说话(或者发出声音)rec_engine_data_read 的返回值会一直增加。这是一个异常么？
Line 1: 22:05:20.639] [0;32mI (85264) DUEROS: Recv Que DUER_CMD_START[0m
	Line 2: [22:05:20.648] [0;32mI (85268) DUEROS: index = 1
	Line 4: [22:05:20.648] [0;32mI (85271) DUEROS: index = 2
	Line 6: [22:05:20.659] [0;32mI (85275) DUEROS: index = 3
	Line 8: [22:05:20.659] [0;32mI (85279) DUEROS: index = 4
	Line 10: [22:05:20.659] [0;32mI (85282) DUEROS: index = 5
	Line 12: [22:05:20.669] [0;32mI (85286) DUEROS: index = 6
	Line 14: [22:05:20.669] [0;32mI (85289) DUEROS: index = 1
	Line 16: [22:05:20.669] [0;32mI (85357) DUEROS: index = 2
	Line 18: [22:05:20.726] [0;32mI (85445) DUEROS: index = 3
	Line 20: [22:05:20.815] [0;32mI (85533) DUEROS: index = 4
	Line 22: [22:05:20.903] [0;32mI (85626) DUEROS: index = 5
	Line 24: [22:05:20.995] [0;32mI (85714) DUEROS: index = 6
	Line 27: [22:05:21.176] [0;32mI (85808) DUEROS: index = 7
	Line 29: [22:05:21.176] [0;32mI (85895) DUEROS: index = 8
	Line 31: [22:05:21.264] [0;32mI (85989) DUEROS: index = 9
	Line 33: [22:05:21.357] [0;32mI (86076) DUEROS: index = 10
	Line 35: [22:05:21.445] [0;32mI (86170) DUEROS: index = 11
	Line 37: [22:05:21.539] [0;32mI (86258) DUEROS: index = 12
	Line 40: [22:05:21.708] [0;32mI (86339) DUEROS: index = 13
	Line 42: [22:05:21.708] [0;32mI (86439) DUEROS: index = 14
	Line 44: [22:05:21.808] [0;32mI (86520) DUEROS: index = 15
	Line 46: [22:05:21.889] [0;32mI (86620) DUEROS: index = 16
	Line 48: [22:05:21.989] [0;32mI (86701) DUEROS: index = 17
	Line 51: [22:05:22.180] [0;32mI (86801) DUEROS: rec_engine_cb - REC_EVENT_VAD_STOP, state:4[0m
	Line 52: [22:05:22.180] [0;32mI (86803) DUEROS: index = 0
	Line 54: [22:05:22.187] [0;32mI (86806) DUEROS: Dueros DUER_CMD_STOP[0m
	Line 60: [22:05:27.209] [0;32mI (91834) DUEROS: rec_engine_cb - REC_EVENT_VAD_START[0m
	Line 61: [22:05:27.209] [0;32mI (91834) DUEROS: RECEIVE DUER_CMD_START
	Line 63: [22:05:27.209] [0;32mI (91834) DUEROS: Recv Que DUER_CMD_START[0m
	Line 64: [22:05:27.220] [0;32mI (91838) DUEROS: index = 60
	Line 66: [22:05:27.220] [0;32mI (91841) DUEROS: index = 61
	Line 68: [22:05:27.229] [0;32mI (91845) DUEROS: index = 62
	Line 70: [22:05:27.229] [0;32mI (91849) DUEROS: index = 63
	Line 72: [22:05:27.229] [0;32mI (91852) DUEROS: index = 64
	Line 74: [22:05:27.240] [0;32mI (91856) DUEROS: index = 65
	Line 76: [22:05:27.240] [0;32mI (91860) DUEROS: index = 66
	Line 78: [22:05:27.240] [0;32mI (91864) DUEROS: index = 67
	Line 80: [22:05:27.248] [0;32mI (91867) DUEROS: index = 68
	Line 82: [22:05:27.248] [0;32mI (91871) DUEROS: index = 69
	Line 84: [22:05:27.258] [0;32mI (91875) DUEROS: index = 70
	Line 86: [22:05:27.258] [0;32mI (91879) DUEROS: index = 1
	Line 88: [22:05:27.258] [0;32mI (91933) DUEROS: index = 2
	Line 90: [22:05:27.303] [0;32mI (92014) DUEROS: index = 3
	Line 92: [22:05:27.383] [0;32mI (92102) DUEROS: index = 4
	Line 94: [22:05:27.470] [0;32mI (92195) DUEROS: index = 5
	Line 96: [22:05:27.564] [0;32mI (92283) DUEROS: index = 6
	Line 99: [22:05:27.745] [0;32mI (92376) DUEROS: index = 7
	Line 101: [22:05:27.745] [0;32mI (92464) DUEROS: index = 8
	Line 103: [22:05:27.833] [0;32mI (92558) DUEROS: index = 9
	Line 105: [22:05:27.927] [0;32mI (92645) DUEROS: index = 10
	Line 107: [22:05:28.014] [0;32mI (92733) DUEROS: index = 11
	Line 109: [22:05:28.103] [0;32mI (92826) DUEROS: index = 12
	Line 112: [22:05:28.283] [0;32mI (92914) DUEROS: index = 13
	Line 114: [22:05:28.283] [0;32mI (93008) DUEROS: index = 14
	Line 116: [22:05:28.377] [0;32mI (93095) DUEROS: index = 15
	Line 118: [22:05:28.464] [0;32mI (93189) DUEROS: index = 16
	Line 120: [22:05:28.558] [0;32mI (93277) DUEROS: index = 17
	Line 122: [22:05:28.646] [0;32mI (93370) DUEROS: index = 18
	Line 125: [22:05:28.826] [0;32mI (93458) DUEROS: index = 19
	Line 127: [22:05:28.826] [0;32mI (93539) DUEROS: index = 20
	Line 129: [22:05:28.908] [0;32mI (93639) DUEROS: index = 21
	Line 131: [22:05:29.008] [0;32mI (93720) DUEROS: index = 22
	Line 133: [22:05:29.089] [0;32mI (93820) DUEROS: index = 23
	Line 135: [22:05:29.189] [0;32mI (93902) DUEROS: index = 24
	Line 138: [22:05:29.371] [0;32mI (94001) DUEROS: index = 25
	Line 140: [22:05:29.371] [0;32mI (94083) DUEROS: index = 26
	Line 142: [22:05:29.454] [0;32mI (94170) DUEROS: index = 27
	Line 144: [22:05:29.540] [0;32mI (94264) DUEROS: index = 28
[... 62 lines cut ...]
	Line 281: [22:05:35.301] [0;32mI (99933) DUEROS: index = 91
	Line 283: [22:05:35.303] [0;32mI (100027) DUEROS: index = 92
	Line 285: [22:05:35.397] [0;32mI (100114) DUEROS: index = 93
	Line 287: [22:05:35.484] [0;32mI (100208) DUEROS: index = 94
	Line 289: [22:05:35.578] [0;32mI (100295) DUEROS: index = 95
	Line 291: [22:05:35.665] [0;32mI (100389) DUEROS: index = 96
	Line 294: [22:05:35.847] [0;32mI (100476) DUEROS: index = 97
	Line 296: [22:05:35.847] [0;32mI (100570) DUEROS: index = 98
	Line 298: [22:05:35.940] [0;32mI (100658) DUEROS: index = 99
	Line 300: [22:05:36.027] [0;32mI (100739) DUEROS: index = 100
	Line 302: [22:05:36.109] [0;32mI (100839) DUEROS: index = 101
	Line 304: [22:05:36.210] [0;32mI (100920) DUEROS: index = 102
	Line 307: [22:05:36.390] [0;32mI (101020) DUEROS: index = 103
	Line 309: [22:05:36.390] [0;32mI (101101) DUEROS: index = 104
	Line 311: [22:05:36.472] [0;32mI (101201) DUEROS: index = 105
	Line 313: [22:05:36.572] [0;32mI (101283) DUEROS: index = 106
	Line 315: [22:05:36.653] [0;32mI (101370) DUEROS: index = 107
	Line 317: [22:05:36.740] [0;32mI (101464) DUEROS: index = 108
	Line 320: [22:05:36.921] [0;32mI (101551) DUEROS: index = 109
	Line 322: [22:05:36.921] [0;32mI (101645) DUEROS: index = 110
	Line 324: [22:05:37.015] [0;32mI (101733) DUEROS: index = 111
	Line 326: [22:05:37.102] [0;32mI (101826) DUEROS: index = 112
	Line 328: [22:05:37.196] [0;32mI (101914) DUEROS: index = 113
	Line 330: [22:05:37.284] [0;32mI (102008) DUEROS: index = 114
	Line 333: [22:05:37.465] [0;32mI (102095) DUEROS: index = 115

espressif-zhanghu, 2020-05-25:
唤醒除了语音还有服务器端调用唤醒。需确认是否服务器端唤醒。

junbowu, 2020-05-27:
测试的时候是把服务器是未连接的(和服务器端应该没什么关系)，只是为了测试唤醒和record，主要是record回来的数据的index看着不太对

espressif-zhanghu, 2020-05-27:
@junbowu 确认一下，有没有连接 dueros 服务器？因为没有连接服务器，下面的打印确实属于不正常。

junbowu, 2020-05-27:
没有连接服务器

junbowu, 2020-05-27:
我本意是想测试record这个模块，实现离线唤醒后，读取录音数据给baidu sr来做在线语音识别，用了dueros这个例子但dueros服务器并没有连接(相关的login和dueros状态修改了下，record返回来的数据大部分时间都是正常的，偶现上面提的问题，感觉是内部的cycle buffer read/write index乱了)

espressif-zhanghu, 2020-05-27:
@junbowu  如果只想使用 “唤醒”，读取音频数据，可以考虑参考 examples/speech_recognition/asr，因为这个可以看见源码。
从log 没法看出问题出在哪了，还有 index 只有在 唤醒结束 或者 VAD 检测结束会重新初始化。

junbowu, 2020-05-28:
好的，谢谢！

closed 2020-05-28 by junbowu