espressif/esp-adf #405: Split i2s_stream to two separate pipelines

Maobuff, 2020-05-25:
Is it possible with current adf api to split i2s_stream to two separete pipelines?
For example to record right and left channels into two different files?

espressif-zhanghu, 2020-05-26:
@Maobuff  what is your purpose? recorder or player? if recoder you can modify after i2s_write.

Maobuff, 2020-05-26:
recorder

espressif-zhanghu, 2020-05-26:
you can find ** i2s_read(i2s->config.i2s_port, buffer, len, &bytes_read, ticks_to_wait) ** after you split **buffer** to  two **buffer**  that are right and left data.

jason-mao, 2020-05-31:
@Maobuff I think you can implement as follow:
- Enable I2s_stream `multi_out_num` ;
- Create a raw_stream, which should be got copy record  data 
- Set raw_stream's ringbuffer  to I2S_stream multi-output as `audio_element_set_multi_output_ringbuf`, 
- Build up two pipelines by your requirements, then select  channel in pipeline

Maobuff, 2020-05-31:
@jason-mao How do I select channels in pipelines?

jason-mao, 2020-05-31:
@Maobuff If your i2s data have two channel, just copy all the left/right channel data to buffer, then fill to ringbuffer.

closed 2020-06-19 by jason-mao