espressif/esp-adf #410: ERROR: mbedtls_ssl_handshake returned -0x7f00

meltdown03, 2020-06-09:
**Board: ESP32-S2-Kaluga-1 v1.2
IDF: master branch 4.2-dev (IDF_PATCHES APPLIED SUCCESSFULLY)
Sketch: most examples (play_http_mp3 in this case)**
```
I (321) cpu_start: Pro cpu up.
I (325) cpu_start: Application information:
I (329) cpu_start: Project name:     play_http_mp3
I (335) cpu_start: App version:      v2.0-118-g5b32779-dirty
I (341) cpu_start: Compile time:     Jun  9 2020 18:19:50
I (347) cpu_start: ELF file SHA256:  ac91191d5026a1d1...
I (353) cpu_start: ESP-IDF:          v4.2-dev-1660-g7d7521367
```

I'm trying to get the Audio board to work with my new Kaluga-1. I've tried all the examples that are supposed to work, but almost all of the HTTP ones do not work and result in the following:

### play_http_mp3 example with default sdkconfig except Board and WiFi settings

```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe8100,len:0x4
load:0x3ffe8104,len:0x1938
load:0x40050000,len:0x18a8
load:0x40054000,len:0x210c
entry 0x40050314
I (49) boot: ESP-IDF v4.2-dev-1660-g7d7521367 2nd stage bootloader
I (49) boot: compile time 21:54:39
I (49) boot: chip revision: 0
I (53) qio_mode: Enabling default flash chip QIO
I (58) boot.esp32s2: SPI Speed      : 80MHz
I (63) boot.esp32s2: SPI Mode       : QIO
I (67) boot.esp32s2: SPI Flash Size : 4MB
I (72) boot: Enabling RNG early entropy source...
I (77) boot: Partition Table:
I (81) boot: ## Label            Usage          Type ST Offset   Length
I (88) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (96) boot:  1 phy_init         RF data          01 01 0000d000 00001000
I (103) boot:  2 factory          factory app      00 00 00010000 00300000
I (111) boot: End of partition table
I (115) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f000020 size=0x24598 (148888) map
I (151) esp_image: segment 1: paddr=0x000345c0 vaddr=0x3ffc9bc0 size=0x03358 ( 13144) load
I (154) esp_image: segment 2: paddr=0x00037920 vaddr=0x40024000 size=0x00404 (  1028) load
0x40024000: _WindowOverflow4 at /home/neal/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (158) esp_image: segment 3: paddr=0x00037d2c vaddr=0x40024404 size=0x082ec ( 33516) load
I (175) esp_image: segment 4: paddr=0x00040020 vaddr=0x40080020 size=0x98350 (623440) map
0x40080020: _stext at ??:?

I (288) esp_image: segment 5: paddr=0x000d8378 vaddr=0x4002c6f0 size=0x0d4cc ( 54476) load
0x4002c6f0: prvAddNewTaskToReadyList at /home/neal/esp/esp-idf/components/freertos/tasks.c:1168

I (313) boot: Loaded app from partition at offset 0x10000
I (313) boot: Disabling RNG early entropy source...
I (313) cache: Instruction cache        : size 8KB, 4Ways, cache line size 32Byte
I (321) cpu_start: Pro cpu up.
I (325) cpu_start: Application information:
I (329) cpu_start: Project name:     play_http_mp3
I (335) cpu_start: App version:      v2.0-118-g5b32779-dirty
I (341) cpu_start: Compile time:     Jun  9 2020 18:19:50
I (347) cpu_start: ELF file SHA256:  ac91191d5026a1d1...
I (353) cpu_start: ESP-IDF:          v4.2-dev-1660-g7d7521367
I (360) cpu_start: Single core mode
I (364) heap_init: Initializing. RAM available for dynamic allocation:
I (371) heap_init: At 3FFD2790 len 00029870 (166 KiB): DRAM
I (377) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (384) cpu_start: Pro cpu start user code
I (454) spi_flash: detected chip: generic
[... 173 lines cut ...]
I (2740) wifi:security type: 4, phy: bgn, rssi: -44
I (2740) wifi:pm start, type: 1

W (2750) PERIPH_WIFI: WiFi Event cb, Unhandle event_base:WIFI_EVENT, event_id:4
I (2760) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (3410) HTTP_MP3_EXAMPLE: [ 4 ] Set up  event listener
I (3410) HTTP_MP3_EXAMPLE: [4.1] Listening event from all elements of pipeline
I (3410) HTTP_MP3_EXAMPLE: [4.2] Listening event from peripherals
I (3420) HTTP_MP3_EXAMPLE: [ 5 ] Start audio_pipeline
E (3420) AUDIO_THREAD: Not found right xTaskCreateRestrictedPinnedToCore. Please apply the $ADF_PATH/idf_patches/idf_v3.3_freertos.patch first
E (3440) AUDIO_THREAD: Error creating RestrictedPinnedToCore http
E (3440) AUDIO_ELEMENT: [http] audio_thread_create failed
E (3450) AUDIO_THREAD: Not found right xTaskCreateRestrictedPinnedToCore. Please apply the $ADF_PATH/idf_patches/idf_v3.3_freertos.patch first
E (3460) AUDIO_THREAD: Error creating RestrictedPinnedToCore mp3
E (3470) AUDIO_ELEMENT: [mp3] audio_thread_create failed
W (3480) AUDIO_ELEMENT: [http] Element has not create when AUDIO_ELEMENT_RESUME
W (3480) AUDIO_ELEMENT: [mp3] Element has not create when AUDIO_ELEMENT_RESUME
E (3490) AUDIO_PIPELINE: audio_pipeline_resume failed
W (3500) AUDIO_PIPELINE: Without stop, st:6
W (3500) AUDIO_PIPELINE: Without wait stop, st:6
W (3510) AUDIO_ELEMENT: [http] Element has not create when AUDIO_ELEMENT_TERMINATE
W (3510) AUDIO_ELEMENT: [mp3] Element has not create when AUDIO_ELEMENT_TERMINATE
```

I applied the patches but stillget the message when SPIRAM is enabled. I've also tried many combinations of SPIRAM/Internal reserve settings.

meltdown03, 2020-06-09:
I cleaned the build environment and rebuilt with default sdkconfig again, now the error is different:
```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0xb (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe8100,len:0x4
load:0x3ffe8104,len:0x17f0
load:0x40050000,len:0x14b0
load:0x40054000,len:0x210c
entry 0x400502dc
I (49) boot: ESP-IDF v4.2-dev-1660-g7d7521367 2nd stage bootloader
I (49) boot: compile time 18:35:35
I (49) boot: chip revision: 0
I (53) boot.esp32s2: SPI Speed      : 80MHz
I (57) boot.esp32s2: SPI Mode       : DIO
I (62) boot.esp32s2: SPI Flash Size : 4MB
I (67) boot: Enabling RNG early entropy source...
I (72) boot: Partition Table:
I (76) boot: ## Label            Usage          Type ST Offset   Length
I (83) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (91) boot:  1 phy_init         RF data          01 01 0000d000 00001000
I (98) boot:  2 factory          factory app      00 00 00010000 00300000
I (106) boot: End of partition table
I (110) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f000020 size=0x24644 (149060) map
I (150) esp_image: segment 1: paddr=0x0003466c vaddr=0x3ffc9af0 size=0x03364 ( 13156) load
I (153) esp_image: segment 2: paddr=0x000379d8 vaddr=0x40024000 size=0x00404 (  1028) load
0x40024000: _WindowOverflow4 at /home/neal/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (157) esp_image: segment 3: paddr=0x00037de4 vaddr=0x40024404 size=0x08234 ( 33332) load
I (174) esp_image: segment 4: paddr=0x00040020 vaddr=0x40080020 size=0x986ac (624300) map
0x40080020: _stext at ??:?

I (305) esp_image: segment 5: paddr=0x000d86d4 vaddr=0x4002c638 size=0x0d4b0 ( 54448) load
0x4002c638: prvAddNewTaskToReadyList at /home/neal/esp/esp-idf/components/freertos/tasks.c:1147

I (331) boot: Loaded app from partition at offset 0x10000
I (331) boot: Disabling RNG early entropy source...
I (331) cache: Instruction cache        : size 8KB, 4Ways, cache line size 32Byte
I (339) cpu_start: Pro cpu up.
I (343) cpu_start: Application information:
I (347) cpu_start: Project name:     play_http_mp3
I (353) cpu_start: App version:      v2.0-118-g5b32779-dirty
I (359) cpu_start: Compile time:     Jun  9 2020 18:35:33
I (365) cpu_start: ELF file SHA256:  e50b8e6e741ae0ed...
I (371) cpu_start: ESP-IDF:          v4.2-dev-1660-g7d7521367
I (378) cpu_start: Single core mode
I (382) heap_init: Initializing. RAM available for dynamic allocation:
I (389) heap_init: At 3FFD26C8 len 00029938 (166 KiB): DRAM
I (395) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (402) cpu_start: Pro cpu start user code
I (472) spi_flash: detected chip: generic
I (473) spi_flash: flash io: dio
I (473) cpu_start: Starting scheduler on PRO CPU.
I (485) HTTP_MP3_EXAMPLE: [ 1 ] Start audio codec chip
I (515) HTTP_MP3_EXAMPLE: [2.0] Create audio pipeline for playback
I (515) HTTP_MP3_EXAMPLE: [2.1] Create http stream to read data
I (525) HTTP_MP3_EXAMPLE: [2.2] Create i2s stream to write data to codec chip
I (535) HTTP_MP3_EXAMPLE: [2.3] Create mp3 decoder to decode mp3 file
I (535) HTTP_MP3_EXAMPLE: [2.4] Register all elements to audio pipeline
I (545) HTTP_MP3_EXAMPLE: [2.5] Link it together http_stream-->mp3_decoder-->i2s_stream-->[codec_chip]
I (555) HTTP_MP3_EXAMPLE: [2.6] Set up  uri (http as http_stream, mp3 as mp3 decoder, and default output is i2s)
I (565) HTTP_MP3_EXAMPLE: [ 3 ] Start and wait for Wi-Fi network
I (585) wifi:wifi driver task: 3ffe4924, prio:23, stack:6656, core=0
I (585) wifi:wifi firmware version: 517093f
I (585) wifi:wifi certification version: v7.0
I (585) wifi:config NVS flash: enabled
I (585) wifi:config nano formating: disabled
I (595) wifi:Init dynamic tx buffer num: 32
[... 23 lines cut ...]
E (7015) MP3_DECODER: Allocate decoder buffer failed. bytes 27796, (line 268)
E (7015) AUDIO_ELEMENT: [mp3] AEL_STATUS_ERROR_OPEN,-1
W (7015) AUDIO_ELEMENT: [mp3] AEL_MSG_CMD_ERROR,6
W (7015) AUDIO_ELEMENT: IN-[i2s] AEL_IO_ABORT
W (7055) HTTP_MP3_EXAMPLE: [ * ] Stop event received
I (7055) HTTP_MP3_EXAMPLE: [ 6 ] Stop audio_pipeline
W (7055) AUDIO_ELEMENT: [mp3] Element already stopped
W (7065) AUDIO_ELEMENT: [i2s] Element already stopped
W (8245) AUDIO_PIPELINE: Without stop, st:1
W (8245) AUDIO_PIPELINE: Without wait stop, st:1
W (8245) AUDIO_PIPELINE: There are no listener registered
W (8245) AUDIO_ELEMENT: [http] Element has not create when AUDIO_ELEMENT_TERMINATE
W (8255) AUDIO_ELEMENT: [i2s] Element has not create when AUDIO_ELEMENT_TERMINATE
W (8265) AUDIO_ELEMENT: [mp3] Element has not create when AUDIO_ELEMENT_TERMINATE
I (8285) wifi:state: run -> init (0)
I (8285) wifi:pm stop, total sleep time: 3858753 us / 5763910 us

I (8285) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
W (8285) PERIPH_WIFI: Wi-Fi disconnected from SSID XXXXXXXX, auto-reconnect disabled, reconnect after 1000 ms
W (8295) PERIPH_WIFI: WiFi Event cb, Unhandle event_base:WIFI_EVENT, event_id:3
I (8305) wifi:flush txq
I (8305) wifi:stop sw txq
I (8315) wifi:lmac stop hw txq
I (8315) wifi:Deinit lldesc rx mblock:10
```

jason-mao, 2022-02-08:
> E (7015) MP3_DECODER: Allocate decoder buffer failed. bytes 27796, (line 268)

Please check the RAM is it still enough.

This topic has become inactive so I'm going to close the issue. Please reopen this if you have any questions or need any further assistance.

closed 2022-02-08 by jason-mao