espressif/esp-adf #412: BT(A2DP) restart crash

iloop2020, 2020-06-13:
If the BT service is start->destroy->start, it crash.

1. bluetooth_service_start()
2. bluetooth_service_destroy()
3. bluetooth_service_start()

Snap of code at esp-adf/components/bluetooth_service/bluetooth_service.c:437
ESP_ERROR_CHECK(esp_bt_controller_mem_release(ESP_BT_MODE_BLE));

Why it is ESP_BT_MODE_BLE ?

Thank you

iloop2020, 2020-10-16:
Hi Any update ?

iloop2020, 2020-12-02:
Hi

If ESP32 is connected with SmartPhone by A2DP sink, can esp32 terminate  the connection ?

Thank you.

jyh-cmd, 2022-04-25:
Hi,
I have the same problem on esp-idf v4.1 and esp-adf v2.3, is there a solution?
The following is the error log information.

`I (7521) BLUETOOTH_EXAMPLE: [ 1 ] Create Bluetooth service
ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x40091f8c
0x40091f8c: _esp_error_check_failed at /home/jiangyinhu/esp/esp-idf/components/esp32/panic.c:766

file: "/home/jiangyinhu/esp/esp-adf/components/bluetooth_service/bluetooth_service.c" line 487
func: bluetooth_service_start
expression: esp_bt_controller_mem_release(ESP_BT_MODE_BLE)

ELF file SHA256: 6abdd6b92f41b565

Backtrace: 0x400918ec:0x3ffbd350 0x40091f8f:0x3ffbd370 0x400db47b:0x3ffbd390 0x400d7098:0x3ffbd400 0x400d74da:0x3ffbd520 0x400932fd:0x3ffbd540
0x400918ec: invoke_abort at /home/jiangyinhu/esp/esp-idf/components/esp32/panic.c:157

0x40091f8f: _esp_error_check_failed at /home/jiangyinhu/esp/esp-idf/components/esp32/panic.c:767

0x400db47b: bluetooth_service_start at /home/jiangyinhu/esp/esp-adf/components/bluetooth_service/bluetooth_service.c:487 (discriminator 1)

0x400d7098: pipeline_bt_sink_start at /home/jiangyinhu/esp/esp32-mitu2-RevA/pipeline_bt_sink_2/build/../main/play_bt_music_example.c:86

0x400d74da: bt_switch_task at /home/jiangyinhu/esp/esp32-mitu2-RevA/pipeline_bt_sink_2/build/../main/play_bt_music_example.c:45

0x400932fd: vPortTaskWrapper at /home/jiangyinhu/esp/esp-idf/components/freertos/port.c:143

Rebooting...
ets Jun  8 2016 00:22:57
`

HengYongChao, 2022-05-09:
Hi @jyh-cmd  @iloop2020 

I suggest you to use pipeline stream way to control bluetooth. Use pipeline run/stop, pause/resume.

The reference routine is this, https://github.com/espressif/esp-adf/tree/master/examples/player/pipeline_a2dp_sink_stream