espressif/esp-adf #417: error: patch failed:

ZYQ-FEIYUE, 2020-07-01:
$ git apply ./idf_v3.3_freertos.patch
error: patch failed: components/freertos/include/freertos/task.h:674
error: components/freertos/include/freertos/task.h: patch does not apply
error: patch failed: components/freertos/tasks.c:756
error: components/freertos/tasks.c: patch does not apply

打补丁失败，我应该怎么打补丁
patch failed, how can I do it

jason-mao, 2020-07-02:
@ZYQ-FEIYUE 
参考如下命令：
`cd $IDF_PATH`
`git apply $ADF_PATH/idf_patches/idf_v3.3_freertos.patch`

closed 2020-07-02 by jason-mao

alibukharai, 2022-08-04:
I am facing the same issue don't know why the patch is not applied. Can you help me to this problem I am new here

Thanks 

[image]