espressif/esp-adf #419: amazon/google example

tcpipchip, 2020-07-02:
Hi,

Do you intend to import this example to this github ?

https://github.com/espressif/esp-va-sdk

Then will be complete!

vikramdattu, 2020-07-08:
@tcpipchip there are no immediate plans to move `esp-va-sdk` to ADF.

Do you have specific requirement for this?

tcpipchip, 2020-07-08:
@vikramdattu 
no more!

**Btw, some more news coming to ESP-VA-SDK ?**

Amazon and Google working assistant working.

Sometimes the GOOGLE crash. But good for tests.

Think about WAKEUP option to GOOGLE.

Thanks to answer, i thought that this esp-va-sdk discussion was dead

vikramdattu, 2020-07-09:
Cannot say anything concrete, but we do keep updating for features/bug fixes.

Can you please share logs for GVA crash?

You can use `Alexa` WW for Google.
There are no immediate plans to update it with different WW.

tcpipchip, 2020-07-09:
@vikramdattu 

Good morning!

Nice that there are updates!

Not got more error! Maybe a power supply problem!

Thanks to info about use Alexa WW.

Someday i talked Alexa and the system pop up and i hadnt understand :)

Is it possible use the Alexa/Google to only form the Phrase and not process it ?
For example, i say Turn On the Lamp, but i dont want the Alexa/Google process this order, just back me the text ?

tcpipchip, 2020-07-09:
ops, got a error
[image]
But still answer to voice! This good!

closed 2021-09-26 by jason-mao