espressif/esp-adf #420: Cannot build Audio HAL = CONFIG_AUDIO_BOARD_CUSTOM

tatsutaigu, 2020-07-03:
Hi,

## Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROVER-B
- ADF path: 
```
$ printenv ADF_PATH
/home/tatsutaigu/esp/esp-adf-v2.0/
```
- ADF version:
```
$cd $ADF_PATH
git describe --tags
v2.0
```

- IDF path: 
```
$ printenv IDF_PATH
/home/tatsutaigu/esp/esp-adf-v2.0/esp-idf 
```
- IDF version:
```
$ cd $IDF_PATH
$ git describe --tags
v3.3.1-211-g6f9a7264c
``` 

- Build System:         idf.py
- Compiler version :
```
$ xtensa-esp32-elf-gcc --version
xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
```

- Operating System:     WSL (Windows Subsystem on Linux)
- Using an IDE?: No

## Problem Description
I select AudioHAL = CONFIG_AUDIO_BOARD_CUSTOM on menuconfig.
Then, run `idf.py build`, but I get error as below.
```
-- IDF_VER: v3.3.1-211-g6f9a7264c
-- Project is not inside a git repository,                         will not use 'git describe' to determine PROJECT_VER.
-- Project version: 1
-- Adding linker script /home/tatsutaigu/esp/adf_pj/play_mp3/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script /home/tatsutaigu/esp/esp-adf-v2.0/esp-idf/components/esp32/ld/esp32.rom.ld
-- Adding linker script /home/tatsutaigu/esp/esp-adf-v2.0/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script /home/tatsutaigu/esp/esp-adf-v2.0/esp-idf/components/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/tatsutaigu/esp/esp-adf-v2.0/esp-idf/components/esp32/ld/esp32.rom.spiram_incompatible_fns.ld
CMake Error: Target idf_component_audio_board has dependency information when it shouldn't.
Your cache is probably stale. Please remove the entry
  idf_component_audio_board_LIB_DEPENDS
from the cache.
-- Current board name is CONFIG_AUDIO_BOARD_CUSTOM
-- Building empty aws_iot component due to configuration
-- Component libraries:
-- Configuring incomplete, errors occurred!
See also "/home/tatsutaigu/esp/adf_pj/play_mp3/build/CMakeFiles/CMakeOutput.log".
See also "/home/tatsutaigu/esp/adf_pj/play_mp3/build/CMakeFiles/CMakeError.log".
FAILED: build.ninja
/usr/bin/cmake -H/home/tatsutaigu/esp/adf_pj/play_mp3 -B/home/tatsutaigu/esp/adf_pj/play_mp3/build
ninja: error: rebuilding 'build.ninja': subcommand failed
ninja failed with exit code 1 
```
And, I could run `idf.py menuconfig` with same as above message.

How do I resolve it?

zloiia, 2020-08-19:
Still no solution?

zloiia, 2020-08-19:
Try to manually delete `build` directory in project folder and then run 
`idf.py menuconfig`

jason-mao, 2020-09-30:
@tatsutaigu Did you solved this issue?

closed 2020-09-30 by tatsutaigu