espressif/esp-adf #421: A2DP Source Stream example doesn't compile on IDF 4.1

howroyd, 2020-07-06:
Two lines in the Pipeline A2DP Source Stream example causing it not to compile:

First one, the third parameter is one byte longer than the destination size, is the `+1` an error?
`memcpy(&remote_bt_device_name, "ESP_SINK_STREAM_DEMO", ESP_BT_GAP_MAX_BDNAME_LEN + 1);`

Second one, I assume this is the old use of the function as it now takes two parameters?
`esp_bt_gap_set_scan_mode(ESP_BT_SCAN_MODE_CONNECTABLE_DISCOVERABLE);`

So I changed it to
`esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, ESP_BT_GENERAL_DISCOVERABLE);`

The audio is pretty bad.  Turns out, that the ESP is still trying to do scans even though it is connected to a device, which is causing the audio to be choppy.  It looks to me like there are two tasks running and fighting each other for the Bluetooth??
```
I (1616) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1626) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1666) BLUETOOTH_SOURCE_EXAMPLE: [ 1 ] Mount sdcard
I (2166) BLUETOOTH_SOURCE_EXAMPLE: [ 2 ] Start codec chip
W (2186) I2C_BUS: i2c_bus_create:57: I2C bus has been already created, [port:0]
I (2196) BLUETOOTH_SOURCE_EXAMPLE: [3.0] Create audio pipeline for playback
I (2196) BLUETOOTH_SOURCE_EXAMPLE: [3.1] Create fatfs stream to read data from sdcard
I (2206) BLUETOOTH_SOURCE_EXAMPLE: [3.2] Create mp3 decoder to decode mp3 file
I (2206) BLUETOOTH_SOURCE_EXAMPLE: [3.3] Create Bluetooth stream
I (2316) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (2686) BLUETOOTH_SOURCE_EXAMPLE: [3.4] Register all elements to audio pipeline
I (2686) BLUETOOTH_SOURCE_EXAMPLE: [3.5] Link it together [sdcard]-->fatfs_stream-->mp3_decoder-->bt_stream-->[bt sink]
I (2696) BLUETOOTH_SOURCE_EXAMPLE: Discovery started.
I (2696) BLUETOOTH_SOURCE_EXAMPLE: [3.6] Set up  uri (file as fatfs_stream, mp3 as mp3 decoder, and default output is i2s)
I (2716) BLUETOOTH_SOURCE_EXAMPLE: [3.7] Create bt peripheral
I (2716) BLUETOOTH_SOURCE_EXAMPLE: [3.8] Start bt peripheral
I (2726) BLUETOOTH_SOURCE_EXAMPLE: [ 4 ] Set up  event listener
I (2736) BLUETOOTH_SOURCE_EXAMPLE: [4.1] Listening event from all elements of pipeline
I (2736) BLUETOOTH_SOURCE_EXAMPLE: [4.2] Listening event from peripherals
I (2746) BLUETOOTH_SOURCE_EXAMPLE: [ 5 ] Start audio_pipeline
I (2756) BLUETOOTH_SOURCE_EXAMPLE: [ 6 ] Listen for all pipeline events
I (2806) BLUETOOTH_SOURCE_EXAMPLE: [ * ] Receive music info from mp3 decoder, sample_rates=44100, bits=16, ch=2
I (15516) BLUETOOTH_SOURCE_EXAMPLE: Device discovery failed, continue to discover...
I (15516) BLUETOOTH_SOURCE_EXAMPLE: Device discovery failed, continue to discover...
I (15516) BLUETOOTH_SOURCE_EXAMPLE: Discovery started.
E (15886) BT_APPL: bta_av_rc_create ACP handle exist for shdl:0
W (17956) BT_SDP: SDP - no spare CCB for orig

W (18226) BT_APPL: new conn_srvc id:18, app_id:0
W (18226) BT_APPL: new conn_srvc id:18, app_id:1
I (28326) BLUETOOTH_SOURCE_EXAMPLE: Device discovery failed, continue to discover...
I (28326) BLUETOOTH_SOURCE_EXAMPLE: Device discovery failed, continue to discover...
W (28326) BT_APPL: [0]: state:1, info:x30, avoid_rs 0
W (28686) BT_APPL: bta_dm_rs_cback:512
E (28686) BT_BTM: tBTM_SEC_DEV:0x3f80811c rs_disc_pending=1

I (28686) BLUETOOTH_SOURCE_EXAMPLE: Discovery started.
I (41486) BLUETOOTH_SOURCE_EXAMPLE: Device discovery failed, continue to discover...
I (41486) BLUETOOTH_SOURCE_EXAMPLE: Device discovery failed, continue to discover...
W (41496) BT_APPL: [0]: state:1, info:x30, avoid_rs 0
I (41506) BLUETOOTH_SOURCE_EXAMPLE: Discovery started.
I (54316) BLUETOOTH_SOURCE_EXAMPLE: Device discovery failed, continue to discover...
I (54316) BLUETOOTH_SOURCE_EXAMPLE: Device discovery failed, continue to discover...
W (54326) BT_APPL: [0]: state:1, info:x30, avoid_rs 0
I (54336) BLUETOOTH_SOURCE_EXAMPLE: Discovery started.
I (67146) BLUETOOTH_SOURCE_EXAMPLE: Device discovery failed, continue to discover...
I (67156) BLUETOOTH_SOURCE_EXAMPLE: Device discovery failed, continue to discover...
W (67156) BT_APPL: [0]: state:1, info:x30, avoid_rs 0
I (67166) BLUETOOTH_SOURCE_EXAMPLE: Discovery started.
I (79986) BLUETOOTH_SOURCE_EXAMPLE: Device discovery failed, continue to discover...
I (79986) BLUETOOTH_SOURCE_EXAMPLE: Device discovery failed, continue to discover...
W (79996) BT_APPL: [0]: state:1, info:x30, avoid_rs 0
I (79996) BLUETOOTH_SOURCE_EXAMPLE: Discovery started.
I (92826) BLUETOOTH_SOURCE_EXAMPLE: Device discovery failed, continue to discover...
I (92826) BLUETOOTH_SOURCE_EXAMPLE: Device discovery failed, continue to discover...
W (92836) BT_APPL: [0]: state:1, info:x30, avoid_rs 0
I (92846) BLUETOOTH_SOURCE_EXAMPLE: Discovery started.
I (105656) BLUETOOTH_SOURCE_EXAMPLE: Device discovery failed, continue to discover...
I (105666) BLUETOOTH_SOURCE_EXAMPLE: Device discovery failed, continue to discover...
W (105666) BT_APPL: [0]: state:1, info:x30, avoid_rs 0
```

I put a simple global boolean to cancel any scan/discovery requests once a device has been found just to test my theory and the audio is now excellent.

Main thing I want to check, as I understand the ADF is still very much in development.  Have I caused the issue through something I have done wrong, or is it purely because the ADF is still new and buggy?  If the latter, then if I have time I am more than happy to do some pull requests to help fix it all up, but want to make sure I am not at fault first!!

Cheers

andrey0303, 2020-07-09:
Hi, I have the same problem.

For the first error, I replaced 
`typedef uint8_t esp_peer_bdname_t[ESP_BT_GAP_MAX_BDNAME_LEN + 1];`
with 
`typedef uint8_t esp_peer_bdname_t[ESP_BT_GAP_MAX_BDNAME_LEN + 2];`

For the second error, I replaced
`#include "esp_gap_bt_api.h"`
with 
`#include %ADF_PATH%\esp-idf\components\bt\bluedroid\api\include\api\esp_gap_bt_api.h"`

But I haven’t received a sound yet, there are other errors.

kylongmu, 2020-09-29:
I got the same problem.
Till now, without any official fix, that's too bad!

jason-mao, 2020-09-29:
@howroyd Thanks for you report. Related issue is https://github.com/espressif/esp-adf/issues/466

closed 2020-09-30 by espressif-bot