espressif/esp-adf #423: error: patch failed: idf_v3.3_freertos.patch

andrey0303, 2020-07-08:
C:\My_Programming\esp-idf-v4.0.1>git apply C:\My_Programming\esp-adf\idf_patches\idf_v3.3_freertos.patch
C:/My_Programming/esp-adf/idf_patches/idf_v3.3_freertos.patch:12: trailing whitespace.
 * xTaskCreateRestrictedPinnedToCore() should only be used in systems that
error: patch failed: components/freertos/include/freertos/task.h:674
error: components/freertos/include/freertos/task.h: patch does not apply
error: patch failed: components/freertos/tasks.c:756
error: components/freertos/tasks.c: patch does not apply

C:\My_Programming\esp-idf-v4.0.1>git config apply.whitespace nowarn

C:\My_Programming\esp-idf-v4.0.1>git apply C:\My_Programming\esp-adf\idf_patches\idf_v3.3_freertos.patch
error: patch failed: components/freertos/include/freertos/task.h:674
error: components/freertos/include/freertos/task.h: patch does not apply
error: patch failed: components/freertos/tasks.c:756
error: components/freertos/tasks.c: patch does not apply

vikramdattu, 2020-07-08:
Patch is meant for if you are using `release/v3.3` branch of IDF.

Not needed for `v4.0`.

shootao, 2020-07-08:
Hi @andrey0303 
 This patch can be put on v4.0.1. Could you confirm that you have modified `tasks. c`?

andrey0303, 2020-07-08:
I am trying to run the example "pipeline_a2dp_source_stream" and get errors
E (2721) AUDIO_THREAD: Not found right xTaskCreateRestrictedPinnedToCore. Please apply the $ ADF_PATH / idf_patches / idf_v3.3_freertos.patch first
E (2741) AUDIO_THREAD: Error creating RestrictedPinnedToCore mp3
E (2741) AUDIO_ELEMENT: [mp3] audio_thread_create failed

I did not modified tasks. c

snailwft, 2020-07-09:
Hi, I have the same problem.

shootao, 2020-07-09:
Hi @andrey0303 & @snailwft 
I have no problem testing in both the Ubuntu environment and the Win32 environment,
Can you confirm your ADF and IDF commit ID?

andrey0303, 2020-07-09:
> Hi @andrey0303 & @snailwft
> I have no problem testing in both the Ubuntu environment and the Win32 environment,
> Can you confirm your ADF and IDF commit ID?

IDF:
```
C:\My_Programming\esp-idf-v4.0.1>git log
commit 4c81978a3e2220674a432a588292a4c860eef27b (HEAD, tag: v4.0.1, origin/release/v4.0)
Merge: fc2199c5d 6153026f2
Author: Angus Gratton <angus@espressif.com>
Date:   Wed May 20 13:17:23 2020 +0800

    Merge branch 'bugfix/doc_version_links_v4.0' into 'release/v4.0'

    doc: Fix broken Release page links when building a stable version (v4.0)

    See merge request espressif/esp-idf!8821

commit 6153026f22b47ef76051ee98a31be2a358d00a04
Author: Angus Gratton <angus@espressif.com>
Date:   Wed May 20 10:50:44 2020 +1000

    doc: Fix broken Release page links when building a stable version

commit fc2199c5d6175ddf7b494b30509c016dd4537818
Merge: 4695eedde 748720a46
Author: Jiang Jiang Jian <jack@espressif.com>
Date:   Sun May 17 17:05:15 2020 +0800

    Merge branch 'bugfix/wpa3_mixed_mode_authype_v4.0' into 'release/v4.0'

    esp_wifi: Add and utilize WPA3/WPA2 Mixed mode Authtype (backport v4.0)

    See merge request espressif/esp-idf!8768
```

ADF:
```

C:\My_Programming\esp-adf>git log -2
commit a4b5ba8e2cf21acab9317a1bd053c888b952e046 (HEAD -> master, origin/master, origin/HEAD)
Merge: 45a7914 b7855e8
Author: Jason-mao <maojianxin@espressif.com>
Date:   Wed Jun 24 19:45:56 2020 +0800

    Merge branch 'feature/update_1906_firmware' into 'master'

    update 1906  firmware

    See merge request adf/esp-adf-internal!571

commit b7855e8ee21f34c87c166f8a397719681badee10
Author: donglianghao <donglianghao@espressif.com>
Date:   Wed Jun 24 14:56:53 2020 +0800

    update firmware and readme
```

shootao, 2020-09-09:
HI @andrey0303 
I've tried this version of IDF and ADF, and they can be patched here. I'm confused
For questions that patch can't answer, you can refer to this   [git-patch-does-not-apply](https://stackoverflow.com/questions/4770177/git-patch-does-not-apply)

closed 2020-10-15 by jason-mao