espressif/esp-adf #424: pipeline_living_stream  can't play "http://yp.shoutcast.com/sbin/tunein-station.m3u?id=99473570"

snailwft, 2020-07-09:
Hi，I use 'example/player/pipeline_living_stream' to play this url "http://yp.shoutcast.com/sbin/tunein-station.m3u?id=99473570",  error occured, I see the log is that:
I (2323) wifi:state: assoc -> run (10)
I (2373) wifi:connected with Divoom-3g, aid = 3, channel 1, 40U, bssid = f4:83:cd:2c:e0:4c
I (2373) wifi:security: WPA2-PSK, phy: bgn, rssi: -47
I (2373) wifi:pm start, type: 1

W (2373) PERIPH_WIFI: WiFi Event cb, Unhandle event_base:WIFI_EVENT, event_id:4
I (2473) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (3103) PERIPH_WIFI: Got ip:10.0.0.171
I (3103) tcpip_adapter: sta ip: 10.0.0.171, mask: 255.255.255.0, gw: 10.0.0.1
I (3103) HTTP_LIVINGSTREAM_EXAMPLE: [ 4 ] Set up  event listener
I (3103) HTTP_LIVINGSTREAM_EXAMPLE: [4.1] Listening event from all elements of pipeline
I (3113) HTTP_LIVINGSTREAM_EXAMPLE: [4.2] Listening event from peripherals
I (3123) HTTP_LIVINGSTREAM_EXAMPLE: [ 5 ] Start audio_pipeline
I (3133) AUDIO_THREAD: The http task allocate stack on external memory
I (3133) AUDIO_ELEMENT: [http-0x3f800520] Element task created
I (3143) AUDIO_THREAD: The aac task allocate stack on external memory
I (3153) AUDIO_ELEMENT: [aac-0x3f800844] Element task created
I (3153) AUDIO_ELEMENT: [i2s-0x3f8006a8] Element task created
I (3163) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:358, MEM Total:4309860 Bytes, Inter:195024 Bytes, Dram:180896 Bytes

I (3173) AUDIO_ELEMENT: [http] AEL_MSG_CMD_RESUME,state:1
I (3183) AUDIO_ELEMENT: [aac] AEL_MSG_CMD_RESUME,state:1
I (3193) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:1
I (3193) I2S_STREAM: AUDIO_STREAM_WRITER
I (3203) AUDIO_PIPELINE: Pipeline started
I (3993) HTTP_STREAM: total_bytes=94
W (3993) HTTP_STREAM: No more data,errno:0, total_bytes:94, rlen = 0
I (3993) AUDIO_ELEMENT: IN-[http] AEL_IO_DONE,0
I (4003) AUDIO_ELEMENT: IN-[aac] AEL_IO_DONE,-2
I (4003) AAC_DECODER: new audio song
I (4013) AAC_DECODER: this audio is RAW AAC
E (4013) AAC_DECODER: Encountered error reading stream
E (4023) AUDIO_ELEMENT: [aac] AEL_STATUS_ERROR_OPEN,-1
W (4023) AUDIO_ELEMENT: IN-[i2s] AEL_IO_ABORT
W (4023) AUDIO_ELEMENT: [aac] AEL_MSG_CMD_ERROR,6
I (4033) AAC_DECODER: Closed by [6]

vikramdattu, 2020-07-09:
@snailwft the audio is mp3 audio. Whereas the example `pipeline_living_stream` tries it as aac audio, hence it's failing.

You can modify this example to use `mp3_decoder` or use other example (`cli` for instance) to be able to play this audio.

snailwft, 2020-07-09:
Hi, it's not mp3, it's m3u.

vikramdattu, 2020-07-09:
@snailwft yes.
That m3u contains, https://streamingv2.shoutcast.com/oroyawave which mp3 content.

snailwft, 2020-07-09:
Oh, amazing, how do you use this m3u to get  https://streamingv2.shoutcast.com/oroyawave.

vikramdattu, 2020-07-09:
> Oh, amazing, how do you use this m3u to get https://streamingv2.shoutcast.com/oroyawave.

You can use `curl` for the same.

e.g.:

`curl --verbose "http://yp.shoutcast.com/sbin/tunein-station.m3u?id=99473570"`

snailwft, 2020-07-09:
Oh, Thank you very much.

closed 2020-07-14 by snailwft