espressif/esp-adf #425: ESP32 LyraT  GPIO isr service already installed

Patsjemoe, 2020-07-13:
Hi,
Running the VOIP example on the LyraT, I get following errors : 
E (2745) gpio: gpio_install_isr_service(394): GPIO isr service already installed
E (3745) ES8388_DRIVER: es_write_reg error
How can this be solved ?
Thanks in advance
best regards
Ludo

andrey0303, 2020-07-15:
Hi, 
 I have the same problem  on the LyraT and A2DP Source Stream example:

E (2111) gpio: gpio_install_isr_service(411): GPIO isr service already installed

jason-mao, 2020-07-17:
@Patsjemoe Which board did you chose?

jason-mao, 2020-07-17:
@andrey0303 Did you got any issue with this error?

andrey0303, 2020-07-17:
@jason-mao 
Yes, this example doesn't work. Perhaps this is due not only to this error but also to other errors.

I found out that gpio_install_isr_service(ESP_INTR_FLAG_LEVEL1) is called first when initializing peripherals for the SD card.
Then gpio_install_isr_service (0) is called from the headphone_detect_init function and this causes an error.

Patsjemoe, 2020-07-17:
Board is LyraT

Patsjemoe, 2020-07-17:
And program runs, so seems not to impact, but want to understand where it comes from.

Patsjemoe, 2020-07-19:
How can I see where the isr is installed..?

closed 2020-07-22 by Patsjemoe