espressif/esp-adf #429: Crash during OGG icecast stream playback

mabuyev, 2020-07-18:
Crashes at the end of each track.

Icecast stream info:

audio_bitrate | 128000
audio_channels | 2
audio_info | samplerate=0;channels=0;quality=3%2e00
audio_samplerate | 44100

`Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400ec356  PS      : 0x00060a30  A0      : 0x800e9c0c  A1      : 0x3ffed8b0  
0x400ec356: oggpack_readinit at /Users/maojianxin/duros/esp-adf-libs-source-dev/esp_codec/esp-tremor/src/bitwise.c:70

A2      : 0x3ffe3070  A3      : 0x00000000  A4      : 0x00001bf2  A5      : 0x3ffdfdd0  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffed8a0  
A10     : 0x3ffe3070  A11     : 0x00000000  A12     : 0x0000000c  A13     : 0x3ffe307c  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000e  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  

ELF file SHA256: b86a20a7d83fa8a0

Backtrace: 0x400ec356:0x3ffed8b0 0x400e9c09:0x3ffed8d0 0x400e96ca:0x3ffed900 0x400e4be9:0x3ffed980 0x400e393f:0x3ffeda10 0x400e394d:0x3ffeda30 0x40114bdb:0x3ffeda50 0x40114d85:0x3ffeda80 0x40088456:0x3ffedab0
0x400ec356: oggpack_readinit at /Users/maojianxin/duros/esp-adf-libs-source-dev/esp_codec/esp-tremor/src/bitwise.c:70

0x400e9c09: vorbis_dsp_synthesis at /Users/maojianxin/duros/esp-adf-libs-source-dev/esp_codec/esp-tremor/src/dsp.c:185

0x400e96ca: _fetch_and_process_packet at /Users/maojianxin/duros/esp-adf-libs-source-dev/esp_codec/esp-tremor/src/vorbisfile.c:631
 (inlined by) ov_read at /Users/maojianxin/duros/esp-adf-libs-source-dev/esp_codec/esp-tremor/src/vorbisfile.c:1713

0x400e4be9: ogg_decoder_process at /Users/maojianxin/duros/esp-adf-libs-source/esp_processing/esp-wrapper/ogg_decoder.c:571

0x400e393f: esp_decoder_process at /Users/maojianxin/duros/esp-adf-libs-source/esp_processing/esp-wrapper/esp_decoder.c:187

0x400e394d: _esp_decoder_process at /Users/maojianxin/duros/esp-adf-libs-source/esp_processing/esp-wrapper/esp_decoder.c:221

0x40114bdb: audio_element_process_running at /home/maranoid/Documents/esp/esp-adf/components/audio_pipeline/audio_element.c:1275

0x40114d85: audio_element_task at /home/maranoid/Documents/esp/esp-adf/components/audio_pipeline/audio_element.c:1275

0x40088456: vPortTaskWrapper at /home/maranoid/Documents/esp/esp-adf/esp-idf/components/freertos/port.c:403

Rebooting...
ets Jun  8 2016 00:22:57
`

jason-mao, 2020-07-24:
@mabuyev Please help to provide your  version and commit ID of ADF and IDF.

mabuyev, 2020-07-24:
ADF v2.0
IDF v3.3.2

houhaiyan, 2020-07-24:
@mabuyev  Can you provide the whole log?

mabuyev, 2020-07-24:
@houhaiyan, here:
[cutecom.log](https://github.com/espressif/esp-adf/files/4971274/cutecom.log)

houhaiyan, 2020-07-27:
can you sent me your music? what steps did you take?

mabuyev, 2020-07-27:
[Phil Thornton - Equilibrium.ogg.zip](https://github.com/espressif/esp-adf/files/4982720/Phil.Thornton.-.Equilibrium.ogg.zip)
Put several ogg files (or several renamed copies) in the folder and stream to icecast with ices or ezstream. 
ADF version below 2.0 - crash at the end of the track
ADF version 2.0 and higher - playback finishes at the end of the track with error "no more data in the stream"
The same stream can be played in web browser/mplayer without any gaps.

houhaiyan, 2020-07-28:
@mabuyev  can you sent me  more musics your tested error?  I want to be more testing.

mabuyev, 2020-07-28:
@houhaiyan, providing more ogg files:
[Rudy Adrian - Ancestral Legacy.ogg.zip](https://github.com/espressif/esp-adf/files/4988392/Rudy.Adrian.-.Ancestral.Legacy.ogg.zip)
[David Helpling - Loss Of Words.ogg.zip](https://github.com/espressif/esp-adf/files/4988393/David.Helpling.-.Loss.Of.Words.ogg.zip)
Ices config: 
[ices.xml.zip](https://github.com/espressif/esp-adf/files/4988398/ices.xml.zip)

houhaiyan, 2020-08-06:
@mabuyev Thank you for your ogg files.  I can not reproduction the crash. The lastest branch of esp-adf-libs repaired a bug. please pull the lastest branch of of esp-adf-libs for testing

houhaiyan, 2020-08-12:
@mabuyev  Does the crash that you encountered reproduce in lastest rdp-adf-lib?

closed 2020-09-30 by jason-mao