espressif/esp-adf #434: Live stream(m3u8) decode error

xiruilin, 2020-07-26:
Hi,

When decode m3u8 stream, after playing a playlist, need `Fetching again...`,
`AAC_READER` got error, log is shown below:

```
I (22494) DCS_AUDIO: Playing audio, offset: 0 url: http://open.ls.qingting.fm/live/332/64k.m3u8?format=ts
I (22511) DISPATCHER: EXE IN, cmd type:2, index:4001, data:0x3f95d0e4, len:12
I (22521) DCS_WRAPPER: duer_dcs_action_audio_play, 0x3f95d0e4, �Ȗ?, 12
I (22532) ESP_AUDIO_CTRL: Enter play procedure, src:801
I (22544) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5, que:0x3ffe30c0
I (22552) ESP_AUDIO_CTRL: Func:_ctrl_play, Line:713, MEM Total:2567268 Bytes, Inter:89408 Bytes, Dram:75760 Bytes

I (22552) ESP_AUDIO_TASK: It's a decoder
I (22567) ESP_AUDIO_TASK: 1.CUR IN:[IN_http],CODEC:[DEC_m3u8],RESAMPLE:[48000],OUT:[OUT_iis],rate:0, ch:0
I (22577) ESP_AUDIO_TASK: Codec:aac, ext:DEC_m3u8
I (22583) ESP_AUDIO_TASK: 2.Handles,IN:0x3f958bb8,CODEC:0x3f95945c,FILTER:0x3f96c428,OUT:0x3f959f2c
I (22593) AUDIO_PIPELINE: audio_pipeline_unlinked
W (22598) AUDIO_PIPELINE: There are no listener registered
I (22604) ESP_AUDIO_TASK: 2.2 Update all pipeline
I (22609) ESP_AUDIO_TASK: 2.3 Linked new pipeline
I (22615) AUDIO_PIPELINE: link el->rb, el:0x3f958bb8, tag:IN_http, rb:0x3f96ab4c
I (22623) AUDIO_PIPELINE: link el->rb, el:0x3f95945c, tag:DEC_aac, rb:0x3f96c3b4
I (22631) AUDIO_PIPELINE: link el->rb, el:0x3f96c428, tag:resample, rb:0x3f96ccc0
I (22639) ESP_AUDIO_TASK: 3. Previous starting...
I (22644) AUDIO_THREAD: The IN_http task allocate stack on external memory
I (22652) AUDIO_ELEMENT: [IN_http-0x3f958bb8] Element task created
I (22658) AUDIO_ELEMENT: [IN_http] AEL_MSG_CMD_RESUME,state:1
I (22665) AUDIO_THREAD: The DEC_aac task allocate stack on external memory
I (22672) AUDIO_ELEMENT: [DEC_aac-0x3f95945c] Element task created
I (22679) AUDIO_ELEMENT: [DEC_aac] AEL_MSG_CMD_RESUME,state:1
I (22899) HTTP_STREAM: total_bytes=780
I (22902) HTTP_STREAM: Live stream URI. Need to be fetched again!
I (23084) HTTP_STREAM: total_bytes=71064
I (23086) AAC_DECODER: new audio song
I (23086) AAC_DECODER: this audio is TS AAC
W (23463) HTTP_STREAM: No more data,errno:0, total_bytes:71064, rlen = 0
I (23464) AUDIO_ELEMENT: IN-[IN_http] AEL_IO_DONE,0
I (23465) AUDIO_ELEMENT: [IN_http] AEL_MSG_CMD_RESUME,state:1
I (23537) HTTP_STREAM: total_bytes=72004
I (23576) RSP_FILTER: reset sample rate of source data : 24000, reset channel of source data : 2
I (23576) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:1, Available:5, que:0x3ffe30c0
I (23582) ESP_AUDIO_TASK: Received muisc info then send MEDIA_CTRL_EVT_PLAY
I (23590) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_PLAY, status:UNKNOWN, 0
I (23597) AUDIO_THREAD: The resample task allocate stack on external memory
I (23605) AUDIO_ELEMENT: [resample-0x3f96c428] Element task created
I (23611) AUDIO_ELEMENT: [resample] AEL_MSG_CMD_RESUME,state:1
I (23618) AUDIO_THREAD: The OUT_iis task allocate stack on external memory
I (23619) RSP_FILTER: sample rate of source data : 24000, channel of source data : 2, sample rate of destination data : 48000, channel of destination data : 2
I (23625) AUDIO_ELEMENT: [OUT_iis-0x3f959f2c] Element task created
I (23647) AUDIO_ELEMENT: [OUT_iis] AEL_MSG_CMD_RESUME,state:1
I (23653) I2S_STREAM: AUDIO_STREAM_WRITER
I (23678) ESP_AUDIO_TASK: ESP_AUDIO status is AEL_STATUS_STATE_RUNNING
I (23679) ESP_AUDIO_TASK: Func:media_ctrl_task, Line:822, MEM Total:2390648 Bytes, Inter:55704 Bytes, Dram:42056 Bytes

I (23686) DUEROS_CPP: ESP_AUDIO_CALLBACK_FUNC, st: 1, err: 0, src: 801
I (23686) ESP_AUDIO_CTRL: Exit play procedure, ret:0
I (23694) DUEROS_CPP: duer_dcs_audio_on_running
I (23700) DISPATCHER: EXE OUT,result type:2, index:4001, ret:0, data:0x0, len:0
I (23707) DISPATCHER: EXE IN, cmd type:2, index:300d, data:0x0, len:0
I (23722) DIS_ACTION: display_action_music_on
I (23728) DISPATCHER: EXE OUT,result type:2, index:300d, ret:0, data:0x0, len:0
I (23715) DCS_AUDIO: Audio ret:0
W (29421) HTTP_STREAM: No more data,errno:0, total_bytes:72004, rlen = 0
I (29422) AUDIO_ELEMENT: IN-[IN_http] AEL_IO_DONE,0
I (29424) AUDIO_ELEMENT: [IN_http] AEL_MSG_CMD_RESUME,state:1
I (29489) HTTP_STREAM: total_bytes=72380
W (36570) HTTP_STREAM: No more data,errno:0, total_bytes:72380, rlen = 0
I (36571) AUDIO_ELEMENT: IN-[IN_http] AEL_IO_DONE,0
I (36572) AUDIO_ELEMENT: [IN_http] AEL_MSG_CMD_RESUME,state:1
I (36666) HTTP_STREAM: total_bytes=73132
W (43774) HTTP_STREAM: No more data,errno:0, total_bytes:73132, rlen = 0
I (43775) AUDIO_ELEMENT: IN-[IN_http] AEL_IO_DONE,0
I (43777) AUDIO_ELEMENT: [IN_http] AEL_MSG_CMD_RESUME,state:1
I (43868) HTTP_STREAM: total_bytes=72568
W (50996) HTTP_STREAM: No more data,errno:0, total_bytes:72568, rlen = 0
I (50997) AUDIO_ELEMENT: IN-[IN_http] AEL_IO_DONE,0
I (50998) AUDIO_ELEMENT: [IN_http] AEL_MSG_CMD_RESUME,state:1
I (51160) HTTP_STREAM: total_bytes=73132
I (53707) DCS_AUDIO: Get player position 244772
W (58228) HTTP_STREAM: No more data,errno:0, total_bytes:73132, rlen = 0
I (58229) AUDIO_ELEMENT: IN-[IN_http] AEL_IO_DONE,0
I (58230) AUDIO_ELEMENT: [IN_http] AEL_MSG_CMD_RESUME,state:1
I (58480) HTTP_STREAM: total_bytes=73884
W (65542) HTTP_STREAM: No more data,errno:0, total_bytes:73884, rlen = 0
I (65543) AUDIO_ELEMENT: IN-[IN_http] AEL_IO_DONE,0
I (65544) AUDIO_ELEMENT: [IN_http] AEL_MSG_CMD_RESUME,state:1
I (65736) HTTP_STREAM: total_bytes=72568
W (72647) HTTP_STREAM: No more data,errno:0, total_bytes:72568, rlen = 0
I (72647) AUDIO_ELEMENT: IN-[IN_http] AEL_IO_DONE,0
I (72649) AUDIO_ELEMENT: [IN_http] AEL_MSG_CMD_RESUME,state:1
I (72655) HTTP_STREAM: Fetching again...
I (72794) HTTP_STREAM: total_bytes=780
W (72795) HLS_PLAYLIST: URI exist
W (72795) HLS_PLAYLIST: URI exist
I (72796) HTTP_STREAM: Live stream URI. Need to be fetched again!
I (72934) HTTP_STREAM: total_bytes=73132
sync_byte errorE (73680) AAC_READER: ts decoding meet with error
I (73981) AAC_DECODER: Closed by [2]
I (73981) ESP_AUDIO_TASK: Received last pos: 404773 bytes
I (74077) AUDIO_ELEMENT: IN-[resample] AEL_IO_DONE,-2
I (74137) AUDIO_ELEMENT: IN-[OUT_iis] AEL_IO_DONE,-2
I (74176) ESP_AUDIO_TASK: Received last time: 49578 ms
```

Is there any solution?
Thanks!

jason-mao, 2020-08-05:
@xiruilin Thanks for your report.

marrog86, 2020-08-05:
I also have Problems with Playlists for playing MP3. 
I solved it by commenting the line like shown, for getting an always correct total_bytes size. But i have no idea how this affects the seeks. 
` /*
    * Due to the total byte of content has been changed after seek, set info.total_bytes at beginning only.
    */
    int64_t cur_pos = esp_http_client_fetch_headers(http->client);
   // if (info.byte_pos <= 0)
    {
        info.total_bytes = cur_pos;
    }`

jason-mao, 2020-08-06:
@xiruilin I think we  provided IDF and ADF  patches  could be fix your issue as 
https://github.com/espressif/esp-adf/tree/master/idf_patches#3idf_v33_esp_http_clientpatch--adf_http_streampatch

xiruilin, 2020-08-07:
@jason-mao 
Thank you for your reply.

We have already made this patch, but the problem has not been solved.

The url ``http://open.ls.qingting.fm/live/332/64k.m3u8?format=ts`` used to play normally, but now there is a problem.
The url ``http://cnlive.cnr.cn/hls/bjyygb.m3u8`` plays normally all the time.

Problems may arise in the content of ``.ts`` files, ``64k.m3u8`` being continuous and ``bjyygb.m3u8`` being independent.

jason-mao, 2020-08-08:
@xiruilin I have test your URL, it is works fine. How is your play operation before error?  Did you pause /resume the URL?

marrog86, 2020-08-08:
The patches worked for me. Thanks.

xiruilin, 2020-08-08:
@jason-mao 
Hi,
``ESP-ADF`` examples ``pipeline_living_stream``,
if ``AAC_STREAM_URI "http://open.ls.qingting.fm/live/274/64k.m3u8?format=aac"``, plays normally
if ``AAC_STREAM_URI "http://open.ls.qingting.fm/live/274/64k.m3u8?format=ts"``, after playing a playlist, get the error:
```
 (51621) HTTP_STREAM: Fetching again...
 (51711) HTTP_STREAM: total_bytes=780
 (51711) HTTP_STREAM: Live stream URI. Need to be fetched again!
 (51771) HTTP_STREAM: total_bytes=71816
sync_byte error (53621) AAC_READER: ts decoding meet with error
 (53621) AAC_DECODER: aac_fill_buffer failed
```

closed 2020-08-20 by xiruilin