espressif/esp-adf #438: Linking error on all Kaluga-1 1.2 Examples

meltdown03, 2020-08-02:
I deleted and recloned the ESP-ADF repo, IDF is at latest Master commit. I am no longer able to compile any of example made for the Kaluga-1. I used to be able to. I'm setting the target to esp32s2, here is the error:
```
[1054/1057] Generating ld/esp32s2.project.ld
warning: the default selection ESP32_KORVO_DU1906_CODEC_TAS5805M (undefined) of <choice ESP32_KORVO_DU1906_DAC> (defined at /home/neal/esp/esp-adf/components/audio_board/Kconfig.projbuild:27) is not contained in the choice
warning: the default selection SR_RUN_WM6_CORE1 (undefined) of <choice SR_RUN_WN6_CORE> (defined at /home/neal/esp/esp-adf/components/esp-sr/Kconfig.projbuild:83) is not contained in the choice
warning: the default selection CONFIG_SR_MN1_MODEL_QUANT (undefined) of <choice SR_MN_MODEL_SEL> (defined at /home/neal/esp/esp-adf/components/esp-sr/Kconfig.projbuild:100) is not contained in the choice
[1056/1057] Linking CXX executable play_http_mp3.elf
FAILED: play_http_mp3.elf
: && /home/neal/esp/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-g++ -mlongcalls  CMakeFiles/play_http_mp3.elf.dir/project_elf_src.c.obj -o play_http_mp3.elf  esp-idf/xtensa/libxtensa.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/efuse/libefuse.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/app_update/libapp_update.a  esp-idf/esp_ipc/libesp_ipc.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/soc/libsoc.a  esp-idf/vfs/libvfs.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/tcpip_adapter/libtcpip_adapter.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/esp_event/libesp_event.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/lwip/liblwip.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/driver/libdriver.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/esp32s2/libesp32s2.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/app_trace/libapp_trace.a  esp-idf/asio/libasio.a  esp-idf/cbor/libcbor.a  esp-idf/coap/libcoap.a  esp-idf/console/libconsole.a  esp-idf/nghttp/libnghttp.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/mdns/libmdns.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/esp_serial_slave_link/libesp_serial_slave_link.a  esp-idf/esp_websocket_client/libesp_websocket_client.a  esp-idf/expat/libexpat.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/fatfs/libfatfs.a  esp-idf/freemodbus/libfreemodbus.a  esp-idf/jsmn/libjsmn.a  esp-idf/json/libjson.a  esp-idf/libsodium/liblibsodium.a  esp-idf/mqtt/libmqtt.a  esp-idf/openssl/libopenssl.a  esp-idf/perfmon/libperfmon.a  esp-idf/spiffs/libspiffs.a  esp-idf/ulp/libulp.a  esp-idf/unity/libunity.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/audio_sal/libaudio_sal.a  esp-idf/adf_utils/libadf_utils.a  esp-idf/esp-sr/libesp-sr.a  esp-idf/esp-adf-libs/libesp-adf-libs.a  esp-idf/audio_pipeline/libaudio_pipeline.a  esp-idf/display_service/libdisplay_service.a  esp-idf/esp_dispatcher/libesp_dispatcher.a  esp-idf/esp_peripherals/libesp_peripherals.a  esp-idf/audio_hal/libaudio_hal.a  esp-idf/audio_board/libaudio_board.a  esp-idf/audio_stream/libaudio_stream.a  esp-idf/battery_service/libbattery_service.a  esp-idf/bluetooth_service/libbluetooth_service.a  esp-idf/coredump_upload_service/libcoredump_upload_service.a  esp-idf/dueros_service/libdueros_service.a  esp-idf/wifi_service/libwifi_service.a  esp-idf/esp_actions/libesp_actions.a  esp-idf/esp_event_cast/libesp_event_cast.a  esp-idf/input_key_service/libinput_key_service.a  esp-idf/ota_service/libota_service.a  esp-idf/playlist/libplaylist.a  esp-idf/main/libmain.a  -Wl,--cref -Wl,--Map=/home/neal/esp/esp-adf/examples/player/pipeline_http_mp3/build/play_http_mp3.map  -fno-rtti  -fno-lto  esp-idf/asio/libasio.a  esp-idf/cbor/libcbor.a  esp-idf/coap/libcoap.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/esp_serial_slave_link/libesp_serial_slave_link.a  esp-idf/esp_websocket_client/libesp_websocket_client.a  esp-idf/expat/libexpat.a  esp-idf/freemodbus/libfreemodbus.a  esp-idf/libsodium/liblibsodium.a  esp-idf/mqtt/libmqtt.a  esp-idf/openssl/libopenssl.a  esp-idf/perfmon/libperfmon.a  esp-idf/unity/libunity.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/protocomm/libprotocomm.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/mdns/libmdns.a  esp-idf/json/libjson.a  esp-idf/adf_utils/libadf_utils.a  esp-idf/jsmn/libjsmn.a  esp-idf/battery_service/libbattery_service.a  esp-idf/bluetooth_service/libbluetooth_service.a  esp-idf/coredump_upload_service/libcoredump_upload_service.a  esp-idf/esp_actions/libesp_actions.a  esp-idf/dueros_service/libdueros_service.a  -L/home/neal/esp/esp-adf/components/clouds/dueros/lightduer  -lduer-device-v4x  esp-idf/wifi_service/libwifi_service.a  -L/home/neal/esp/esp-adf/components/wifi_service/airkiss_config  -lairkiss_aes  esp-idf/esp_event_cast/libesp_event_cast.a  esp-idf/input_key_service/libinput_key_service.a  esp-idf/ota_service/libota_service.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/audio_stream/libaudio_stream.a  esp-idf/display_service/libdisplay_service.a  esp-idf/esp_peripherals/libesp_peripherals.a  esp-idf/audio_hal/libaudio_hal.a  esp-idf/audio_board/libaudio_board.a  esp-idf/display_service/libdisplay_service.a  esp-idf/esp_peripherals/libesp_peripherals.a  esp-idf/audio_hal/libaudio_hal.a  esp-idf/audio_board/libaudio_board.a  esp-idf/console/libconsole.a  esp-idf/fatfs/libfatfs.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/spiffs/libspiffs.a  esp-idf/esp-adf-libs/libesp-adf-libs.a  esp-idf/audio_pipeline/libaudio_pipeline.a  /home/neal/esp/esp-adf/components/esp-adf-libs/esp_audio/lib/esp32s2/libesp_audio.a  esp-idf/esp-adf-libs/libesp-adf-libs.a  esp-idf/audio_pipeline/libaudio_pipeline.a  /home/neal/esp/esp-adf/components/esp-adf-libs/esp_audio/lib/esp32s2/libesp_audio.a  esp-idf/esp-sr/libesp-sr.a  -L /home/neal/esp/esp-adf/components/esp-sr/lib  -L /home/neal/esp/esp-adf/components/esp-sr/wake_word_engine  -L /home/neal/esp/esp-adf/components/esp-sr/speech_command_recognition  -L /home/neal/esp/esp-adf/components/esp-sr/acoustic_algorithm  -ldl_lib  -lc_speech_features  -lwakenet  -lhilexin_wn3  -lhilexin_wn4  -lhilexin_wn5  -lhilexin_wn5X2  -lhilexin_wn5X3  -lhijeson_wn5X3  -lnihaoxiaozhi_wn5  -lnihaoxiaozhi_wn5X2  -lnihaoxiaozhi_wn5X3  -lnihaoxiaoxin_wn6  -lnihaoxiaoxin_wn5X3  -lcustomized_word_wn5  -lcustomized_word_wn6  -lmultinet  -lmultinet1_ch  -lmultinet1_en  -lesp_audio_processor  -L /home/neal/esp/esp-adf/components/esp-adf-libs/esp_audio/lib/esp32s2  -L /home/neal/esp/esp-adf/components/esp-adf-libs/esp_codec/lib/esp32s2  -L /home/neal/esp/esp-adf/components/esp-adf-libs/esp_ssdp/lib/esp32s2  -L /home/neal/esp/esp-adf/components/esp-adf-libs/esp_upnp/lib/esp32s2  -L /home/neal/esp/esp-adf/components/esp-adf-libs/esp_sip/lib/esp32s2  -L /home/neal/esp/esp-adf/components/esp-adf-libs/recorder_engine/lib/esp32s2  -Wl,--start-group  -lesp_processing  -lesp-aac  -lesp-amr  -lesp-amrwbenc  -lesp-flac  -lesp_ssdp  -lesp-ogg-container  -lesp-opus  -lesp-tremor  -lesp_sip  -lesp_upnp  -lesp-mp3  -lcodec-utils  -lesp-g711  -lrecorder_engine  -Wl,--end-group  -L/home/neal/esp/esp-adf/components/audio_hal/driver/zl38063/firmware  -lfirmware  esp-idf/esp_dispatcher/libesp_dispatcher.a  esp-idf/playlist/libplaylist.a  esp-idf/audio_sal/libaudio_sal.a  esp-idf/xtensa/libxtensa.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/efuse/libefuse.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/app_update/libapp_update.a  esp-idf/esp_ipc/libesp_ipc.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/soc/libsoc.a  esp-idf/vfs/libvfs.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/tcpip_adapter/libtcpip_adapter.a  esp-idf/esp_netif/libesp_netif.a  esp-idf [... cut ...]
/home/neal/esp/esp-adf/components/audio_hal/driver/es8311/es8311.c:202: undefined reference to `i2c_bus_write_bytes'
/home/neal/esp/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.2.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/audio_hal/libaudio_hal.a(es8311.c.obj): in function `es8311_codec_deinit':
/home/neal/esp/esp-adf/components/audio_hal/driver/es8311/es8311.c:258: undefined reference to `i2c_bus_delete'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

meltdown03, 2020-08-02:
I solved the issue by editing the `COMPONENT_PRIV_REQUIRES` in the `$ADF_PATH/components/audio_board/CMakeLists.txt` file. I had to set `esp_peripherals` as the the first item. I attached a patch. Should I make a PR for this issue?
[esp-adf.diff.txt](https://github.com/espressif/esp-adf/files/5012259/esp-adf.diff.txt)

meltdown03, 2020-08-03:
reopening until fixed upstream

jason-mao, 2020-10-12:
@meltdown03 Thanks for your contribution. But, I have not reproduced your issue.
Is it still build error on ADF latest master without modify `$ADF_PATH/components/audio_board/CMakeLists.txt` file?

meltdown03, 2020-10-12:
Yes, it still fails with the same 'undefined references' errors unless I
modify the CMakeLists.txt file.

jason-mao, 2020-10-15:
@meltdown03 Which IDF version and commit ID do you use?

meltdown03, 2020-10-15:
v4.3-dev-1472-g0b71a0a46

meltdown03, 2020-10-20:
I had an issue when renaming due to all the master merges I pulled into the original pull request, but PR #483 should be good to merge

commit 54ae6f09 (2020-10-20): Merge branch 'bugfix/cmake_linking_error' into 'master'

Fix dependency order for 'audio_board' components, fixes #438 linking error

See merge request adf/esp-adf-internal!647
commit 741b8fb5 (2020-10-20): Fix dependency order for 'audio_board' components, fixes #438 linking error
closed 2020-10-20 by espressif-bot