espressif/esp-adf #440: esp-adf-v2.1 compile issues

Patsjemoe, 2020-08-05:
hi, 
when I compile VOIP with adf-v2.1, I get following errors : 
In file included from C:/msys32/path/esp/esp-adf-v2.1/components/audio_stream/tone_stream.c:35:0:
C:/msys32/path/esp/esp-adf-v2.1/components/audio_stream/include/tone_stream.h:112:39: error: unknown type name 'esp_                          app_desc_t'
 esp_err_t tone_partition_get_app_desc(esp_app_desc_t *desc);
                                       ^
C:/msys32/path/esp/esp-adf-v2.1/components/audio_stream/tone_stream.c: In function '_get_flash_tone_info':
C:/msys32/path/esp/esp-adf-v2.1/components/audio_stream/tone_stream.c:101:58: error: 'esp_app_desc_t' undeclared (fi                          rst use in this function)
         start_adr = sizeof(flash_tone_header_t) + sizeof(esp_app_desc_t) + FLASH_TONE_FILE_INFO_BLOCK * index;
                                                          ^
C:/msys32/path/esp/esp-adf-v2.1/components/audio_stream/tone_stream.c:101:58: note: each undeclared identifier is re                          ported only once for each function it appears in
C:/msys32/path/esp/esp-adf-v2.1/components/audio_stream/tone_stream.c: At top level:
C:/msys32/path/esp/esp-adf-v2.1/components/audio_stream/tone_stream.c:174:39: error: unknown type name 'esp_app_desc                          _t'
 esp_err_t tone_partition_get_app_desc(esp_app_desc_t *desc)
Any suggestion on how to solve ?
regards
Ludo

JosephTang, 2020-08-05:
Hi @Patsjemoe,
I had checkout the tags/v2.1 and built the example `examples/advanced_examples/voip`, but no errors occurred. `esp_app_desc_t` is defined in `esp_image_format.h` and this file is included in 'tone_stream.h'. So please check the build environment and try, thanks.

Patsjemoe, 2020-08-05:
Hi Joseph,
thank you for your quick feedback.
Could not find the files under adf-v2.1, but found them under
ESP32-A1S-AudioKit/esp-idf.
Then looked under adf-v2.1/esp-idf, and there the files were...indicating
that I used an old idf_path, corrected it and it worked 😀.
Thanks again
regards
Ludo

closed 2020-08-05 by Patsjemoe